nkolban/esp32-snippets #1: Adafruit neoPixelType order

Smeedy, 2017-01-24:
When setting the pixel the neoPixelType is not taken into account (RGB versus GRB for example). So colours could be off as I had in my setup. One can make a crude hack to shift the pixel colors according to the used type, but a constructor parameter would be nice.

See https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.h

great work

nkolban, 2017-02-04:
Thank you sir ... I'll take a look and do as you ask. ... Neil

nkolban, 2017-02-16:
Howdy @Smeedy  ... I have attempted a fix.  The default order in which colors are sent to the pixels remains "RGB".  However, you can now customize the order.  There is now a function on the WS2812 object class called `setColorOrder(char *)`.  This expects a three character string composed or the characters 'R', 'G' and 'B'.  For example:

```
myWS212.setColorOrder("GRB")
```
Would then send the colors in `GRB` order next time `show()` is called.  I hope this works, if not ... please post back.

nkolban, 2017-04-13:
Closing ... believed resolved.

closed 2017-04-13 by nkolban

nkolban, 2017-04-27:
Awesome ... TY again kind sir and let me know if I can ever help again in the future.