nkolban/esp32-snippets #5: Any details on Makefile setup ?

lakidd, 2017-03-12:
Hi kolban
Love your video's on youtube, but in all your examples, you do a 
`make flash monitor` (I think ?) to kick the whole thing off, however there doesn't appear to be a makefile that I can find in the repo.
Any hints ?

nkolban, 2017-03-12:
Every project that is a user project should be built from this template ...

https://github.com/espressif/esp-idf-template

lakidd, 2017-03-13:
ah got it thanks...so get the blank template, and copy and paste the code across into it ?

nkolban, 2017-03-14:
Yes sir ... that's exactly correct.  For every new project, clone the esp-idf-template and use that as the basis for your new work.  Well .. that's what I do and so far it seems to have worked out ok.

lakidd, 2017-03-15:
Great stuff, thanks and keep up the good work on the books and videos, they
are great !

nkolban, 2017-04-13:
Closing believed resolved.

closed 2017-04-13 by nkolban