nkolban/esp32-snippets #6: C++ "Hello World" Project

GarageGadget, 2017-03-14:
Hi,

I would appreciate it if you can provide the C++ configuration details e.g example "Hello World" project for C++.

This would help fast track myself and others with C++ development on the ESP32.

Keep up the good work.

BR,
Brent

nkolban, 2017-03-14:
@GarageGadget 

Great idea ... thank you for that sir.   Here is a first pass ... have a go and lets see if this meets the need.   Please post back with comments/results.

https://github.com/nkolban/esp32-snippets/blob/master/skeletons/cpp/main_helloworld.cpp

GarageGadget, 2017-03-14:
I replaced the main.c with main_helloworld.cpp file a project based on esp-idf-template project.

I had to add the include directory to the include path to resolve ESP_LOGD declarations. 
The code compiled and flashed without issue but unfortunately none of the application messages were sent to the uart.

A full sample project with compiler configuration etc. would help.

nkolban, 2017-03-14:
Ahhh ... yes ... sorry ... by using ESP_LOGD() that logs at the debug level which is a lower level logging than default.  But it sounds like you got it figured.

Close this issue when you are happy.

closed 2017-03-14 by GarageGadget

mumairanwar92, 2017-07-07:
@GarageGadget I'm having the same issue, will you guide me how did you adjust the log level to get the output ? Thanks in advance.
and @nkolban , Great tutorial and awesome effort writing the book about esp32. Really appreciated.

nkolban, 2017-07-07:
@mumairanwar92 The logging is documented here ... 

http://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html

At a high level, and assuming your are using ESP-IDF, run "make menuconfig" and navigate through the menus to the log settings.  Set the default log output level to debug or lower.  If you need further assistance, don't hesitate to post back.

mumairanwar92, 2017-07-07:
@nkolban thanks for the comment. It has been resolved. you and your tutorials have been a great help throughout.

arunbm123, 2018-03-07:
hi
I am unable to import includes file c_includes.xml.
after giving path. and finish nothing happens

wdouglas1999, 2018-04-17:
Hi all,
I am too unable to import includes from the c_includes.xml. Exactly the same situation as arunbm123.
Has any solution been found? 
Thank you.

chegewara, 2018-04-17:
https://github.com/nkolban/esp32-snippets/blob/master/eclipse/c_includes.xml

burkulesomesh43, 2018-08-10:
@nkolban 
I ma using using esp idf environment on esp32 board. I want to decode url. 
URL>>file:///F:/s?ssid=cabin+336&Passkey=Cabin336%40efc

In above url, 'Cabin336%40efc' includes '%40' which has its decoded value '@'
so how to decode this?
I tried to use base64.h library file from esp idf but gives an error..
fatal error: base64.h: No such file or directory

chegewara, 2018-08-10:
Check you esp-idf installation:
https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/include/wpa2/utils/base64.h

burkulesomesh43, 2018-08-10:
@chegewara 
yes, this file is on this location but when i used    #include "base64.h"  i gives error..

chegewara, 2018-08-10:
Try like this:
https://github.com/espressif/esp-adf/blob/master/examples/cloud_services/google_translate_device/main/google_sr.c#L31

burkulesomesh43, 2018-08-10:
@chegewara 
hey Its working.
thank you so much for help and quick response..