nkolban/esp32-snippets #7: Websocket full example

williamesp2015, 2017-03-29:
would be grateful if you provide websocket full example (send and receive binary and text; sync and Asynchronous).
Thank you

mikepang, 2017-03-30:
There's a nice example here, with working code for linux.  I haven't tried it on esp32 yet.

http://www.linuxhowtos.org/C_C++/socket.htm

SamuelM333, 2017-05-05:
A websocket full example using Mongoose would be very useful. I can't find something clear on the documentation.

Gfast2, 2017-08-19:
Hi @williamesp2015 @SamuelM333 ,
I've tried [_this one_](https://github.com/cesanta/mongoose/tree/master/examples/websocket_chat) from the official [_**mongoose repo**_](https://github.com/cesanta/mongoose). Perhapse you'd like give it a try too. 😃 
If you can't go too much further. I'll try to make one for you.