nkolban/esp32-snippets #8: Compile error building curl - following esp32-snippets/curl/build_files/

mgleason, 2017-04-01:
BTW, thanks for creating the curl build instructions.  I followed the instructions at esp32-snippets/curl/build_files/.  Unfortunately when I make the project, I get the following error:

CC lib/vauth/cleartext.o
/Users/mike/Dev/esp/i2s/components/curl/lib/vauth/cleartext.c:165:1: fatal error: opening dependency file lib/vauth/cleartext.d: No such file or directory
 }
 ^
compilation terminated.
make[1]: *** [lib/vauth/cleartext.o] Error 1
make: *** [curl-build] Error 2

nkolban, 2017-04-01:
Hi my friend ... looking at it now.   If you can, ping me at `#ESP32` on IRC channel that would be great and we can chat live.  Failing that run "ls -l /Users/mike/Dev/esp/i2s/components" and attach the results in a ZIP to this issue.

mgleason, 2017-04-01:
Hi, tried ping you, but you may have stepped away - or it may be my lack of IRC experience...

mgleason, 2017-04-01:
Here's the output from ls -l /Users/jeff/Dev/esp/i2s/components
[out.txt](https://github.com/nkolban/esp32-snippets/files/887949/out.txt)

mgleason, 2017-04-02:
nkolban - Thanks for your help today!

Here's the notes captured from for "Getting the ESP32 curl build working if you're developing on MacOS/OSX":

1. Follow the "Building curl" directions: https://github.com/nkolban/esp32-snippets/tree/master/curl/build_files.
2. Change back to project root (cd ../..).
3. Go ahead and "make" the project and let it fail.  This creates the build/curl/lib/ directory among others.
4. Create two directories:
> mkdir build/curl/lib/vauth
> mkdir build/curl/lib/vtls
5. Modify the last line of component.mk so that it reads:
> CFLAGS+=-DHAVE_CONFIG_H -DBUILDING_LIBCURL -DMBEDTLS_FS_IO

mgleason, 2017-04-02:
Oops, hadn't actually included curl yet.  Now that I've added "#include <curl/curl.h>", the following errors are generated:

/Users/jeff/Dev/esp/i2s/build/curl/libcurl.a(mbedtls.o):(.literal.mbed_connect_common+0x8c): undefined reference to `mbedtls_x509_crt_parse_file'
/Users/jeff/Dev/esp/i2s/build/curl/libcurl.a(mbedtls.o):(.literal.mbed_connect_common+0x90): undefined reference to `mbedtls_x509_crt_parse_path'
/Users/jeff/Dev/esp/i2s/build/curl/libcurl.a(mbedtls.o):(.literal.mbed_connect_common+0x94): undefined reference to `mbedtls_pk_parse_keyfile'
/Users/jeff/Dev/esp/i2s/build/curl/libcurl.a(mbedtls.o):(.literal.mbed_connect_common+0x98): undefined reference to `mbedtls_x509_crl_parse_file'
/Users/jeff/Dev/esp/i2s/build/curl/libcurl.a(mbedtls.o): In function `mbed_connect_step1':
/Users/jeff/Dev/esp/i2s/components/curl/lib/vtls/mbedtls.c:996: undefined reference to `mbedtls_x509_crt_parse_file'
/Users/jeff/Dev/esp/i2s/components/curl/lib/vtls/mbedtls.c:996: undefined reference to `mbedtls_x509_crt_parse_path'
/Users/jeff/Dev/esp/i2s/components/curl/lib/vtls/mbedtls.c:996: undefined reference to `mbedtls_x509_crt_parse_file'
/Users/jeff/Dev/esp/i2s/components/curl/lib/vtls/mbedtls.c:996: undefined reference to `mbedtls_pk_parse_keyfile'
/Users/jeff/Dev/esp/i2s/components/curl/lib/vtls/mbedtls.c:996: undefined reference to `mbedtls_x509_crl_parse_file'

nkolban, 2017-04-02:
Find me on `#ESP32` ... we'll go for round 2 :-)

nkolban, 2017-04-02:
I have a new guess ... are you using the VERY latest ESP-IDF distribution ... see the following which was added in March 2017.   This would be consistent with the problems described ...

https://github.com/espressif/esp-idf/commit/74817c35f3826939337f911263b210f9b7a60efe

If your ESP-IDF is back-level to this ... this would be the problem and the solution would be to use the latest ESP-IDF.

mgleason, 2017-04-02:
That seems to be it!

nkolban, 2017-04-02:
If it works, the recipe in this thread needs to be changed to remove `-DMBEDTLS_FS_IO` which is now the default.

mgleason, 2017-04-03:
Yes, seems to work well.  Tested it with your [testCurl.c](https://github.com/nkolban/esp32-snippets/blob/master/curl/fragments/testCurl.c) code (modified to use example.com instead of the raspberrypi:) successfully returned "<!doctype html><html><head>    <title>Example Domain</title>..."  Interestingly, I believe it worked even with -DMBEDTLS_FS_IO.  Next I need to configure TLS/SSL.  Luckily I've got [your book](https://leanpub.com/kolban-ESP32)!

1. To ensure I was on the latest esp-idf distribution, I needed to clone it from Git.  But, since I originally installed esp-idf following the "official"  [DevKit](https://www.espressif.com/sites/default/files/documentation/esp32-devkitc_getting_started_guide_en.pdf) and [esp-idf](http://www.espressif.com/sites/default/files/documentation/esp-idf_getting_started_guide_en.pdf) "Getting Started Guide"s, I moved the /esp-idf folder to a safe backup location (and later deleted it).
2. git clone --recursive https://github.com/espressif/esp-idf.git
    ...or, if you already installed it from Git...
    git pull
    git submodule update --init --recursive
3. cd to your project folder, edit "component.mk" and remove the "-DMBEDTLS_FS_IO" flag - so the last line reads:
`CFLAGS+=-DHAVE_CONFIG_H -DBUILDING_LIBCURL`

Awesome, thanks again - don't know how you do it!

hassfers, 2017-06-19:
i have a similar question, 
i get an compilation error during the first build caused by `error: unknown type name 'curl_off_t'`. 
do you know where this came from?

nkolban, 2017-06-20:
The `curl_off_t` issue has been tracked down and resolved in this issue:

https://github.com/nkolban/esp32-snippets/issues/18

nkolban, 2017-07-17:
closing with completion.  Don't hesitate tto re-open as needed.

closed 2017-07-17 by nkolban