nkolban/esp32-snippets #9: how to use  cpp_util in couple with esp-idf/example/blufi

rogerustb, 2017-04-11:
Hi nkolban:
Thanks for your esp32-snippets, they are very useful to me.

I want to use SockServ.h/cpp in esp-idf/example/blufi .  So I copied cpp_util/ to blufi/main/ and add one line(COMPONENT_ADD_INCLUDEDIRS=./cpp_utils/) in blufi/main/component.mk.
When I typed command "make", I got the following 

`$ make
CC blufi_example_main.o
In file included from /home/roger/esp/myapp/bluetooth/blufi/main/./blufi_example_main.c:38:0:
/home/roger/esp/myapp/bluetooth/blufi/main/cpp_utils/SockServ.h:6:18: fatal error: string: No such file or directory
compilation terminated.
/home/roger/esp/esp-idf/make/component_wrapper.mk:176: recipe for target 'blufi_example_main.o' failed
make[1]: *** [blufi_example_main.o] Error 1
/home/roger/esp/esp-idf//make/project.mk:386: recipe for target 'main-build' failed
make: *** [main-build] Error 2
`
Then, I renamed blufi_example_main.c to blufi_example_main.cpp. And got another error:
`make[1]: *** No rule to make target '/home/roger/esp/myapp/bluetooth/blufi/main/./blufi_example_main.c', needed by 'blufi_example_main.o'.  Stop.
/home/roger/esp/esp-idf//make/project.mk:386: recipe for target 'main-build' failed
make: *** [main-build] Error 2
`
So, what should I do ?

nkolban, 2017-04-11:
Howdy ... are you using the ESP-IDF build system?  It seems to automatically add the necessary paths to the compilation.

The missing include is called <string> (yes this is "string" and not "string.h").  In C++ we have what is called the Standard Template Library (STL) which is a set of pre-defined C++ classes that are assured to be available in a C++ implementation.  These libraries and header files are supplied by the compiler.

In my experience, running "make" against the latest Espressif ESP-IDF has always built without issue.

I'm at your disposal to get to the bottom of this.

rogerustb, 2017-04-12:
Thanks for your reply. I try to update my disposal.

`$ git fetch https://github.com/espressif/esp-idf.git
remote: Counting objects: 248, done.
remote: Compressing objects: 100% (243/243), done.
Receiving objects: 100% (248/248), 283.89 KiB | 265.00 KiB/s, done.
remote: Total 248 (delta 92), reused 0 (delta 0), pack-reused 0
Resolving deltas: 100% (92/92), done.
From https://github.com/espressif/esp-idf
 * branch            HEAD       -> FETCH_HEAD
√ 07:10:31 roger:~/esp/esp-idf (master) [6 days ago] ?
$ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/master
  remotes/origin/release/v2.0
√ 07:10:44 roger:~/esp/esp-idf (master) [6 days ago] ?
$ git merge origin/master
Already up-to-date.`

I have copied `cpp_util/` to `eps-idf/components/`.

Then, I add `#include "SockServ.h"` to `esp/myapp/main/main.c` . And type command `make` , the result is 
`In file included from /home/roger/esp/myapp/main/./main.c:8:0:
/home/roger/esp/esp-idf/components/cpp_utils/./SockServ.h:6:18: fatal error: string: No such file or directory
compilation terminated.
/home/roger/esp/esp-idf/make/component_wrapper.mk:176: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/roger/esp/esp-idf//make/project.mk:386: recipe for target 'main-build' failed
make: *** [main-build] Error 2`

nkolban, 2017-04-12:
I'm not sure if you saw this ... if not, please take a look ...
https://www.youtube.com/watch?v=-ttiPfmrehU
Let me ask you to study this a couple of times.  It is a general video on building an arbitrary C++ application.

Lets see if that helps.
If not, are you able to compile and link in an arbitrary C++ piece of code?

Failing all this, what we can do is arrange an appointment for an hour on the weekend and setup a skype screen share where you can show me live the problem and we can look at it together and see if we can spot the issue.  I'm certain that there is something "wrong" in how you have set up the environment but I don't know what it might be.  I'd suggest studying in depth exactly what it takes to build and compile arbitrary ESP32 applications especially those involving C++.

rogerustb, 2017-04-12:
Thanks. I have not saw the video before.
My environment is wrong.

rogerustb, 2017-04-26:
I have saw the video.  But, I use vim rather than eclipse . So, could you make a video on  building an arbitrary C++ application in vim environment ?

rogerustb, 2017-04-27:
I removed esp-idf and git clone ...again .  It just works. Thanks a lot.

closed 2017-04-27 by rogerustb

jonface, 2017-11-23:
(replied to wrong issue, deleted it, just in case you saw it)

I'm getting on a clean checkout of the esp-idf (today), 

```
In file included from C:/Users/jmasters/Downloads/esp32_win32_msys2_environment_and_toolchain-20170918/msys32/home/jmasters/esp/gatt_client/main/gattc_demo.c:4:0:
C:/Users/jmasters/Downloads/esp32_win32_msys2_environment_and_toolchain-20170918/msys32/home/jmasters/esp/gatt_client/components/cpp_utils/BLEUtils.h:15:18: fatal error: string: No such file or directory
compilation terminated.
make[1]: *** [/home/jmasters/esp/esp-idf/make/component_wrapper.mk:243: gattc_demo.o] Error 1
make: *** [C:/Users/jmasters/Downloads/esp32_win32_msys2_environment_and_toolchain-20170918/msys32/home/jmasters/esp/esp-idf/make/project.mk:448: component-main-build] Error 2
```
I've hijacked the gatt_client example in case you were wondering, I've looked in menuconfig, do I need to enable the STL somehow?

Thanks

mecowhy, 2017-11-27:
@jonface rename your main.c to main.cpp

nkolban, 2017-11-27:
@jonface I think you are using this demo ... https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/gatt_client

This demo uses the ESP-IDF supplied BLE APIs.  This is unrelated to the code found in this repository.  The code here is a C++ encapsulation and model using those lower level APIs.  I'm afraid that this repository isn't  a good place to ask questions on low level BLE APIs ... might I suggest posting to the ESP-IDF forum found here?

https://esp32.com/index.php