nkolban/esp32-snippets #11: problem making i2c scanner

stiivik, 2017-04-15:
Hi

I'm trying to make i2c scanner through Eclipse, but it won't make. If I try to make esp-idf [i2c main example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2c/main/i2c_example_main.c) then it compiles and runs fine. The error I'm getting with i2c scanner is following:
`make all 
CC main.o
AR libmain.a
LD app-template.elf
C:/esp32idf/workspace/myap/build/esp32\libesp32.a(cpu_start.o):(.literal.main_task+0x8): undefined reference to `app_main'
C:/esp32idf/workspace/myap/build/esp32\libesp32.a(cpu_start.o): In function `main_task':
C:/esp32idf/esp-idf/components/esp32/cpu_start.c:284: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
make: *** [C:\esp32idf\esp-idf/make/project.mk:323: /c/esp32idf/workspace/myap/build/app-template.elf] Error 1`

Been searching everywhere for answer, but nothing comes up. Maybe anyone here can help me?

nkolban, 2017-04-15:
Howdy .... from the error, it looks like you are building your own app and this is not an issue related to the ESP32-snippets project?  If that's the case, post your question to the ESP32 forum and it can be addressed there:
https://esp32.com/index.php

nkolban, 2017-04-15:
Are you using the ESP32 Snippet code found in this project?

stiivik, 2017-04-16:
Hi. Yes I'm using the i2c scanner from the snippet. The weird thing is that it the esp32 i2c main example works perfect, so it may be due to some library or something? Code itself doesn't give any errors, just when i try to make.

stiivik, 2017-04-17:
Hi again. Found my mistake, i had the app_main missing totally. I just copied your code and tried to run it. I just received my unit and i'm totally new to this, so sorry for wasting your time. Most of the info I'm using is from your videos, they really help a lot so thanks again!

closed 2017-04-17 by stiivik

nkolban, 2017-04-17:
No problem my friend.  Glad you got it solved.  I'm here for you as needed.