nkolban/esp32-snippets #13: Build error - do not seem to have socketpair as part of components

ponnappatallac, 2017-04-25:
I followed the instruction on building libcurl (Thank you for the detail steps and video).  However I seem to be getting the following error : 
/home/user/esp/libcurl4/components/curl/lib/curl_ntlm_wb.c: In function 'Curl_ntlm_wb_cleanup':
/home/user/esp/libcurl4/components/curl/lib/curl_ntlm_wb.c:91:9: error: implicit declaration of function 'kill' [-Werror=implicit-function-declaration]
         kill(conn->ntlm_auth_hlpr_pid, SIGTERM);
         ^
/home/user/esp/libcurl4/components/curl/lib/curl_ntlm_wb.c: In function 'ntlm_wb_init':
/home/user/esp/libcurl4/components/curl/lib/curl_ntlm_wb.c:187:6: error: implicit declaration of function 'socketpair' [-Werror=implicit-function-declaration]
   if(socketpair(AF_UNIX, SOCK_STREAM, 0, sockfds)) {
      ^
/home/user/esp/libcurl4/components/curl/lib/curl_ntlm_wb.c:187:17: error: 'AF_UNIX' undeclared (first use in this function)
   if(socketpair(AF_UNIX, SOCK_STREAM, 0, sockfds)) {
                 ^
/home/user/esp/libcurl4/components/curl/lib/curl_ntlm_wb.c:187:17: note: each undeclared identifier is reported only once for each function it appears in

nkolban, 2017-05-08:
Thanks guys.  Sorry for the delay in getting back to y'all.   I have updated the project sample files here and all should now work.  The addition was

```
#define CURL_DISABLE_NTLM 1
```
`into lib/curl_config.h`

Many thanks to @mgleason for spending live time helping me debug this.

@ponnappatallac  Please check that it works for you now and let me know.

ponnappatallac, 2017-05-09:
Thank you for your help.

nkolban, 2017-07-07:
May we close this issue?

mgleason, 2017-08-19:
yes (from my viewpoint)

closed 2017-08-19 by nkolban