nkolban/esp32-snippets #16: Webserver example

skateone, 2017-06-08:
hi there. 
does anyone have a webserver example for me based on the code snippets here.

nkolban, 2017-06-16:
Howdy bud ... I'll work with you to get what you need.  Lets collaborate together to build a simple sample.  Email me at kolban1@kolban.com

skateone, 2017-06-16:
> Howdy bud ... I'll work with you to get what you need. Lets collaborate together to build a simple sample. Email me at kolban1@kolban.com <mailto:kolban1@kolban.com>

hi there..
Firstly i must say thank you for the access to your code snippets. they have shed a lot of light on this architecture.

i found someone that ported the Arduino 8266 WebServer over and that seems to be working great so far. 
Would love to use mongoose as it offers all the required interfaces i need.  MQTT HTTP WEBSOCKETS

i would like to use your framework and can help get a sample going as well for you..

Skype: shaun.reynolds.skate
regards
Shaun

nkolban, 2017-09-14:
The whole story on HTTP Servers in this library has been re-written.
See:
https://github.com/nkolban/esp32-snippets/blob/master/Documentation/C%2B%2B%20HTTP%20Server.pdf

closed 2017-11-30 by nkolban

snahmad, 2018-05-24:
Is anyone using WebServer class which uses mongoose.