nkolban/esp32-snippets #18: Unable to make curl/fragments/testCurl.c

ben39, 2017-06-20:
I am using Eclipse on Ubuntu VM. The nkolban instructions for setting up Eclipse have been followed. esp template cloned to myapp, ran the make menuconfig, curl/build_files copied, and Eclipse includes set up. Not sure what is causing the issue here.
[http://i.imgur.com/FWHG7RP.png](url)

nkolban, 2017-06-20:
testing now ...

nkolban, 2017-06-20:
recreated ...

nkolban, 2017-06-20:
Problem found ... CURL changed their build system ... https://github.com/curl/curl/commit/73a2fcea0b4adea6ba342cd7ed1149782c214ae3

nkolban, 2017-06-20:
Problem needs to be researched to be resolved properly.

nkolban, 2017-06-20:
A solution has been documented.   New instructions for building libcurl can now be found here.

https://github.com/nkolban/esp32-snippets/tree/master/curl/build_files

Please test and report if all now works or if we have further problems.

ben39, 2017-06-21:
This fixes the curl_off_t errors, thanks.
curl/include/curl/system.h #define CURL_TYPEOF_CURL_SOCKLEN_T was also causing error until I changed to unsigned int