nkolban/esp32-snippets #19: No Application Message on UART

mumairanwar92, 2017-07-07:
Great tutorial writing C++ Applications and awesome effort writing the book about esp32. Really appreciated.
I have followed your tutorial, the code was compiled and flashed without issue but unfortunately none of the application messages were sent to the UART. Someone mentioned a little tweaking in the ESP_LOG file but I couldn't figure it out.
Can anybody help me out in this issue ?

nkolban, 2017-07-07:
I believe that you need to study here:

http://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html

I think you will need to set the debug output level via "make menuconfig".  However, if this doesn't solve the issue, post back and we'll pick it up again.

mumairanwar92, 2017-07-07:
@nkolban thanks for the comment. It has been resolved. you and your tutorials have been a great help throughout.

closed 2017-07-07 by mumairanwar92