nkolban/esp32-snippets #20: BLE.h: No file or Directory found

mumairanwar92, 2017-07-10:
I'm writing a BLE C++ Application following the Kolban's tutorial. I have included all the source files as mentioned in the tutorials, Everyfile is accepted but only one is excluded "BLE.h". the file is present in the include directory but error is still there, I have check the "menuconfig", bluetooth is enabled there. 
A little help would be appreciated.

[image]
[image]
[image]

nkolban, 2017-07-10:
Howdy ... my focus this week will be polishing up the documentation and producing samples.  If I may, let me write this up ASAP and post a link to the materials here.  Hopefully it will serve us both.  I'll work with you directly so that you get built what you want and you can give me feedback on how easy to understand the documentation is/will be.

mumairanwar92, 2017-07-10:
Sure, that would be great. Thankyou. I'll glad if i can in some way.

Sent from iOS

nkolban, 2017-07-17:
Samples and a guide are now up.

nkolban, 2017-09-17:
Howdy ... is there anything else we can do related to this issue?  Is it ready to be closed out?

mumairanwar92, 2017-09-18:
Yes, thank for helping, I'm closing this issue.
I'll PM you if I will have any problem. Thanks Again.

closed 2017-09-18 by mumairanwar92