nkolban/esp32-snippets #21: GPIO interrupt example

skateone, 2017-07-11:
Hi there again. i love your code... at the moment i am playing with you GPIO classes and have noticed you allow to enable and disable interrupts.  
My question would be, where are the Interrupt ISR's handled as i dont see any way for notification on interrupt triggering. normally you expose a callback function pointer or something

nkolban, 2017-07-12:
Let me look at this ... it is possible I haven't added interrupt handling ... if that's the case, I'll make that a priority to get added.

skateone, 2017-07-12:
added some basic methods to the class will share a bit later with you…

AirCycler, 2018-01-05:
Hi,
I need help with interrupts. Can I use two interrupts on ESP-WROOM-32 like
void setup(){
Serial.begin(115200); 
attachInterrupt(digitalPinToInterrupt(16), ONEISR, HIGH);
 attachInterrupt(digitalPinToInterrupt(17), TWOISR, HIGH);
}

chegewara, 2018-01-05:
Sure think you can. You can have up to 16 interrupts if i remember, or maybe even on every GPIO pin.
But i can asure you that your code is good, i have similar and it works.

AirCycler, 2018-01-05:
I am facing a problem here. Without touching the interrupt pin, it is changing its state. I don't understand.  When I change 16 pin interrupt, then after sometime automatically 17 pin interrupt is getting called.

could you please help?

chegewara, 2018-01-05:
Try with PULLDOWN setup.
```
  pinMode(16, INPUT_PULLDOWN);
  pinMode(17, INPUT_PULLDOWN);

```

AirCycler, 2018-01-05:
I am using hardware pull-ups. So, I am just using  
pinMode(16, INPUT);
pinMode(17, INPUT);

chegewara, 2018-01-05:
Maybe hardware pullups are to weak? Also just for try use CHANGE triggering in place of HIGH.

AirCycler, 2018-01-05:
Actually, I was using internal pull-ups then I faced same behavior. I though they are too week so I used 10K pull-ups later but the behavior is still the same. I want to call the ISR if the pins go HIGH. so CHANGE won't work for me.

chegewara, 2018-01-05:
I understand, just asked you to make test with CHANGE to see if this also will cause triggering interrupts on pin 17. Besides there is no interrupt triggering type HIGH, its ONHIGH.
https://github.com/espressif/arduino-esp32/blob/a6c02bc71024308deadc3c0e04e137bc621069f6/cores/esp32/esp32-hal-gpio.h#L51-L58

nkolban, 2018-01-05:
A couple of things to check ... first of all ... is it just these pins ... if we test with other pins do we get the same results?

If we don't use interrupt handling but instead place the GPIOs in a tight polling loop, do still see spurious data?

What do you have connected to these GPIOs and by how long a connection?  If your wires are too long, you can pick up noise.

Finally, do you have a logic analyzer you can connect to the pins?  I'd be curious to see if there is indeed a signal on the pins when you detect the change.

AirCycler, 2018-01-08:
@chegewara If I use ONHIGH , even though i do not make it HIGH it is calling ISR continuously . CHANGE is giving me the same thing like HIGH. It seems like working for a while but it is not stable.

AirCycler, 2018-01-16:
@nkolban I checked with other pins also and I am getting the same result.

If I use polling loop also I am getting the same result. 

I have connected push buttons to these GPIO pins with 10k pull up resistor and its on PCB. I didn't understand on how long a connection?
I am not checking with logic analyzer but i am using scope to monitor the voltage on pins. They are changing continuously with out me operating on push buttons.

Could you please help me? I was thinking the processor is bouncing the push buttons pins all the time once after I pressed the buttons. I f I don't press the push buttons after I dumped the code its working fine.

vtongcoder, 2018-03-13:
I have the same issue.
My setup: INPUT_PULLUP.
When press button, many interrupts occurred continuously.

kgotsis, 2018-03-16:
Correct me if I am wrong, but while the pin remains high you will be getting new interrrupts continiously. A single button press is a really long time for the processor, so when you exit your ISR re-enabling interrupts, it will fire again.