nkolban/esp32-snippets #23: Webserver Mongoose Wrapper Memory leak

skateone, 2017-07-12:
Another interesting thing i picked up using mongoose with your CPP wrappers is the esp32 eats into the free heap and doesnt release it.
tried two examples.
1: with a uint8_t encoded home page with size of around 24KB.. the initial call consumes about 50KB ram then after the response most that is released. page refreshes seem stable.
2: have two URL paths that respond with JSON  ( to be used on the client side )
- the first using cJSON to construct the payload then doing the send and then releasing the cJSON
   --this example consumes about 1kb ram for each request from my client without releasing.

- a simple path that responds with JSON ( built up as a constant )
  -- this also eats ram and doesn't release.

i know not my cJSON constructors as is do delete the objects after construction.

Have you had similar results?

nkolban, 2017-07-12:
At this point, I haven't tried any soak tests nor looked at memory consumption over time.  I have seen that there is an initial usage cost for things like WiFi and BLE that isn't present until after some use.  For example, I found that when I ran a web server, free memory at start declined sharply after the first couple of connections but then seemed to become stable.  I didn't look to see if there were "finer" grained leaks ... but there could certainly be.  We'll have to do analysis to see if they are coming from Mongnoose, C++ wrapper code, ESP-IDF or Standard C++ libraries.  But ... we'll get there.

nkolban, 2017-11-30:
Unfortunately my interest in Mongoose has wained now that we have a native C++ HttpServer set of classes that don't involve that library.  While I don't know whether or not there is a problem with the C++ wrappers for mongoose, I can't promise that they will be examined.

closed 2017-11-30 by nkolban