nkolban/esp32-snippets #24: ble lib compile errors

rami-alisawi, 2017-07-14:
Had the following when try to use ReadSampl.cpp from ble tests folder.

I guess it is because the difference between my esp-idf version and the one you have developed on. Can you share the version that you are sure it is working on. 

`
Compiling .pioenvs/esp32dev/lib/cpputils/BLEClientCallbacks.o
Compiling .pioenvs/esp32dev/lib/cpputils/BLEDesciptorMap.o
lib/cpputils/src/BLE.cpp: In static member function 'static void BLE::gapEventHandler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*)':
lib/cpputils/src/BLE.cpp:118:8: error: 'ESP_GAP_BLE_SEC_REQ_EVT' was not declared in this scope
case ESP_GAP_BLE_SEC_REQ_EVT: {
^
lib/cpputils/src/BLE.cpp:119:22: error: '::esp_ble_gap_security_rsp' has not been declared
esp_err_t errRc = ::esp_ble_gap_security_rsp(param->ble_security.ble_req.bd_addr, true);
^
lib/cpputils/src/BLE.cpp:119:56: error: 'union esp_ble_gap_cb_param_t' has no member named 'ble_security'
esp_err_t errRc = ::esp_ble_gap_security_rsp(param->ble_security.ble_req.bd_addr, true);
^
lib/cpputils/src/BLE.cpp: In static member function 'static void BLE::initServer(std::__cxx11::string)':
lib/cpputils/src/BLE.cpp:146:2: error: 'esp_bt_controller_config_t' was not declared in this scope
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
^
lib/cpputils/src/BLE.cpp:147:44: error: 'bt_cfg' was not declared in this scope
esp_err_t errRc = esp_bt_controller_init(&bt_cfg);
^
lib/cpputils/src/BLE.cpp:190:2: error: 'esp_ble_io_cap_t' was not declared in this scope
esp_ble_io_cap_t iocap = ESP_IO_CAP_NONE;
^
lib/cpputils/src/BLE.cpp:191:10: error: '::esp_ble_gap_set_security_param' has not been declared
errRc = ::esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
^
lib/cpputils/src/BLE.cpp:191:43: error: 'ESP_BLE_SM_IOCAP_MODE' was not declared in this scope
errRc = ::esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
^
lib/cpputils/src/BLE.cpp:191:67: error: 'iocap' was not declared in this scope
errRc = ::esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
^
lib/cpputils/src/BLE.cpp: In static member function 'static void BLE::initClient()':
lib/cpputils/src/BLE.cpp:205:3: error: 'esp_bt_controller_config_t' was not declared in this scope
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
^
lib/cpputils/src/BLE.cpp:206:45: error: 'bt_cfg' was not declared in this scope
esp_err_t errRc = esp_bt_controller_init(&bt_cfg);
^
`

`
*** [.pioenvs/esp32dev/lib/cpputils/BLE.o] Error 1
Compiling .pioenvs/esp32dev/lib/cpputils/BLEDescriptor.o
lib/cpputils/src/BLECharacteristic.cpp: In constructor 'BLECharacteristic::BLECharacteristic(BLEUUID, uint32_t)':
lib/cpputils/src/BLECharacteristic.cpp:36:23: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties         = 0;
^
lib/cpputils/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setBroadcastProperty(bool)':
lib/cpputils/src/BLECharacteristic.cpp:345:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties |= ESP_GATT_CHAR_PROP_BIT_BROADCAST;
^
lib/cpputils/src/BLECharacteristic.cpp:347:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties &= ~ESP_GATT_CHAR_PROP_BIT_BROADCAST;
^
lib/cpputils/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setIndicateProperty(bool)':
lib/cpputils/src/BLECharacteristic.cpp:369:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties |= ESP_GATT_CHAR_PROP_BIT_INDICATE;
^
lib/cpputils/src/BLECharacteristic.cpp:371:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties &= ~ESP_GATT_CHAR_PROP_BIT_INDICATE;
^
lib/cpputils/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setNotifyProperty(bool)':
lib/cpputils/src/BLECharacteristic.cpp:379:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties |= ESP_GATT_CHAR_PROP_BIT_NOTIFY;
^
lib/cpputils/src/BLECharacteristic.cpp:381:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties &= ~ESP_GATT_CHAR_PROP_BIT_NOTIFY;
^
lib/cpputils/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setReadProperty(bool)':
lib/cpputils/src/BLECharacteristic.cpp:389:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties |= ESP_GATT_CHAR_PROP_BIT_READ;
^
lib/cpputils/src/BLECharacteristic.cpp:391:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties &= ~ESP_GATT_CHAR_PROP_BIT_READ;
^
lib/cpputils/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setWriteNoResponseProperty(bool)':
lib/cpputils/src/BLECharacteristic.cpp:429:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties |= ESP_GATT_CHAR_PROP_BIT_WRITE_NR;
^
lib/cpputils/src/BLECharacteristic.cpp:431:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties &= ~ESP_GATT_CHAR_PROP_BIT_WRITE_NR;
^
lib/cpputils/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setWriteProperty(bool)':
lib/cpputils/src/BLECharacteristic.cpp:439:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties |= ESP_GATT_CHAR_PROP_BIT_WRITE;
^
lib/cpputils/src/BLECharacteristic.cpp:441:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties &= ~ESP_GATT_CHAR_PROP_BIT_WRITE;
^

lib/cpputils/src/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setWriteProperty(bool)':
lib/cpputils/src/BLECharacteristic.cpp:439:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties |= ESP_GATT_CHAR_PROP_BIT_WRITE;
^
lib/cpputils/src/BLECharacteristic.cpp:441:16: error: invalid conversion from 'int' to 'esp_gatt_char_prop_t' [-fpermissive]
m_properties &= ~ESP_GATT_CHAR_PROP_BIT_WRITE;
^
*** [.pioenvs/esp32dev/lib/cpputils/BLECharacteristic.o] Error 1
lib/cpputils/src/BLEDescriptor.cpp: In member function 'void BLEDescriptor::executeCreate(BLECharacteristic*)':
lib/cpputils/src/BLEDescriptor.cpp:67:12: error: invalid conversion from 'int' to 'esp_gatt_perm_t' [-fpermissive]
&control);
^
In file included from lib/cpputils/src/BLEService.h:13:0,
from lib/cpputils/src/BLEDescriptor.cpp:16:
/Users/rami/.platformio/packages/framework-espidf/components/bt/bluedroid/api/include/esp_gatts_api.h:393:11: note:   initializing argument 3 of 'esp_err_t esp_ble_gatts_add_char_descr(uint16_t, esp_bt_uuid_t*, esp_gatt_perm_t, esp_att
r_value_t*, esp_attr_control_t*)'
esp_err_t esp_ble_gatts_add_char_descr (uint16_t service_handle,
^
*** [.pioenvs/esp32dev/lib/cpputils/BLEDescriptor.o] Error 1
========================== [ERROR] Took 7.49 seconds ==========================

`

nkolban, 2017-07-14:
Howdy,
I tend to always use the very latest version of the ESP-IDF from Github ... the one that is "master".  So far this has worked out for me well.  To be honest, I don't know what the _minimum_ version of ESP-IDF would be in order to support the C++ BLE classes.  If you know what version you want to use, I can likely test that to ensure that it works.  Other than that, download the latest ESP-IDF and lets see what happens.

If the problem persists, we can set up a live chat (kolban on IRC in channel ESP32) or via skype for a screen share.  Either way, realize that I'm more than happy to assist (with respect to the code that I make available here).

rami-alisawi, 2017-07-14:
solved, thanks.

closed 2017-07-14 by rami-alisawi

nkolban, 2017-07-14:
What was the resolution?

rami-alisawi, 2017-07-14:
"download the latest ESP-IDF and lets see what happens." it worked with the latest.

nkolban, 2017-07-14:
Awesome ... thanks for reporting.