nkolban/esp32-snippets #34: task_test_SSD1306i2c  for Arduino IDE + ESP32

blotfi, 2017-08-11:
Hi
Thanks for the HIL
but I didn't succeed in compiling a simple example with my environnement
I copier u8g2_esp32_hal.c/.h to the h8g2 lib directory : c:\Users\myname\Documents\Arduino\libraries\U8g2\src\clib\
but the compiler complains about the h8g2.h (didn't find it)
though I just compiled and use successfully, the h8g2 lib with a Node MCU (8266) but with ESP32 Lolin32, it didn't
here is my example that dos'nt compile
(my environnement works, because I just compiled installed Web Radio ESP32: and it works great, I can https://github.com/Edzelf/ESP32-Radio

it worked well, but I want to display the canal radio and the song, so how can I correct the code 

`//#include <driver/gpio.h>
//#include <driver/spi_master.h>
//#include <esp_log.h>
//#include <freertos/FreeRTOS.h>
//#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include <u8g2.h>

//#include "sdkconfig.h"
#include "u8g2_esp32_hal.h"

// SDA - GPIO21
#define PIN_SDA 21

// SCL - GPIO22
#define PIN_SCL 22

static const char *TAG = "ssd1306";

void task_test_SSD1306i2c(void *ignore) {
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.sda   = PIN_SDA;
	u8g2_esp32_hal.scl  = PIN_SCL;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; // a structure which will contain all the data for one display
	u8g2_Setup_ssd1306_128x32_univision_f(
		&u8g2,
		U8G2_R0,
		//u8x8_byte_sw_i2c,
		u8g2_esp32_msg_i2c_cb,
		u8g2_esp32_msg_i2c_and_delay_cb);  // init u8g2 structure
	u8x8_SetI2CAddress(&u8g2.u8x8,0x78);

	ESP_LOGI(TAG, "u8g2_InitDisplay");
	u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,

	ESP_LOGI(TAG, "u8g2_SetPowerSave");
	u8g2_SetPowerSave(&u8g2, 0); // wake up display
	ESP_LOGI(TAG, "u8g2_ClearBuffer");
	u8g2_ClearBuffer(&u8g2);
	ESP_LOGI(TAG, "u8g2_DrawBox");
	u8g2_DrawBox(&u8g2, 0, 26, 80,6);
	u8g2_DrawFrame(&u8g2, 0,26,100,6);

	ESP_LOGI(TAG, "u8g2_SetFont");
    u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	ESP_LOGI(TAG, "u8g2_DrawStr");
    u8g2_DrawStr(&u8g2, 2,17,"Hi nkolban!");
	ESP_LOGI(TAG, "u8g2_SendBuffer");
	u8g2_SendBuffer(&u8g2);

	ESP_LOGI(TAG, "All done!");

	vTaskDelete(NULL);
}

int val = 0;
void setup() {
  Serial.begin(9600);
  task_test_SSD1306i2c();
    }

void loop() {
  // put your main code here, to run repeatedly:
  val = hallRead();
  // print the results to the serial monitor:
  //Serial.print("sensor = ");
  Serial.println(val);//to graph 
}`

blotfi, 2017-08-11:
Solved
I didn't copy the lib to the directory of ESP32 which is different from other targets (arduino ESP8266)
Also, instead of including 
#include <u8g2.h>
we have to put
#include <U8g2lib.h>

nkolban, 2017-08-11:
Thank you sir ... I'll check the instructions out at the weekend and see if we can't make them clearer (or working).

blotfi, 2017-08-12:
No, it is all my fault
I thought it was for Arduino IDE and it is not
now for Arduino IDE it works fine without your HAL

nkolban, 2017-08-15:
Great news.  Closing ... feel free to open again as needed.

closed 2017-08-15 by nkolban