nkolban/esp32-snippets #35: BLE_scan error!    Other 3 work fine

mxwang101, 2017-08-12:
Arduino:1.8.1 (Windows 7), 开发板："ESP32 Dev Module, QIO, 80MHz, 921600, None"

C:\Users\wangxin\Documents\Arduino\BLE20170813\BLE\examples\BLE20170813\BLE\examples\BLE_scan\BLE_scan.ino: In function 'void setup()':

BLE_scan:23: error: invalid new-expression of abstract class type 'MyAdvertisedDeviceCallbacks'

   pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());

                                                                          ^

C:\Users\wangxin\Documents\Arduino\BLE20170813\BLE\examples\BLE20170813\BLE\examples\BLE_scan\BLE_scan.ino:11:7: note:   because the following virtual functions are pure within 'MyAdvertisedDeviceCallbacks':

 class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {

       ^

In file included from C:\Users\wangxin\Documents\Arduino\libraries\ESP32_BLE\src/BLEScan.h:15:0,

                 from C:\Users\wangxin\Documents\Arduino\libraries\ESP32_BLE\src/BLE.h:20,

                 from C:\Users\wangxin\Documents\Arduino\BLE20170813\BLE\examples\BLE20170813\BLE\examples\BLE_scan\BLE_scan.ino:6:

C:\Users\wangxin\Documents\Arduino\libraries\ESP32_BLE\src/BLEAdvertisedDevice.h:102:15: note: 	virtual void BLEAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)

  virtual void onResult(BLEAdvertisedDevice advertisedDevice) = 0;

               ^

BLE_scan:25: error: conversion from 'BLEScanResults' to non-scalar type 'std::vector<BLEAdvertisedDevice*>' requested

   std::vector<BLEAdvertisedDevice*> foundDevices = pBLEScan->start(30);  //scan for 30 seconds

                                                                   ^

exit status 1
invalid new-expression of abstract class type 'MyAdvertisedDeviceCallbacks'

nkolban, 2017-08-12:
Howdy,
Can you pastebin or attach your Arduino script that contains the BLE functions?  I think I'll need to see the code in context to assist.

copercini, 2017-08-12:
Probably you are running an updated code (after https://github.com/nkolban/esp32-snippets/commit/4401dc05e44b442a34e85400ab9ce92ca80b2d29) from this repo, but with old example

mxwang101, 2017-08-13:
@nkolban ,

attach the Arduino script:
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLE.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
#include <esp_log.h>
#include <string>
#include "sdkconfig.h"

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice *pAdvertisedDevice) {
      Serial.printf("Advertised Device: %s \n", pAdvertisedDevice->toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLE::initClient();
  BLEScan* pBLEScan = BLE::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  std::vector<BLEAdvertisedDevice*> foundDevices = pBLEScan->start(30);  //scan for 30 seconds
  Serial.print("Devices found: ");
  Serial.println(foundDevices.size());
  Serial.println("Scan done!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}

mxwang101, 2017-08-13:
AND the errors

Arduino:1.8.1 (Windows 7), 开发板："ESP32 Dev Module, QIO, 40MHz, 921600, None"

C:\BLE\examples\BLE_scan\BLE_scan.ino: In function 'void setup()':

BLE_scan:26: error: invalid new-expression of abstract class type 'MyAdvertisedDeviceCallbacks'

   pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());

                                                                          ^

C:\BLE\examples\BLE_scan\BLE_scan.ino:14:7: note:   because the following virtual functions are pure within 'MyAdvertisedDeviceCallbacks':

 class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {

       ^

In file included from C:\Users\wangxin\Documents\Arduino\libraries\ESP32_BLE\src/BLEScan.h:15:0,

                 from C:\Users\wangxin\Documents\Arduino\libraries\ESP32_BLE\src/BLE.h:20,

                 from C:\BLE\examples\BLE_scan\BLE_scan.ino:6:

C:\Users\wangxin\Documents\Arduino\libraries\ESP32_BLE\src/BLEAdvertisedDevice.h:102:15: note: 	virtual void BLEAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)

  virtual void onResult(BLEAdvertisedDevice advertisedDevice) = 0;

               ^

BLE_scan:28: error: conversion from 'BLEScanResults' to non-scalar type 'std::vector<BLEAdvertisedDevice*>' requested

   std::vector<BLEAdvertisedDevice*> foundDevices = pBLEScan->start(30);  //scan for 30 seconds

                                                                   ^

exit status 1
invalid new-expression of abstract class type 'MyAdvertisedDeviceCallbacks'

在文件 -> 首选项开启
“编译过程中显示详细输出”选项
这份报告会包含更多信息。

mxwang101, 2017-08-13:
more detail :Arduino:1.8.1 (Windows 7), 开发板："ESP32 Dev Module, QIO, 40MHz, 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\wangxin\AppData\Local\Arduino15\packages -hardware C:\Users\wangxin\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\wangxin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\wangxin\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=40,UploadSpeed=921600,DebugLevel=none -ide-version=10801 -build-path C:\Users\wangxin\AppData\Local\Temp\arduino_build_769564 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\BLE\examples\BLE_scan\BLE_scan.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\wangxin\AppData\Local\Arduino15\packages -hardware C:\Users\wangxin\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\wangxin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\wangxin\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=40,UploadSpeed=921600,DebugLevel=none -ide-version=10801 -build-path C:\Users\wangxin\AppData\Local\Temp\arduino_build_769564 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\BLE\examples\BLE_scan\BLE_scan.ino
Using board 'esp32' from platform in folder: C:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\wangxin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-var [... cut ...]
  virtual void onResult(BLEAdvertisedDevice advertisedDevice) = 0;

               ^

BLE_scan:28: error: conversion from 'BLEScanResults' to non-scalar type 'std::vector<BLEAdvertisedDevice*>' requested

   std::vector<BLEAdvertisedDevice*> foundDevices = pBLEScan->start(30);  //scan for 30 seconds

                                                                   ^

使用 1.0  版本的库 ESP32_BLE 在文件夹： C:\Users\wangxin\Documents\Arduino\libraries\ESP32_BLE 
exit status 1
invalid new-expression of abstract class type 'MyAdvertisedDeviceCallbacks'

nkolban, 2017-08-13:
Howdy @mxwang101 

About a week ago we changed the API for performing a BLEScan.  When we changed the API, we also changed the sample ... if you have a look at:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp

You will see the latest sample code.  I just verified from my end and the version that is there now looks correct.  Can I ask you to check again the URL above and compare it to the code you are using.  I think you will it different and hopefully see what changes were made.

If this isn't clear or you still have problems, post back.  I'll be here this evening (US time) and tomorrow (Sunday US time) to assist.

mxwang101, 2017-08-13:
Thanks@nkolban&copercini!

BLE_scan work part time, other 3 never work when reset.

1) BLE_scan work part time when reset, the logfile show, first reset, not work, reset again, works.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:708
load:0x40078000,len:0
load:0x40078000,len:12168
entry 0x40078a68
Scanning...
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (Btc_task) 
Register dump:
PC      : 0x4015baef  PS      : 0x00060536  A0      : 0x801577fd  A1      : 0x3ffdcd40  
A2      : 0x00000008  A3      : 0xfffffffd  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000007  A7      : 0x00000001  A8      : 0x00000008  A9      : 0x00000007  
A10     : 0x3ffdce10  A11     : 0x3ffe33eb  A12     : 0x400836c1  A13     : 0x00000010  
A14     : 0x00000007  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x4015baef:0x3ffdcd40 0x401577fa:0x3ffdd110 0x4015153a:0x3ffdd1d0 0x4012fd34:0x3ffdd220 0x4012f3f7:0x3ffdd650 0x40130337:0x3ffdd670 0x400ece60:0x3ffdd700

CPU halted.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:708
load:0x40078000,len:0
load:0x40078000,len:12168
entry 0x40078a68
Scanning...
Advertised Device: Name: , Address: 7d:d7:ea:8f:b1:a6, manufacturer data: 4c0010020a00 
Advertised Device: Name: Cardio.E0DD, Address: ef:81:ff:7b:e0:dd 
Advertised Device: Name: iCT, Address: fb:89:36:a1:57:dc, appearance: 0, manufacturer data: 1ec192a9, serviceUUID: c3888939-adfa-d623-5aa0-ca5bd7ac1a22 
Advertised Device: Name: iCT, Address: e7:2b:5c:ca:40:00, appearance: 0, manufacturer data: 1ec192f7, serviceUUID: c3888939-adfa-d623-5aa0-ca5bd7ac1a22 
E (147364) BT: btc_search_callback  BLE observe complete. Num Resp 4

Devices found: 4
Scan done!

2)BLE_scan
 
Build OK; 

Wrote 1087744 bytes (618384 compressed) at 0x00010000 in 11.3 seconds (effective 767.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 1445.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

Reset no work:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:708
load:0x40078000,len:0
load:0x40078000,len:12168
entry 0x40078a68
1- Download and install an BLE scanner app in your phone
2- Scan for BLE devices in the app
3- Connect to MyESP32
4- Go to CUSTOM CHARACTERISTIC in CUSTOM SERVICE and write something
5- See the magic =)
abort() was called at PC 0x400d7013 on core 0

Backtrace: 0x40088804:0x3ffdd400 0x400888cf:0x3ffdd420 0x400d7013:0x3ffdd440 0x400d8682:0x3ffdd460 0x400d880e:0x3ffdd480 0x400d8b59:0x3ffdd4c0 0x400d8b99:0x3ffdd4f0 0x400d8ba9:0x3ffdd510 0x400d4cea:0x3ffdd530 0x400d3bdd:0x3ffdd550 0x400d2a21:0x3ffdd590 0x400f1481:0x3ffdd5b0 0x400ed688:0x3ffdd5f0

CPU halted.

copercini, 2017-08-13:
@mxwang101 This is a know issue and depends of changes in the compilation of Arduino core for ESP32 https://github.com/espressif/arduino-esp32/issues/567

nkolban, 2017-08-13:
@mxwang101 ... based on your testing (and we thank you for this), we have made an update to the documentation to further explain what you can try to alleviate the issue.  See:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

mxwang101, 2017-08-14:
@nkolban :+1: Thanks, Now All Works!

nkolban, 2017-08-14:
Awesome.  Closing the issue ... re-open as needed.

closed 2017-08-14 by nkolban