nkolban/esp32-snippets #39: WS2812.cpp not working for red colour on ESP32 DEVKIT V1

charlesfitzhenry, 2017-08-19:
@nkolban I am using three 144LED/M strips connected in series which are using the WS2812 LED chips. Whether I am doing something wrong, I don't know, but when I run the following code from my main method The LEDs begin to flash irregularly with different colours. The strangest thing is this only happens when the RED colour is set, all other colours work fine. See the linked videos showing the different scenarios. Is there something that I have done incorrectly for it to be doing this?

```c++
for(int i=0; i<144; i++){
    //Set strip 1 red
    led.setPixel(i, 255,0,0);

    //Set strip 2 green
    led.setPixel(287-i, 0,255,0);

    //Set strip 3 blue
    led.setPixel(288+i, 0,0,255);

    led.show();
}
```

Videos:
[Using Red-Green-Blue](https://drive.google.com/open?id=0B03BwgQorpOma2s2X2JuTzJKLUU) <-- output from code above
[Red only](https://drive.google.com/open?id=0B03BwgQorpOmSXV6U1dJMHdVcnc)
[Green only](https://drive.google.com/open?id=0B03BwgQorpOmLURjb3FpVWVRN1E)
[Blue only](https://drive.google.com/open?id=0B03BwgQorpOmeVE0TnZURXhZTVk)

Any advice as to why this could be happening?

nkolban, 2017-08-19:
The WS2812s are "finicky".  The way they work is by sending VERY fast serial pulses ... down to the nano seconds if I remember correctly.  That means that they are very susceptible to noise.  I am assuming you are using a separate power supply for the pixels to able to handle the current draw.  If I were you, I'd try two things:

1) try using an Arduino board to drive the pixels.  If that works, then we have a better feel that the issue is the ESP32 and not the environment.

2) Try dramatically shortening the wires from the ESP32 to the Data In of the LEDS ... ideally ... as short as is humanly possible.  Also consider "shielding" the pixels from radio or electrical interference.

See if there isn't an alternate WS2812 library for the ESP32 and see if the same results appear there.

charlesfitzhenry, 2017-08-19:
Thanks for that advice, I will give it a try and let know what happens

YvedD, 2017-10-27:
> I will give it a try and let know what happens

any progress on this issue?

Grtz,
Yves

LectureTracking, 2017-10-27:
Sorry I had forgotten to close this issue. I was able to resolve the issue by using another power supply. Apparently, the one I was using was not producing very clean power. It works fine now.

charlesfitzhenry, 2017-10-27:
Sorry I had forgotten to close this issue. I was able to resolve the issue by using another power supply. Apparently, the one I was using was not producing very clean power. It works fine now.

closed 2017-10-27 by charlesfitzhenry