nkolban/esp32-snippets #41: NodeMCU-32s does not work using your arduino library

cupid0426, 2017-08-23:
Serial output info:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:10472
entry 0x40078a28
Starting BLE work!

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:10472
entry 0x40078a28
Starting BLE work!

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:10472
entry 0x40078a28
Starting BLE work!

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:10472
entry 0x40078a28

could you help me find what's wrong, thank you.

nkolban, 2017-08-23:
The notion of brownout detection seems to come from the ability for the ESP32 itself to detect a low voltage situation in its own operation.  How are you powering the device?  If USB, consider trying alternate USB outlets or different cabling.  There also appears to be some ESP32 specific settings that one can change to affect the level of brownout detection.

I have listed some forum posts at the end here that appear to provide some additional thoughts.  They seem show show brownout detection when the WiFi radio is initialized.  This would also be a clue to our puzzle here.  When we enable the BLE subsystem, although obviously not WiFi, they have a common denominator which is radio work.  Radio is a big power draw and if the source power is weak, such a draw could trigger an apparent brownout detection (guess on my part).

Let's see if we can first resolve the issue through alternate power source/location/cable and see where we go from there.

See also:

https://esp32.com/viewtopic.php?f=2&t=2784
https://www.esp32.com/viewtopic.php?t=2462

closed 2017-08-23 by cupid0426

nkolban, 2017-08-23:
Howdy my friend, I saw you closed the issue ... I hope that means something is now working.  Could you comment on what changed to help me help others better?