nkolban/esp32-snippets #42: Building curl

Renata27, 2017-08-25:
Hi Kolban,

With your config files for building curl for ESP IDF - ESP 32, i could get my code working.

But now, after downloading the newest curl from github, I get this issue:
https://github.com/curl/curl/issues/1827

[image]
[image]

Would you please help me understand what I am doing wrong and how to use curl with ESP IDF?

I rely on this for my project to work!

Apreciate your help!!

jay, 2017-08-26:
@nkolban possibly related:
Changes made for the next curl release in October will require premade configuration files to have the values of SIZEOF_LONG and SIZEOF_CURL_OFF_T due to curl/curl@80d9e35. If you have any related problems please let us know at curl/curl#1816.

Renata27, 2017-08-27:
Yes, jay. Thank you! Back to previously curl versions, I was able to build. 
But with the new versions, it isn't working, as you saw above!

Best regards.

nkolban, 2017-08-28:
Howdy guys ... been on vacation for last week ... flying home today.  Normal service/response will resume this evening.

nkolban, 2017-08-28:
Howdy, again ... sorry tor the delay.  I made the changes recommended by @jay and got a new clean compile.  I have checked in the changes and if we can follow the recipe and try it again, lets see how far we get now.

Renata27, 2017-08-29:
Thank you @nkolban ! I am going to give that a try and come back to report!

Renata27, 2017-08-31:
Excellent dear @nkolban !

Thank you for your effort and quick response!
I am very glad to have contacted you, because you have saved my life/file again!

See you!

nkolban, 2017-08-31:
Howdy @Renata27  he he he ... hopefully not your "life" ...
Glad all is well ... come back anytime with bug reports and we'll try and knock them out quickly and for questions, use the ESP32 forum at esp32.com.

Close this issue when happy.

closed 2017-08-31 by Renata27