nkolban/esp32-snippets #47: Example to read from client before notify

rrajkum, 2017-09-04:
Looking for an example, where BLE server (in ESP32) reads from client and then responds with a notify or BLE server reads from client the response received to notify.

nkolban, 2017-09-04:
Howdy my friend ... 
Let's have a discussion on what you are looking for in an example.

I think you are asking for a sample where the ESP32 is being a BLE Server.  See the samples here:

https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests/Arduino

* BLE_notify
* BLE_server
* BLE_write

are all examples of he ESP32 being a BLE server.

The sample called "BLE_write" takes an incoming "set characteristic value" from the BLE client and sets the value.

The sample called "BLE_notify" performs a notification every two seconds to a BLE client.

See also the documentation here:

https://github.com/nkolban/esp32-snippets/tree/master/Documentation

If you could, please study these samples and get back to me with what is not clear.  I am sure we can make the documentation clearer and it is only with help from folks like yourself that we can learn what can be improved.

rrajkum, 2017-09-05:
Thanks a million Neil, 

I got it now. I misunderstood write, but perusing again - it does what I am looking for. I good forward to instantiate the examples!

Cordially,
Raj

closed 2017-09-05 by rrajkum