nkolban/esp32-snippets #49: Compile error building curl - undefined reference to `access'

aivark6, 2017-09-06:
Hello. I am quite new to ESP32 and had a problem while compiling the curl functionality for ESP made by Kolban. Special thanks to Kolban for awesome book, snippets and tutorials.

I am receiving following error:
/home/aivarkoodi/git/gen2/esp-idf/developement/curl/build/curl/libcurl.a(mime.o):(.literal.curl_mime_filedata+0x10): undefined reference to `access'
/home/aivarkoodi/git/gen2/esp-idf/developement/curl/build/curl/libcurl.a(mime.o): In function `curl_mime_filedata':
/home/aivarkoodi/git/gen2/esp-idf/developement/curl/components/curl/lib/mime.c:617: undefined reference to `access'
/home/aivarkoodi/git/gen2/esp-idf/make/project.mk:337: recipe for target '/home/aivarkoodi/git/gen2/esp-idf/developement/curl/build/curl.elf' failed

I think it is some sort of linking error related with **curl_easy_perform()**, because if I remove this function everything compiles nicely.

highwalker, 2017-09-07:
I faced the same issue, but considering that file access will never be used, I commented the specific section and compile went through. Curl has been working since for me. Hope my 2 cents help you!

nkolban, 2017-09-07:
Would it be possible to post your code snippets that relate to invoking curl from your C application?  I want to see what it is that you are asking curl to do for you to see if I see a pattern in the story.

highwalker, 2017-09-11:
Hi Neil - first of all, thanks for your ebook on esp32, it's a great compendium on the esp32 which actually doesn't have as much written on it other than just the manuals.

Well I am using the curl for controlling a WEMO switch, which is basically POSTing XML formatted command. Atleast I checked based on the code coverage that the section I mentioned above is not invoked, so it should be alright ATM. 

Ideally since 'access' has got to do with file pointers/access, it should be able to be cleaned up with preprocessor directive IMO. I haven't had time to sit on this to sort this out yet, but if you think otherwise, I'd like to learn your views.

nkolban, 2017-11-30:
Watching https://github.com/espressif/esp-idf/issues/1085 for access() shim resolution.

nkolban, 2017-11-30:
Believed resolve/worked around here:

https://github.com/nkolban/esp32-snippets/issues/108

closed 2017-11-30 by nkolban