nkolban/esp32-snippets #52: Event handler as a class member function?

jpan127, 2017-09-07:
Hi Kolban,

I was wondering if you could answer another question of mine.

I have been creating a lot of classes but this problem has persisted.  I was wondering if the event_handler can be a member function of my class ExampleTask that inherits from Task (your Task implementation).

Thanks!

nkolban, 2017-09-07:
Howdy my friend.
Can you elaborate on the "event_handler" story?   Generically, an event handler is code that is asynchronously invoked when something happens.  When you say "wondering if the event_handler ..." ... I don't know "which" event handler you are referring to.  There are many event handlers about.  Can you elaborate on what you mean by the "event handler" in the context of your question?

jpan127, 2017-09-08:
Sorry, for example if I had a wifi class and I had my own wifi event_handler.  In my wifi class's function, `Wifi::Initialize()` I call `esp_event_loop_init( (system_event_cb_t)wifi_event_handler, NULL )`.

This `wifi_event_handler` is outside of the Wifi class' scope.

What if I wanted to share private member variables with the `wifi_event_handler`?  I would like to make it a member function.

I have tried simply making it a member of class Wifi but ran into errors whichever way I tried.  One method (suggestion by the compiler error, as in the compiler told me to do this after showing an error) was to do this: 
`esp_event_loop_init( (system_event_cb_t)&Wifi::wifi_event_handler, NULL )`

This allowed me to compile but still threw a long warning.

So I was wondering if you have ever achieved this successfully?

nkolban, 2017-09-08:
Howdy,
The answer is "yes" we can do this.  However the answer also has nothing to do with ESP32 or with the C++ classes supplied for this library.  The answer is pure C++ study I'm afraid.

Imagine you have a function in C that looks like:

```
void registerMyCallback( (void *myCallback)());
```

This says we are declaring a function called "registerMyCallback" which takes as input a function reference.

For example:
```
void myRealCallback() {
   ...
}

...
registerMyCallback(myRealCallback);
```

Great ... but what if the function we want to callback is a method within a class?

This is where things get interesting.  If the method is a "static" then there is one story, if the method is non-static (i.e. a method associated within a class instance) then there is another story.  I'll assume the second one is what you want.

In this case, you would provide a reference to a C function that takes a pointer to a class instance as a parameter.

eg.

```
void registerMyCallback( (void *myCallback)(MyClassType *classInstance), MyClassType *classInstance);
```

Your callback function might then be:
```
void myRealCallback(MyClassType *classInstance) {
   classInstance->myTargetMethod();
}

...
registerMyCallback(myRealCallback, &myClassInstance);
```

There really isn't an easier way for me to write this summary short of writing a book chapter on C++ and function pointers.   I'd suggest getting a good book on C++ and/or googling around:

For example:
http://www.cprogramming.com/tutorial/function-pointers.html
http://www.learncpp.com/cpp-tutorial/78-function-pointers/
http://www.codeguru.com/cpp/cpp/article.php/c17401/C-Tutorial-PointertoMember-Function.htm
http://umich.edu/~eecs381/handouts/Pointers_to_memberfuncs.pdf
etc etc

jpan127, 2017-09-08:
Thank you so much, cleared up some confusion for me.
This is my first time being exposed to callbacks and function pointers, so wasn't sure what I'm dealing with.

closed 2017-09-08 by jpan127