nkolban/esp32-snippets #53: Bluetooth BLEServer createService error

maxcww, 2017-09-07:
Hello, maybe you could help me out? In the following code an error comes:
`#define SERVICE_UUID        "9175cd3a-538b-4689-a588-4ecbc2d9f308"`
`BLEDevice::init("my_device");`
`Serial.println("first");`
`BLEServer *server = new BLEServer();`
`Serial.println("second");`
`BLEService *service = server->createService(BLEUUID(SERVICE_UUID));`
`Serial.println("1");`

The error:
`first`
`second`
`abort() was called at PC 0x400da963 on core 0`

`Backtrace: 0x400887fc:0x3ffdd810 0x400888fb:0x3ffdd830 0x400da963:0x3ffdd850 0x400dbfd2:0x3ffdd870 0x400dc15e:0x3ffdd890 0x400dc4a9:0x3ffdd8d0 0x400dc4e9:0x3ffdd900 0x400dc4f9:0x3ffdd920 0x400d7d5a:0x3ffdd940 0x400d6951:0x3ffdd960 0x400d58d1:0x3ffdd9a0 0x400f4e85:0x3ffdd9c0 0x400f0ffc:0x3ffdda00`

`Rebooting...`

I have no idea what it causes...

nkolban, 2017-09-07:
Ouch ... that's not pretty.  Lets see what we can do.  The first thing we want to try is switching on extra debug at the BLE level.  I'm going to assume you know how to do that, if not, ping back.  If we switch on verbose logging, we will see all the log messages issued by the BLE.  Also ... are you coding in Arduino or in ESP-IDF?

maxcww, 2017-09-08:
Hi, I have switched the debugging on in sdkconfig right? Where do I see the available logging? Further, I am using Arduino IDE.

nkolban, 2017-09-09:
If you are willing to pastebin the sketch as a whole, I'll be happy to take a look.  It looks like only a few lines. If I can reproduce, problem should be as good as solved.  If I can't reproduce, then we can look carefully to see what the distinction between my environment and yours could be.

maxcww, 2017-09-09:
It was a pretty big sketch, so I have created a new sketch where the problem still persists. They are all in the same pastebin https://pastebin.com/D9LxtnzD

maxcww, 2017-09-09:
Problem is, it worked before, but now it doesn't anymore... I strongly suspect it is something with my ESP, but I have cleaned its flash memory already and it still doesn't work.

maxcww, 2017-09-09:
I solved the problem, seems I had to change a variable in the Arduino/hardware/espressif/esp32/tools/sdk/include/config/sdkconfig.h file 
`#define CONFIG_LOG_DEFAULT_LEVEL 1`
to 
`#define CONFIG_LOG_DEFAULT_LEVEL 5`

nkolban, 2017-09-09:
Howdy @maxcww 
Oooh ... thanks for the tip ... I have updated the docs to reflect the instructions for debugging.  If I may ask, when you say "You solved the problem".  My thinking is that you switched on debugging and the problem no longer manifests.  The good news may be that it works, the bad news is that it leaves debugging enabled.  Can you confirm that if you run with debugging enabled, no problem but if you switch off debugging, the problem returns?

Waltibaba, 2017-09-14:
I had a similar issue with it calling abort(), since there seems to be ESP_LOG logging within ISRs. This calls a lock on a recursive mutex deep within esp-idf code, and that's not allowed in the context of an interrupt.
I solved it by removing the log message from my inherited BLEServerCallbacks::onConnect and onDisconnect, and replaced them with printf statements. They were only for debugging purposes anyway.

closed 2018-11-13 by chegewara