nkolban/esp32-snippets #54: Can't build BLE Arduino examples

DonWasabi, 2017-09-09:
Hi,
i tried to build the examples for the Arduino BLE wrapperclasses, but i failed :(
I used this tutorial: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

I am able to build the zip library and i can also import it into the Arduino IDE and open the provided examples. When i press compile, he can't find the header files, when i copy all classes&headers from 
~/sketchbook/libraries/ESP32_BLE/src
to 
~/sketchbook/libraries/ESP32_BLE/

The compiler says a file called sdkconfig is needed.

Seems like i'm missing something,
like the libraries are not stand alone and i need the idf files too or something.
Kind regards

nkolban, 2017-09-09:
@DonWasabi 
Howdy my friend.  Let's see if we can't find the nature of the problem.  I'm on the look out for puzzles like this because this is my first project that involved Arduino libraries and I have every belief that we may not have the instructions just right or the recipe written correctly.

First I want to check that in your Arduino IDE, you are able to build "other" ESP32 apps and flash them.  This will give me a good feel that there isn't something wrong in your Arduino IDE as a whole.  If all that is good, then the next thing I'll ask is that you upload a copy of the ZIP library to here (https://uploadfiles.io/) and paste the link in the a reply to this post.  I'll have a look at it and see if there is something wrong.  Failing all that, we can then use Skype to do a screen share and you can show me live the problem and we can work together to get it fixed.  You can find me on IRC in channel `#ESP32` as user `Kolban`.

DonWasabi, 2017-09-09:
Howdy :)
i do have some good news on my issue:
I deleted and reinstalled the Arduino IDE with this command:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md

and now i'm able to compile it, which made me curious if i can get it to compile in the atom plugin platformio (http://platformio.org/) aswell. I would prefer to use it, because it offers some features like autocomplete, which unfortunately the Arduino IDE lacks. But i can't get it to compile yet.
PS: Wow that was a fast reply.

nkolban, 2017-09-09:
Howdy,
I'm afraid I haven't worked in the platformio space.  The vast majority of my time has so far been spent working in pure ESP-IDF and the Arduino library platform is somewhat new to me.  I'm playing catch-up.  My initial goal for this BLE library was to provide C++ classes for ESP-IDF BLE ... and a colleague then found that they compiled and ran just fine on the ESP32 Arduino platform and did a lot of work packaging them for that.  If there are problems with the ESP32 BLE C++ code then it should manifest in both Arduino and ESP-IDF and of course I'll be keen to resolve them.  However, getting to grips with other Arduino platforms is unfortunately further down in my ESP32 work list.

I try and respond as quickly as I can ... don't confuse speed for quality though ... many of my quick responses are just requests for further clarification or more docs.

DonWasabi, 2017-09-11:
Howdy,
i have some more or less good/bad news (for me ;) ), i am able to compile the example now both in in platformio and Arduino IDE without issues, which is good.
But as soon as i want to mix your libraries with my code, it behaves strange.
Interestingly it builds when i include your libraries before my includes, when i include yours after mine,
it crashes and this error appears: 

https://pastebin.com/utFTm6Mz

Another problem is that when i want to use my SD card wrapper class, the huzzah32 keeps rebooting?
This problem does not seem to be related to the SPI part, because my camera works fine and it uses SPI & I2C.
Here is the wrapper class:
https://pastebin.com/VUcEgrQv
https://pastebin.com/P9gSrZP2

ahachicha, 2017-09-21:
I have the same problem.
I tried to add a characteristic but after the init() method the esp32 keeps rebooting.

DonWasabi, 2017-09-21:
@ahachicha Try to set the debug lvl to 5 and/or the board definition to esp32dev, that may not fix the problem but it can help to get it working somehow :)

nkolban, 2017-11-30:
Does the puzzle still exist?

nkolban, 2017-12-14:
Closing for just now.

closed 2017-12-14 by nkolban