nkolban/esp32-snippets #55: Unable to built Arduino BLE Examples

rrajkum, 2017-09-09:
Hi Niel, 

I am naive and use Ardunio IDE in Windows PC to work with ESP32. So far I managed to get ESP32 work with Ardunio for different functionalities. My next critical stage in project is to use ESP BLE capabilities. 

Many thanks, your BLE Ardunio support for ESP32 comes at right moment. I would like to try out your Arduino BLE examples. 
First I try to follow the instructions at
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md
to create BLE library for cpp_utilities. 
The problem is since I am a windows user above instructions did not work out for me. Then I tried to get some help from a friend with Linux OS and generate the BLE library for me. 

Second I add the BLE library through Ardunio IDE - The library was successfully added.  

Third I try to built one of your examples BLE_server 
[ardunio esp32 ble built error.txt](https://github.com/nkolban/esp32-snippets/files/1290013/ardunio.esp32.ble.built.error.txt)
[Arduino ESP32 BLE Built Error.pdf](https://github.com/nkolban/esp32-snippets/files/1290015/Arduino.ESP32.BLE.Built.Error.pdf)

- that's where I fail.

Any help Niel? 
May be in first place I can't build the library from Linux machine and try to use in in Windows PC? Please can you advise how to make the library for Arduino in Windows env? 

Or it is some other problem, this is the error message (attached) I get in Ardunio IDE

Many thanks to guide me, 
Raj

nkolban, 2017-09-09:
Howdy Raj,
Those error messages would be consistent with trying to build the app with a "board" other than an ESP32 board selected.  Can you look in the Arduino IDE Tools menu and report back on the current setting for "Board"?

Your thinking on building the BLE ZIP file is sound.  What I'll do is create an issue for creating regular "snapshots" of the ZIP file that will be available for "download and go".

rrajkum, 2017-09-10:
Hi Neil, 

I am using the board ESP32 Dev Module. I will try again to uninstall and reinstall Arduino. I shall get back. 

It is really great that you will try to provide a downlodable library link in coming days. It is going to very helpful for me and others. 

Many thanks,
Raj

rrajkum, 2017-09-10:
Hi Neil, 

Still no luck. I get the same built errors. I had reinstalled the arduino IDE and updated the ESP32 repository. 

I attach the snapshot of Dev. Broad selected in Arduino as well the built error. 

Out of curiosity since I see the error points to esp_log.h, I tried #include "esp_log.h to example code and also add esp_log_level_set("*", ESP_LOG_VERBOSE); in setup(). I also tried to change CONFIG_LOG_DEFAULT_LEVEL=1 to 5 in sdkconfig to see if that will help. Still the result was the same compile error. 

Am I missing something? Do you think I try something different?

Many thanks, 
Raj
[Arduino ESP32 Dev Board.pdf](https://github.com/nkolban/esp32-snippets/files/1290425/Arduino.ESP32.Dev.Board.pdf)
[Arduino ESP32 BLE Built Error 0910.pdf](https://github.com/nkolban/esp32-snippets/files/1290426/Arduino.ESP32.BLE.Built.Error.0910.pdf)

nkolban, 2017-09-10:
Howdy Raj,
Just in case there is something not right in the ESP32_BLE.zip file, I have uploaded the version I am using.  This can now be found here:

https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/Arduino

We also have some notes on installing a new version found in this doc:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

Let's try using the ZIP file from this location.

Thanks for your Arduino.ESP32.Dev.Board.pdf file.  That shows exactly what I was looking for and that looks fine.

Can I also assume that if you try and compile and flash other ESP32 applications from your Arduino IDE that they compile and run cleanly?

rrajkum, 2017-09-10:
Hi Neil, 

So many thanks, I will try out the library you provided and get back to you. Yes, that is right I am able to compile and run other ESP applications on my Arduino IDE. 

Many thanks,
Raj

nkolban, 2017-09-10:
If you need some live support later today we can arrange something if you desire.  I am in US central time.  Skype will be the best way to reach me at "neil.kolban".  My availability is about 20% today ... apologies for that.

rrajkum, 2017-09-10:
Hi Neil, 

That is very kind you offer to help with live support. I am based in Denmark. I am just about the try the new library. I would not like to distrub your weekend. Please allow to grab your help if I continue to face issues. Let me get back.

Many thanks,
Raj
Skype rraj121

rrajkum, 2017-09-10:
Hi Neil, 

I tried with new library. The same issue persists. May be I need to grab your offer to help via Skype. I feel uncomfortable to trouble you today. How about tomorrow 11 Sept any time slot that suits you between 09:00 - 13:00 your time? It is a time difference of minus 7 hrs between US central and Denmark. 

I hope to catch-up on Skype. 

Many thanks for all your support,
Raj

ADeeds, 2017-09-12:
I had a similar issue. I'm using Neil's library via PlatformIO and not the Arduino IDE, so this may or may not help you. In my case I was linking against an old version of the ESP SDK (I think 0.9?) which didn't have some of those Bluetooth constants defined.

nkolban, 2017-09-12:
Howdy @ADeeds 
I had the opportunity to chat with mr @rrajkum this evening and that was exactly the issue (so far).  The problem was that the ESP32 Arduino package installed into the Arduino IDE was indeed back level (just as you thought).  Mr rrajkum is going to download the latest from Github, re-install and re-test.

rrajkum, 2017-09-12:
Many thanks Neil, with your support things are working good after downloading the latest repo from Github. 

There seems to be link issue with ESP32_BLE_Arduino-master.zip. We can close this issue and track separate. 

Now I confirm that examples compiles, links and executes good!

closed 2017-09-12 by rrajkum