nkolban/esp32-snippets #56: Distribute the Arduino BLE ZIP file

nkolban, 2017-09-09:
As of today, we are making users jump through hoops to build the Arduino BLE ZIP file.  Let us examine the state of Arduino builds and see how can we correctly distribute the Arduino BLE ZIP.  Ideally, we could leverage the package management tools ... lets see if we can.

nkolban, 2017-09-10:
While we think about the correct distribution of the ZIP, an interim story is now available with the ZIP available here:

https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/Arduino

with replacement installation instructions here:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

nkolban, 2017-09-10:
I found the following link which seems to include all the requirements and recipes necessary to distribute an Arduino library.

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

nkolban, 2017-09-10:
First attempt at making the library available through the library manager in the IDE:

https://github.com/nkolban/ESP32_BLE_Arduino

rrajkum, 2017-09-12:
Hi Neil, 

Instead of raising new issue. I like to state that new library from library manager fails at the link stage after successfully compiling. 

Thanks,
Raj

nkolban, 2017-09-13:
Apparently I messed up what it took to create a Managed Library.

What I did was the following.

I read this web page:

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

And read about a flag called `dot_a_linkage`.  My assumption was that if I added this flag and set it to `true` then the Arduino BLE code would be compiled and placed in an archive file (a `.a` file) and would produce faster/easier linking at the linking step.  I then enabled this flag and pushed it to the Github repository.  Since the Arduino Manage Libraries system scans the repositories every hour for changes, I then wandered away thinking I would test it in a couple of hours after the latest refresh.  I then promptly forgot to retest.  Now we find that my addition of `dot_a_linkage` didn't work and produces a linker error.  I have now removed the `dot_a_linkage` line from the `library.properties` file.  This has been pushed to both the Arduino library manager "pickup" location as well as to the master repository.  Hopefully normality will be restored from version 0.4.2 onwards.

rrajkum, 2017-09-14:
Super Neil, 

I will try them. In meanwhile your examples are working good with the library you had shared. 

Many thanks,
Raj

nkolban, 2017-09-14:
I just had the opportunity to retest and all appears to be working now.  What this means is that from a virgin Arduino IDE with the ESP32 Arduino support added, we can now install the Library called "ESP32 BLE Arduino".

We'll wait a few more weeks to ensure it is working but if there are no reported issues, we'll consider this item completed.

rrajkum, 2017-09-16:
Many thanks Neil, I am trying this out.

elsabz, 2017-09-17:
Hi Kolban, see a BLE Library for Arduino is great news for me, thankyou!
I tried installing it on my Windows 10 system. Since a few months ago I installed the arduino core for ESP32 and this works fine, I can complete the projects and upload them on ESP32. Unfortunately I get an error while compiling sample projects in the ESP32_BLE.zip library, BLE_Client and Ble_Server, Arduino says: Error while compiling for the ESP32 Dev Module board. What can I do? Do I need to enable BLE with make.config? How do I open the make.config in Arduino on Windows 10?

nkolban, 2017-09-17:
Good thinking ... I have only ever tested on Linux ... building myself an Arduino environment for ESP32 on Windows 10 now and will test.

nkolban, 2017-09-17:
I installed from scratch the Arduino IDE and then followed the instructions here for installing the ESP32 Arduino environment on top of that:

https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

I switched my board to "ESP32 Dev Module".

I then installed the Arduino BLE ESP32 library from the Library manager.

Finally I went to the BLE ESP32 samples, select BLE client and it compiled cleanly.

So all appears "good" on my end.  Looks like there is a puzzle to work out.  Make sure you have the latest Arduino IDE and the latest installation of Arduino ESP32 libraries and the latest ESP32 BLE library (0.4.2 as of today).  If the problem persists, capture any and all logs that you see and post them as patebin files here.

elsabz, 2017-09-17:
YES (y) (y) (y)
i update the Arduino IDE (my version war 1.8.2 and now is 1.8.4) and i update the latest ESP32 libraries.
Now i can compile ESP32_BLE without error... In the next Day i test the example on the ESP32 board, Thank you very much (y)

nkolban, 2017-09-21:
believe completed.  Reopen as needed.

closed 2017-09-21 by nkolban

pdxalz, 2017-09-23:
Hi Neil,

I am unable to build the BLE_Client example.
I have tried both Window 10 and Ubuntu.  Everything was freshly installed today.
I have Arduino version 1.8.4,  not sure how the find the version of the espressif stuff.
On windows I used the BLE lib from the manager, and on Ubuntu I built the zip file and did an "Add .zip library".  I get the exact same result:

home/alan/Arduino/libraries/ESP32_BLE/src/BLEClient.cpp: In member function 'void BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
/home/alan/Arduino/libraries/ESP32_BLE/src/BLEClient.cpp:126:78: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'
    BLERemoteService *pBLERemoteService = getService(BLEUUID(evtParam->notify.srvc_id.id.uuid));
                                                                              ^
In file included from /home/alan/Arduino/libraries/ESP32_BLE/src/BLEClient.cpp:9:0:
/home/alan/Arduino/libraries/ESP32_BLE/src/BLEClient.cpp:128:104: error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no member named 'srvc_id'
     ESP_LOGE(LOG_TAG, "Could not find service with UUID %s for notification", BLEUUID(evtParam->notify.srvc_id.id.uuid).toString().c_str());
                                                                                                        ^

It looks like a mismatch between the espressif code and the BLE lib because it looks like the elements in the structure have been moved around.  I'm assuming I installed something wrong because it looks like it was working just a few days ago.

The complete list of build errors can be found here: https://pastebin.com/hQ7JDxXp

ciotlosm, 2017-09-23:
@pdxalz see https://github.com/nkolban/esp32-snippets/issues/82

ThorvaldAagaard, 2017-10-03:
I have been reading this and especially #82, but it is unclear to me what the current state is.

Is it currently possible to use Arduino BLE ESP32 functionality?

Versions doesn't really matter as we are just prototyping at the moment

nkolban, 2017-10-03:
Howdy my friend,
Unfortunately Arduino support for the BLE C++ classes is suspended ... it just won't work.  However, there is good news.

To summarize:
- Arduino-ESP32 libraries are built as part of the Arduino-ESP32 project
- The Arduino-ESP32 project is built on top of ESP-IDF (the native SDK for the ESP32)
- The latest release of the ESP-IDF broke the ability to run C++ classes (see: https://github.com/espressif/esp-idf/issues/1032)
- The current Arduino-ESP32 project refreshed its distribution to be built on top of the the latest ESP-IDF

The result is that the latest Arduino-ESP32 uses the latest ESP-IDF which uses the latest ESP32 toolchain which contains a runtime bug that prevents some C++ classes that should work from working.  These classes are used by the BLE C++ classes in this project.

The good news is that Espressif have found the bug referenced in the issue above ... and they have shipped me a test to make sure it works (it does).  However we need to wait for Espressif to release a fix in their Github master and then we need to wait for the Arduino-ESP32 project to refresh itself with the new ESP-IDF.

The things should be back to normal.

idtres, 2017-10-03:
Hey nkolban i just sent you an email regarding this, sorry to bother you.
Can we estimate a timeframe for the next release ? I{m looking foward for a ibeacon scanner using esp32

Thanks

nkolban, 2017-10-04:
Howdy,
Execution on the latest Arduino IDE is broken because of a problem with the latest toolchain and C++ standard libraries.

This can be read about here:

https://github.com/espressif/esp-idf/issues/1032

In summary, the ESP-IDF introduced two major changes in their pre-release 3.0 simultaneously.   One was a change to the exposed APIs of ESP-IDF BLE APIs that the C++ classes in this project were using.  That was resolved in a few days.  The second was a change to the underlying toolchain which broke C++ standard library usage.  That has taken longer to fix but I have been given a test patch by Espressif which (in my environment) works just fine.  However we need to wait for that patch to be incorporated in ESP-IDF before we can call that issue closed.   Combine the above (which are strictly ESP-IDF issues) with the story that the latest Arduino-ESP32 based itself on the very latest ESP-IDF (changes and breakages and all) ... then we have a chain of problems.  First we need ESP-IDF puzzles resolved and then we need a new release of the Arduino-ESP32 to incorporate those changes when available.  Only then will we be in a stable state to release a new Arduino-BLE library.

As to when all this might happen.  I'm going to assume that the ESP-IDF toolchain changes are at least two weeks away because of holiday season in China and then we will have some latency after that before Arduino-ESP32 picks up the changes.  I can pretty much assure that 24 hours after Arduino-ESP32 being available a new BLE C++ library will be made available.

mahdikan, 2017-12-23:
hello to mr nkolban
i downloaded the BLEscan zip file and added to arduino v1.8.2 but it dosen't work end there are som errors
please help me to run the blescan project
thanks a lot
Mahdi

idtres, 2017-12-23:
Hey I just tested a few days ago, it works just fine
Download latest arduino, and re download ESP32 as the instruccions of the
github oficial page says
then run the exe con the tool folder

the ble examples will be available, try rthe ble scan ( change the 30
seconds to 5 seconds ) and it works just fine

mahdikan, 2017-12-27:
hello dear friends i want to search bluetooth mac addresses with esp32  using arduino IDE but i haven't been do that until now if there is a way, please send me the program.
best regards 
Mahdi

idtres, 2017-12-27:
This worked for me

/*
   Based on Neil Kolban example for IDF:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 5; //In seconds

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n",
advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);

}

void loop() {

   Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get
results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  delay(100);
}

mahdikan, 2017-12-27:
hello idtres 
thanks for your response, i downloaded this code to esp32 and turned on my android phone bluetooth, but the esp32 couldn't decet my phone mac, and every time it prints this:

Devices found: 1
Scan done!
Scanning...
Advertised Device: Name: , Address: 1b:d3:d6:33:33:58, manufacturer data: 060001092000dc23418796fd4fce39689e1e6601fd3c9e6a69084eecf2 
E (965235) BT: btc_search_callback  BLE observe complete. Num Resp 3

please tell me what is that mean
thanks

nkolban, 2017-12-27:
mahdikan ... might I suggest that you create a new issue to cover your question ... your posting starting from here:

> hello dear friends i want to search bluetooth mac addresses with esp32 using arduino IDE but i haven't been do that until now if there is a way, please send me the program.
> best regards
> Mahdi

Seem to cover an issue that is not the same as distributing a BLE ZIP file Arduino.

idtres, 2017-12-27:
The code works, its looking for BLE devices, my ble keychan works, and its
finding 1 device
now we depend on your phone, the settings of your phone, the bt app that
you are using and so,,,
You need a deep understanding of the ble protocol, I sugest to check out
this 2 videos

https://www.youtube.com/watch?v=2mePPqiocUE
https://www.youtube.com/watch?v=osneajf7Xkg