nkolban/esp32-snippets #57: BLE server idf example

DonWasabi, 2017-09-13:
Howdy @nkolban ,

im trying to get your BLE c++ wrapper classes to run with idf, do you have something like a helloworld server example? My approach is kinda typed from your video: https://www.youtube.com/watch?v=2_vlF_02VXk&t=324s

My code:
https://pastebin.com/XFCQKFQf

And it compiles well, but i can't find the esp ble device with my phone. After watching your video again and looking inside the BLEDevice.h it seems like this line 
BLEDevice::init("SERVERNAME");
should do the trick, but when i add this line, i get lots of compile errors:
https://pastebin.com/Mj2U6cwg

nkolban, 2017-09-13:
Howdy my friend,
By any chance have you had the opportunity to look at this doc.

https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf

My goal is to provide good documentation on how to use these classes and if something doesn't work or isn't clear, I want to hear about that so that I can fix it. 

There are also a growing number of samples found here:

https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests

If after looking at these we aren't making progress, realize that I'm fully here for you to assist and we'll get there in the end (I hope).

nkolban, 2017-10-04:
Howdy closing for just now.  Re-open as needed.

closed 2017-10-04 by nkolban