nkolban/esp32-snippets #58: ESP32_BLE.a: No such file or directory

WALLTECH, 2017-09-13:
Trying to compile any Arduino BLE examples results in this error: xtensa-esp32-elf-gcc: error: libraries\ESP32_BLE\ESP32_BLE.a: No such file or directory.

nkolban, 2017-09-13:
Recreated ... over the weekend, I worked on getting the library added to the master Library manager and broke something.  Checking/fixing now.

WALLTECH, 2017-09-13:
Alright, thanks.

nkolban, 2017-09-13:
I think it has been corrected. See #56 for details.

nkolban, 2017-09-14:
@WALLTECH I am hoping that the issue has now been resolved.  We should now be able to install the library from the Arduino IDE Library Manager.

WALLTECH, 2017-09-14:
Yep, no issues now. Thanks!

nkolban, 2017-09-14:
Thanks @WALLTECH  ... am closing the issue ... but don't hesitate to re-open if this problem occurs again or open a new issue for any new problems.  I'm tracking what I have to do based on the issues.

closed 2017-09-14 by nkolban