nkolban/esp32-snippets #60: Service list empty (on nRF Connect) when using SampleServer or SampleNotify examples

dadosch, 2017-09-14:
When I flash the SampleServer example as is, with Bluetooth enabled in menuconfig, and then connect using the nRF Connect app, the custom service is not shown in the device's tab, but its id in the "scanner" tab.

It only shows Generic Attribute 0x1801 and Generic Access 0x1800 in the device's tab.
Debug log doesn't show any error at all.

nkolban, 2017-09-14:
Howdy my friend,
Can you have a review of this open issue:

https://github.com/espressif/esp-idf/issues/857

Can you describe your environment?  Are you building with native ESP-IDF or Arduino?  If native ESP-IDF, what version of the ESP-IDF?  If using the master, can you try using 2.1 as described in the referenced issue above?

dadosch, 2017-09-15:
Ahh, ok, thanks.

My environment is the latest esp-idf (master), native ESP-IDF.

Yes, if i use version2.1-branch, it works fine!

Thank you 😄

closed 2017-09-15 by dadosch