nkolban/esp32-snippets #61: RFID library doesn't work

CarloMara, 2017-09-15:
Hi There,

I'm trying to get the RFID module to work with the port that you have written (MFRC522). The module is connected to the ESP32 according to the default configuration (MOSI13, MISO 12, CLK 14 CS 15) and the example code reports an unknown firmware version [0x11].

Still, the module didn't seem to respond to any tag that I have on hand, not even to my public transport pass, that I know for sure does work. I tried with another module, but still, no luck. I then moved to see if the SPI communication was properly happening, and according to my trusted rigol it is.

I thus decided to go with printf-debugging, and it showed me a wired behavior. When a tag is paced close to the module, it gets detected [to say that we manage to escape the _PICC_IsNewCardPresent()_ jail ] but the program gets stuck at detecting the card uid [aka at _PICC_ReadCardSerial_ ]. In case you are wondering I'm referring to _DumpInfo.cpp_ 

On a side note, I tried to scan the tags with my phone and they definitely work.

Hope you can give me a clue,
Carlo

nkolban, 2017-09-15:
Howdy Carlo,
I'm probably not going to be able to help that much.  I took the public library for the MFRC522 and ported it exactly to the ESP32 ... which basically meant that I replaced only the logic that was hardware related and changed those to ESP32 SPI function.  I have no idea how the library itself works :-(

Do you have a link to the MFRC522 module you are using?

CarloMara, 2017-09-16:
Hi Nick,

I kinda imagined that. The module that I'm using is a bog standard one like [this](https://home.ctw.utwente.nl/slootenvanf/wp-content/uploads/2016/05/RFID-RC522-pinout.png).

Anyway, I cloned the original repo and found out that they initialize the class instance with
` MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 instance `
while on your repo we simply do 
` MFRC522 mfrc522;  // Create MFRC522 instance `
I haven't had any time to look at the code yet, but if you could explain me how to set the reset pin in the port of the library it would be great. 

Anyway I'm going to try and use the modules with an Arduino, so we can start removing unknowns form the equation.

Thanks again for your time,
Carlo

P.S. Can you suggest me a debugger that can be used with the ESP? I tried a segger clone, but it's deadly slow.

nkolban, 2017-09-19:
I tend to work mostly in ESP-IDF land rather than Arduino ... however, there is a JTAG debugger ability and it seems to work.  That said, I mostly do "printf" statements.

For JTAG debugging, I went for the easiest approach which was to buy the Espressif "WROVER Dev Kit" which has JTAG built in and ready to go.

CarloMara, 2017-09-20:
Thanks for the tip, I think I'll do the same thing.

As for the library, I tried with an Arduino and it works flawlessly. Go figure.
I don't know exactly how I'm going to proceed. One way could be trying  to rebase your port on the last version of the Arduino library. Another way would be to step through the code with both the Arduino and the ESP in order to see where it hangs.

If you have any better ideas, drop a line,

Carlo

CarloMara, 2018-02-28:
No, unfortunately I coudn't get it to work.

cheers,
Carlo

CarloMara, 2018-11-09:
> Same problem for me unfortunately...
> I am not so much experimented and it is only a supposition, but I guess
> there might be a problem at reading / writing registers (so before speaking
> about scanning any card), because there is already a problem at
> PCD_DumpVersionToSerial(). Bad values are returned too when calling
> PCD_GetAntennaGain() for example.
> For the moment I have no idea about how to solve this, but at least it
> reduces possibilities.

Hello everybody,

I've come to the conclusion that it's a problem of the chip as Paul
suggested. I've tried to implement my library, straight from the datasheets
and the results are similar, at some point it will return an unexpected
value for no reason.
I don't have the code nor the results right now, as they are on my computer
at home. If someone is interested I might fish them out but frankly it's
just poor code and excels spreadsheets.

I don't know what is causing this behavior, but I highly suspect poor
quality chips. Supply chain management is a mess these days.

Cheers all,
Carlo

PaulBreugnot, 2018-11-09:
Hello,

I'm pretty sure that **this can't be an hardware problem**, because I tried the module on the ESP32 with the same connections, but compiling the code from the Arduino IDE (so, using the **original** MFRC522 library), and it works like a charm.

So I focused on the read / write registers functions and I think I discovered quite an interesting thing.
Let's assume that the write operation is performed correctly.
According to the read SPI protocol describe in the datasheet (https://www.nxp.com/docs/en/data-sheet/MFRC522.pdf) we could suppose that for a single byte we have : 
MOSI : adress 0, 0 (stop reading)
MISO : X            , value 0
So, currently, to read a register, we do : 
```
byte MFRC522::PCD_ReadRegister(PCD_Register reg) {
	uint8_t data[2];
	data[0] = reg | 0x80;
	data[1] = 0;
	m_spi.transfer(data, 2);
	return data[1];
} // End PCD_ReadRegister()
```
But I made some prints during those operations, and this is not what we get, but values doesn't seem to be random (see my attached screenshot).
```
byte MFRC522::PCD_ReadRegister(PCD_Register reg) {
	uint8_t data[2];
	data[0] = reg | 0x80;
	data[1] = 0;

	ESP_LOGI("REG", "Data[0] (complete address transfered) : %.2x", data[0]);

	m_spi.transfer(data, 2);

	ESP_LOGI("REG", "Data[0] response from reg %.2x : %.2x", reg >> 1, data[0]);
	ESP_LOGI("REG", "Final 0 response from reg %.2x : %.2x (returned value)", reg >> 1, data[1]);
	return data[1];
} // End PCD_ReadRegister()
```
[image]
It seems that in response of the final 0, we actually have the address previously transferred (this bad value is returned!) and our correct **data seems to be in data[0] !**

I don't know from where this behavior come from (different implementations of the SPI protocol in Arduino or in ESP-IDF?).

Anyway,

 I still have hope to solve this and getting this library working with the ESP-IDF!

PaulBreugnot, 2018-11-11:
It finally works!

I just added the SPI slave select / release around each SPI transfer (and the definition of `_chipSelectPin` as an output), and now it seems to work! I created a pull request to include the corrections : 
https://github.com/nkolban/esp32-snippets/pull/713

Notice that for now, we need to connect the module to the HSPI (see HSPI / VSPI tables there : https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#gpio-matrix-and-iomux). Currently, to use VSPI, I modify SPI.h with the good gpios. I think it would be a good idea to be able to select HSPI / VSPI directly at `PCD_Init()` call, maybe I'll make a pull request later.

hansmbakker, 2018-12-04:
@CarloMara I think this issue can be closed since the problem was in the chip? Could you please close this to get the issue list clean?

@chegewara

closed 2018-12-05 by CarloMara