nkolban/esp32-snippets #62: BLE Sample notify is not working

chegewara, 2017-09-16:
Hi Neil,
i have two issues with ble sample notify. First one is easy to solve. In line https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleNotify.cpp#L95 is created new BLE2902 descriptor, but by default notification i turned off in it (easy to fix).
But after fixing it there is another issue: 
`E (3358) BT: btc_gatts_arg_deep_copy 8 no mem` 
and i cant find solution. And i think so at the end service and/or characteristic does not start.

nkolban, 2017-09-16:
Howdy ... it is my understanding that the flag which says that notifications are enabled should NOT be initially set.   Instead, a client that wants notifications should make a request to set the flag on.  The server then looks at the value of the flag before actually sending a notification and then uses this as the indication of whether or not to send the notification.  Thus the server does not send notifications unless a client has actively expressed an interest in receiving them.

Have you increased the amount of memory for the Blue Tooth stack in the makemenu config settings.

See the following:

https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf

chegewara, 2017-09-16:
Hi,
just a different approach. In my case notify is ready when esp32 starts and is waiting ble client to connect, then starts to sent notifications.

Yes, i did increase BLE stack to 10000 but still same error. Most likely its esp-idf error, but i dont know how to debug it.

nkolban, 2017-09-17:
Do you have a sample application that demonstrates the issue?  Have you been able to switch on BLE C++ library debugging/trace to examine what is going on at the API level?

chegewara, 2017-09-17:
Basically its sample code with 1 change: 
`    BLE2902* p2902Descriptor = new BLE2902();
    p2902Descriptor->setNotifications(true); `
in line 95
`	pCharacteristic->addDescriptor(p2902Descriptor);
`

chegewara, 2017-09-17:
The weirdest thing is when i revoke esp-idf to commit 27574a3 then i cant compile the code cause for some reason files in components directory are not included and  i got this message:
`F:/esp32/programs/workspace/bletest/main/SampleNotify.cpp:26:23: fatal error: BLEDevice.h: No such file or directory
compilation terminated.`
but with latest esp-idf i can compile and flash but with issue described above.

nkolban, 2017-09-17:
What is esp-idf commit 27574a3?   Do you have a link?  What is your thinking?  Are you trying to run with release of ESP-IDF v2.1?

For the compilation issue, try deleting the `F:/esp32/programs/workspace/bletest/build` directory and then try re-building.   The BLEDevice.h header should be found in `F:/esp32/programs/workspace/bletest/cpp_utils`

chegewara, 2017-09-17:
Ive found this post of yours https://github.com/espressif/esp-idf/issues/857, and this is mentioned commit https://github.com/espressif/esp-idf/commit/27574a31e7c41f6e22e10e9b9d7713c6d49f795d, its esp-idf v2.1 release.

When i change esp-idf to new commit with fetch or checkout i always do make clean and then recompile code. This dont help this time.

nkolban, 2017-09-17:
If desired, we can always do live chat and/or screen shares as needed.  You can always find me when I am around on IRC in channel `#ESP32`.

chegewara, 2017-09-17:
I will make it working somehow, cause your library is great, nice and clean to read and to work with.
Thank you for all your hard work.

closed 2017-09-17 by chegewara