nkolban/esp32-snippets #63: Disconnects when using notify and write together

rrajkum, 2017-09-16:
Hi Neil, 

Your examples are working good. In my use case I would like to use ESP32 as BLE server and nRF Connect (as client). I execute examples BLE_server, BLE_write and BLE_notify. All work perfectly fine individually. Using BLE_notify I am able to receive the notifications from ESP32 into nRF connect. Similarly with BLE_write if something is written in nRF connect it is received in the ESP32 (as seen from serial monitor). 

Next I would like to use notify and write together. The idea is when ESP32 (BLE server) notifies the client, it sends back something. This is where I fail.  First I try to modify the notify example so that I can call BLECharacteristic *pCharacteristic->getValue() to get the value from client. Second I attempt to merge BLE_write and BLE_notify sketches. In both attempts the connection disconnects every few minutes, I am not able to read or write from the nRF connect app. May be I am goofing up something... 

An example to using both notify and write would really help. Any chance? 

Many thanks,
Raj

nkolban, 2017-09-17:
What I'd suggest is to try and create the design yourself, code up as simple an example as possible and try it.  If you get stuck, then you can post questions to the forums or issues here in Github.  I'm here to assist and answer questions.

rrajkum, 2017-09-17:
Hi Neil,

I catch the mistake I was making in my attempts - was using same TX and RX uuid. May be I can post the exampke after throughly testing. 

Many thanks for you library, it makes working with BLE easy.

closed 2017-09-17 by rrajkum