nkolban/esp32-snippets #64: Webserver with Websocket: how to start?

squonk11, 2017-09-17:
Dear Neil,
thank You for Your great work on these ESP32 libraries - I highly appreciate it.
Currently I am planning to setup a webserver for an embedded device using the ESP32. Here I need to exchange fastly some data with the embedded device and the webbrowser. So, I think a webserver with websocket (instead of REST) could be a good choice. Checking Your library I see two different approaches: one using mongoose.c and the other one using Your WebSocket.cpp code. Now I ask myself which approach to choose. My questions are:
1. Mongoose seems to have a powerful featureset and is easy to use. So, why not choose mongoose? Are there issues concerning the license or speed of encryption in case of using https and wss (not using the hw accelerator of the ESP32)?
1. What is the advantage of using Your WebSocket.cpp code?

Best regards

   Lothar

nkolban, 2017-09-17:
Howdy Lothar,
I'm a big fan of C++ these days and think it makes a great environment for ESP32 programming.  To that end I'm looking at all the major functions and seeing if they can be encapsulated as C++ classes.  Web server function was an obvious one given the nature of the ESP32.  To that end we wrote a set of C++ classes that attempted to encapsulate the function of the 3rd party Mongoose web server library.  That worked great ... no issues.  Mongoose is super mature and works as advertized.

However my thinking went as follows:

1. The mongoose library is licensed material ... happily, it is open source and 100% available for re-distribution as long as the complete source of the surrounding application is also open source.  For me and my purposes this is fine.  However I envisioned some scenarios where users of the C++ library I am tinkering with might inadvertently include my free for use library and pull in mbedtls for commercial purposes and then there would be a mess.

2. The mongoose library (of course) uses sockets.  However in my play, I wanted to create a C++ class that encapsulates sockets ... which would then an app using the mongoose wrapped HttpServer would now be running through native sockets and my wrapped sockets.  Not a big deal ... but something I was hoping to avoid.

3. ... and this is the biggie.   In my day job I recently had to study the HTTP protocol is some great depth.  Most of us in IT space have done this at some point or another and the result was that I said "it might be fun to see what would be involved in writing an HTTP server"

4. I'm tinkering with an ESP32 project that I call "ESP32 Explorer" .. see:

https://github.com/nkolban/ESP32_Explorer

This is a library which can be linked with an ESP32 programmers on apps that provides visibility at runtime into the operation of that app.  It behaves as an HTTP Server to serve up web pages that show the result (see screen shots in the repository above).  Because I wanted to have this embedable into user's own ESP32 apps, I got nervous about basing that part on Mongoose ... so now we have an extremely simple Web Server that can be included that I know 100% how it works and it has no polling loops.   I also needed WebSocket support so that got added too.

I'm not trying to "push" either the Mongoose code for your use or my own new code for your use ... I'll happily support the C++ libraries for either.  When I write any code, I try and push it up there as quickly as I can in the hope that it may be of use to someone/somewhere.  No magic agenda on my part ... all this code is free and the only smile that comes to my face is when someone finds some benefit from it.

chegewara, 2017-09-17:
Hi,
ive found nice open source project which is simillar to pubnub and can be alternative for mongoose. Its socketcluster, https://socketcluster.io/#!/. And [here ](https://github.com/sacOO7/socketcluster-client-C) you can find c client. Its very nice framework which have standalone server. Server can be setup on any node server (at home on R-Pi or on heroku) and many clients. 

@nkolban If you will have some time and you see it as good tool to work with esp32 then maybe you could make a client for it.

squonk11, 2017-09-18:
I have still one question on this: In case of using an encrypted connection (https and wss), do mongoose and Neil's solution use the HW accelerators of the ESP32? I think if one of the solutions does not use it, it might make a big difference in performance, right?

nkolban, 2017-09-18:
To the best of my knowledge, the Mongoose logic uses the mbedTLS library and the library I am tinkering with also uses mbedTLS.  It is my understanding that leveraging the mbedTLS supplied with ESP-IDF does indeeded leverage  ESP32 native capabilities.  I just got SSL/TLS working on the HttpServer classes last night (not yet committed to the repository).

squonk11, 2017-12-17:
Now I am trying to use the Websocket class in combination with the HttpServer class. Unfortunately I do not understnd how to use it. Is there somewhere an example or at least a skeleton showing how it is going to be used?

nkolban, 2017-12-17:
Howdy ... lets start at the beginning ... have you had a chance to look at this documentation?

https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf

Also ... I'm now available live on IRC in channel `#ESP32` as user `kolban`

squonk11, 2017-12-17:
currently I am able to activate a handler:
`static void wsHandler(HttpRequest* pRequest, HttpResponse* pResponse) {
	static char tag[] = "wsHandler";
	ESP_LOGD(tag, "Request-Path: %s", pRequest->getPath().c_str());
	WebSocket *pWebSocket = pRequest->getWebSocket();
}`
but what to do in the handler?

nkolban, 2017-12-17:
Once you has a reference to a WebSocket object (pWebSocket in your example) ... you can now send data to the partner by calling `pWebSocket->send(data)`.

If you want to receive data from the partner, then once you have the WebSocket object, you can register a handler for WebSockets using:

```
MyWebSocketHandler myWebSocketHandler;
pWebSocket->setHandler(myWebSocketHandler);
```

We do this because data coming from the web socket partner can arrive at any time (asynchronously).  The WebSocketHandler has callback methods for:

* onClose() - You are told when the partner closes the web socket
* onError() - You are told if there is an error in the web socket
* onMessage() - You are told when there is a new message from the partner

squonk11, 2017-12-17:
if I do this:
`static void wsHandler(HttpRequest* pRequest, HttpResponse* pResponse) {
	static char tag[] = "wsHandler";
	ESP_LOGD(tag, "Request-Path: %s", pRequest->getPath().c_str());
	WebSocket *pWebSocket = pRequest->getWebSocket();
	pWebSocket->send("Hello", WebSocket::SEND_TYPE_TEXT);
}`
then I get this;
`D (27102) wsHandler: Request-Path: /api/v1/ws
D (27107) WebSocket: >> send: Length: 5
D (27110) Socket: send: Raw binary of lengthh: 2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.`

nkolban, 2017-12-17:
When we get an exception, we should also be able to get a stack trace to determine where within the code the exception was thrown.  I tend to use the flags that say that when an exception occurs, halt the processor and place me in GDB.  Then I can run the GDB command "bt" (back trace) and determine exactly where the problem happened.

squonk11, 2017-12-17:
The log also has:
> Register dump:
PC      : 0x4017a41f  PS      : 0x00060630  A0      : 0x80145e3c  A1      : 0x3ffd
82a0
0x4017a41f: Socket::getSSL() const at E:/msys32/opt/devel/HttpServer/components/cp
p_utils/Socket.cpp:244
A2      : 0x00000008  A3      : 0x3ffb5864  A4      : 0x3f4118d4  A5      : 0x0000
0104
A6       : 0x00000001  A7      : 0x00000004  A8      : 0x80082af8  A9      : 0x3ff
d8240
A10     : 0x00000034  A11     : 0x3ffd55a8  A12     : 0x3ffd8280  A13     : 0x0000
000c
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000
001c
EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffff
fffe
Backtrace: 0x4017a41f:0x3ffd82a0 0x40145e39:0x3ffd82c0 0x401418d6:0x3ffd82f0 0x401
0aa81:0x3ffd8320 0x4017a44d:0x3ffd8370 0x4014a365:0x3ffd8390 0x4014ae3d:0x3ffd86b0
 0x40145984:0x3ffd9810
0x4017a41f: Socket::getSSL() const at E:/msys32/opt/devel/HttpServer/components/cp
p_utils/Socket.cpp:244
0x40145e39: Socket::send(unsigned char const*, unsigned int) const at e:\msys32\op
t\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/basic_string.h:4787
0x401418d6: WebSocket::send(std::__cxx11::basic_string<char, std::char_traits<char
>, std::allocator<char> >, unsigned char) at E:/msys32/opt/devel/HttpServer/compon
ents/cpp_utils/WebSocket.cpp:352
0x4010aa81: wsHandler(HttpRequest*, HttpResponse*) at E:/msys32/opt/devel/HttpServ
er/main/main.cpp:88
0x4017a44d: PathHandler::invokePathHandler(HttpRequest*, HttpResponse*) at E:/msys
32/opt/devel/HttpServer/components/cpp_utils/HttpServer.cpp:483
0x4014a365: HttpServerTask::processRequest(HttpRequest&) at E:/msys32/opt/devel/Ht
tpServer/components/cpp_utils/HttpServer.cpp:278
0x4014ae3d: HttpServerTask::run(void*) at E:/msys32/opt/devel/HttpServer/component
s/cpp_utils/HttpServer.cpp:278
0x40145984: Task::runTask(void*) at E:/msys32/opt/devel/HttpServer/components/cpp_
utils/Task.cpp:58

squonk11, 2017-12-17:
I think my problem is that the pointer returned by:
`WebSocket *pWebSocket = pRequest->getWebSocket();`
could be invalid?

squonk11, 2017-12-17:
I think I found the problem:
During the check if it is a WebSocket in the constructor of HttpRequest a check is done like this:
`getHeader(HTTP_HEADER_CONNECTION) == "Upgrade"`
but in my request there is:
> D (12410) HttpRequest: name="connection", value="keep-alive, Upgrade"

There is an additional "keep-alive" in the request. If I comment this our, then it works.
Is there a way to change the request (coming from Firefox/57.0)?

nkolban, 2017-12-17:
Great find.  I had assumed that the Connection header property would exclusively be `Upgrade` and nothing else.  The code has been changed to assume that the Connection property is now 1 or more tokens separated by the `,` character.  We now read the Connection, split it into its parts and then examine each of the parts to see if any of them are `Upgrade`.  The code changes have just been checked in.  It compiles cleanly but not yet tested.  Can you give it a spin and see how this feels?

squonk11, 2017-12-17:
I will test it tomorrow evening.

squonk11, 2017-12-18:
I tested the new version of cpp_utils. Now the problem seems to be fixed.

nkolban, 2017-12-18:
That's great news.  If/when you are ready, close out this issue.  New problems can be raised at any time as we are happy to handle as many issues as needed.  We want to avoid re-using the same issue for distinct and un-related problem reports.  I.e. we'd rather have as many issues to track individually than one large issue that spans multiple problems.

closed 2017-12-18 by squonk11

snahmad, 2018-05-25:
Hi,
Have you people used HttpServer with SSL enabled. It is not working for me crashing and closing socket connection when use from web browser my sd card file system.

I am improve c++ WebServer class which uses mongoose web serrver C API.
Currently it does not support SSL. I am putting it. not huge success so far.

squonk11, 2018-05-25:
No, until now I did not use HttpServer with SSL enabled. But I will have to use ist as soon as I want to use my webserver online on the internet. But currently I have no idea how to achieve that.

snahmad, 2018-05-25:
oh ok. can kolban tell us have he tested HttpServer with SSL.
It is single threaded. if we make it work with two thread as esp32 platform have two cores. I think it will speed up serving web pages.

flipatoms, 2018-09-10:
Hi Kolban,
I'm looking at mongoose for a small project and noticed your comment:
The mongoose library is licensed material ... happily, it is open source and 100% available for re-distribution as long as the complete source of the surrounding application is also open source. 

From what I understand the Apache 2.0 license does not force you to have a open source license on your project code just because you use the mongoose framework. Do you know if the mongoose framework had another license in 2017?