nkolban/esp32-snippets #65: TFTP.cpp issues error message

squonk11, 2017-09-17:
Dear Neil,
I am trying to use Your TFTP server SW (TFTP.cpp). Unfortunately it does not work. I am initializing it like You suggest:
	TFTP tftp;
	tftp.setBaseDir("/sdcard");
	tftp.start();
The log output shows:
D (5743) MyWiFi: DNS Server[0]: 192.168.1.1
D (5748) TFTP: Starting TFTP::start() on port 69
D (5753) Socket: bind: port=69, address=0x0
E (5753) Socket: listen: Operation not supported on socket
D (5763) TFTP: TFTP: Waiting for a request
D (5768) TFTP: Un-handled opcode: 61647
D (5768) TFTP: TFTP: Waiting for a request
D (5773) TFTP: Un-handled opcode: 61647

I think I am missing something. Do You have a hint for me?

Best regards

   Lothar

nkolban, 2017-09-17:
Oh oh ... its been a while since I tested that ... let me run up a test and see what happens.

squonk11, 2017-09-17:
I am also currently checking why it fails. Unfortunately I do not have the possibility to debug (e.g. using OCD). As far as I could track this issue so far it seems as if the call:
err = netconn_listen_with_backlog(sock->conn, (u8_t)backlog);
in the lwip sources fails with the error code "EOPNOTSUPP". Diving deeper seems to be too tricky for my skills...

nkolban, 2017-09-17:
Seeing as you are working on it live now ... let me do the same.   Can you open up a chat with me on IRC in channel `#ESP32`?

squonk11, 2017-09-17:
ups, I never used IRC. How can I do that?

nkolban, 2017-09-17:
I retested the TFTP code and sure enough, got exactly the error you reported.  I worked through the code and found that I had introduced some stupid errors.  This tells me that I have to perform far better regression testing and I apologize for the inconvenience.  The goods news is that I have corrected the coding errors and pushed a new version up to Github.  If you follow the steps to download the cpp_utils library and try again, I have more faith that it will work now.  I tested with an end-to-end TFTP test and it all seemed to work.  Please restest (if you are willing) and post back how you get on.

squonk11, 2017-09-17:
I updated the sources and encontered two problems:

1. I need to comment out "#include <iostream>" in Socket.cpp and Socket.h; otherwise I get a compile error in the esp-idf libraries
1. After that the start of the TFTP server seems to work but I get a stackoverflow:
D (5288) TFTP: Starting TFTP::start() on port 69
D (5293) Socket: >> listen: port: 69, isDatagram: 1
D (5298) Socket: >> createSocket: isDatagram: 1
D (5303) Socket: << createSocket: sockFd: 2
D (5308) Socket: >> bind: port=69, address=0x0
D (5313) Socket: << bind
D (5313) Socket: << listen
D (5318) TFTP: TFTP: Waiting ***ERROR*** A stack overflow in task main has been dete
cted.
abort() was called at PC 0x40088428 on core 0
0x40088428: vApplicationStackOverflowHook at E:/msys32/opt/esp/esp-idf/components/es
p32/panic.c:553

Do I need to give some task more stack size?

Best regards

   Lothar

nkolban, 2017-09-17:
Howdy, what is the precise error message you get during compilation with the source exactly as distributed?  I'd like to look into that in more depth to make sure that all is well.

If you get a stack overflow then yes, the amount of storage allocated to thread has been used up and you need to increase the stack size.  We should also check that you aren't somehow "leaking" stack resources.  Stack sizing for FreeRTOS is a concept of pure ESP-IDF and isn't directly covered in these libraries.  The assumption is that you have allocated enough storage as needed.

squonk11, 2017-09-17:
The error message I get if I include the iostream header is:
`In file included from e:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\iostream:39:0,
                 from E:/msys32/opt/devel/nk_wifi/main/Socket.h:20,
                 from E:/msys32/opt/devel/nk_wifi/main/TFTP.h:12,
                 from E:/msys32/opt/devel/nk_wifi/main/main.cpp:16:
e:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\ostream:335:49: error: macro "write" requires 3 arguments, but only 2 given
       write(const char_type* __s, streamsize __n);`

The esp-idf I am using is: `IDF Version: v3.0-dev-616-g9a26296a-dirty;`

If I comment it these #include lines out (in Socket.h and Socket.cpp) it compiles without errors. Maybe You are using a different compiler version?

Best regards

   Lothar

squonk11, 2017-09-17:
Concerning the stackoberflow issue: to be honest I do not know where to adjust this. In menuconfig? Or in Task.cpp?

nkolban, 2017-09-17:
On the stackoverflow issue, could you pastebin your application?  Assuming its not too large I can take a look through it.  I will be looking for anywhere you create a task.

squonk11, 2017-09-17:
I do not have pastebin account. Alternatively I could send You a zip of my sources - if You tell me Your eMail adress.

nkolban, 2017-09-17:
On the iostream error ... can you run a test for me?

Open up socket.h

and look at lines 28 onwards .. they will look like:

```
#undef accept
#undef bind
#undef close
#undef connect
#undef listen
#undef recv
#undef send
```

can you add
```
#undef write
```
so that it looks like:

```
#undef accept
#undef bind
#undef close
#undef connect
#undef listen
#undef recv
#undef send
#undef write
```
and try it again?

nkolban, 2017-09-17:
email address is kolban1@kolban.com

squonk11, 2017-09-17:
I added #undef write to Socket.h but the error is still the same:
`In file included from e:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\iostream:39:0,
                 from E:/msys32/opt/devel/nk_wifi/main/Socket.h:20,
                 from E:/msys32/opt/devel/nk_wifi/main/TFTP.h:12,
                 from E:/msys32/opt/devel/nk_wifi/main/main.cpp:16:
e:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\ostream:335:49: error: macro "write" requires 3 arguments, but only 2 given
       write(const char_type* __s, streamsize __n);`

nkolban, 2017-09-17:
One more test ... can you re-arrange the header in Socket.h to the following at the start:

```
#ifndef COMPONENTS_CPP_UTILS_SOCKET_H_
#define COMPONENTS_CPP_UTILS_SOCKET_H_

#include <mbedtls/platform.h>

#include <mbedtls/ctr_drbg.h>
#include <mbedtls/debug.h>
#include <mbedtls/entropy.h>
#include <mbedtls/error.h>
#include <mbedtls/net.h>
#include <mbedtls/ssl.h>

#include <lwip/inet.h>
#include <lwip/sockets.h>

#undef accept
#undef bind
#undef close
#undef connect
#undef listen
#undef recv
#undef read
#undef send
#undef write

#include <string>
#include <iostream>
#include <streambuf>
#include <cstdio>
#include <cstring>
```

squonk11, 2017-09-17:
after rearranging according to You latest post it now compiles without errors. But the problem with the stackoverflow prevails:
`D (5803) TFTP: TFTP: Waiting for a requ***ERROR*** A stack overflow in task main has
 been detected.`
The message "TFTP: Waiting for a requ" shows that the `TFTP::TFTP_Transaction::waitForRequest(Socket *pServerSocket)` is interrupted by the stackoverflow. But I do not have any idea why...

nkolban, 2017-09-17:
For the task stack size, run `make menuconfig` in your project directory.
Go to 
```
Component config ->
-> ESP32-specifc ->
```

Now look for the setting called `Main task stack size` and change it to something higher ... for example, 16000.

Let us then recompile and see what changes.   For the header files, am now committing that into the primary repository as a fix.

squonk11, 2017-09-17:
With setting the main task stack size to 16000 the statckoverflow issue is solved also. Tomorrow I will check if I am able to transfer files to the SD card.

nkolban, 2017-09-17:
Great news ... please let me know how we get on.

closed 2017-10-03 by squonk11