nkolban/esp32-snippets #66: ble is not working for sparkfun

Akskaypohankar, 2017-09-18:
When I tried to use this BLE Arduino library for Sparkfun things board it gave me following error I tried the number of different things but the problem is not solved.
  the error message 
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8
abort() was called at PC 0x400d8cd3 on core 0

Backtrace: 0x4008832c:0x3ffdc360 0x4008842b:0x3ffdc380 0x400d8cd3:0x3ffdc3a0 0x400da342:0x3ffdc3c0 0x400da4ce:0x3ffdc3e0 0x400da819:0x3ffdc420 0x400da859:0x3ffdc450 0x400da869:0x3ffdc470 0x400d667e:0x3ffdc490 0x400d52f9:0x3ffdc4b0 0x400d410d:0x3ffdc4f0 0x400f340d:0x3ffdc510 0x400ef4b2:0x3ffdc550

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8
abort() was called at PC 0x400d8cd3 on core 0

Backtrace: 0x4008832c:0x3ffdc360 0x4008842b:0x3ffdc380 0x400d8cd3:0x3ffdc3a0 0x400da342:0x3ffdc3c0 0x400da4ce:0x3ffdc3e0 0x400da819:0x3ffdc420 0x400da859:0x3ffdc450 0x400da869:0x3ffdc470 0x400d667e:0x3ffdc490 0x400d52f9:0x3ffdc4b0 0x400d410d:0x3ffdc4f0 0x400f340d:0x3ffdc510 0x400ef4b2:0x3ffdc550

Rebooting...﻿
the error message repeats itself

nkolban, 2017-09-18:
Howdy,
A stack trace without context doesn't help that much.  What we need to do is pin down within the code where the error is occurring.  Sprinkle in some logging statements and keep recompiling to find where in your code the last succesful statement can be found.

Also, have a good read of the following:

https://github.com/me-no-dev/EspExceptionDecoder

And get this going.  What this will do is decode a back trace to tell us where within the code the issue is being encountered.

Akskaypohankar, 2017-09-18:
SparkFun esp 32thing code used BLE-uart from examples,
ERROR message 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8
abort() was called at PC 0x400d8cd3 on core 0

Backtrace: 0x4008832c:0x3ffdc360 0x4008842b:0x3ffdc380 0x400d8cd3:0x3ffdc3a0 0x400da342:0x3ffdc3c0 0x400da4ce:0x3ffdc3e0 0x400da819:0x3ffdc420 0x400da859:0x3ffdc450 0x400da869:0x3ffdc470 0x400d667e:0x3ffdc490 0x400d52f9:0x3ffdc4b0 0x400d410d:0x3ffdc4f0 0x400f340d:0x3ffdc510 0x400ef4b2:0x3ffdc550

EspExceptionDecoder reasult 

Decoding 17 results
0x400d8cd3: std::__throw_length_error(char const*) at ?? line ?
0x4008832c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x4008842b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x400d8cd3: std::__throw_length_error(char const*) at ?? line ?
0x400da342: std::__cxx11::basic_string  , std::allocator  >::_M_create(unsigned int&, unsigned int) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h line 2200
0x400da4ce: std::__cxx11::basic_string  , std::allocator  >::_M_mutate(unsigned int, unsigned int, char const*, unsigned int) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h line 2200
0x400da819: std::__cxx11::basic_string  , std::allocator  >::_M_replace(unsigned int, unsigned int, char const*, unsigned int) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h line 2200
0x400da859: std::__cxx11::basic_string  , std::allocator  >::assign(char const*) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h line 2200
0x400da869: std::__cxx11::basic_string  , std::allocator  >::operator=(char const*) at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h line 2200
0x400d667e: FreeRTOS::Semaphore::give() at C:\Users\Akshay\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/FreeRTOS.cpp line 126
0x400d52f9: BLEServer::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at C:\Users\Akshay\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEServer.cpp line 47
0x400d410d: BLEDevice::gattServerEventHandler(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at C:\Users\Akshay\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEDevice.cpp line 43
0x400f340d: btc_gatts_cb_to_app at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c line 54
:  (inlined by) btc_gatts_cb_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c line 731
0x400ef4b2: btc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/core/btc_task.c line 85

nkolban, 2017-09-18:
My immediate guess is that the task that is running the BLE callbacks doesn't have enough stack size.  When was the last time your refreshed your Arduino ESP32 drivers.

I am running with the latest Arduino IDE and then have installed the latest Arduino core libraries from here:

https://github.com/espressif/arduino-esp32

Specifically, it appears you will need a driver that is post August 17th ... see:

https://github.com/espressif/arduino-esp32/pull/583

Akskaypohankar, 2017-09-18:
Arduino ide 1.8.4 and esp32 updated just 3 days ago when you asked me to do for previous error

Akskaypohankar, 2017-09-18:
[image]
stack size is 8192

nkolban, 2017-09-18:
Are you compiling and running any of the sample apps or is this a custom app?  If you could elaborate on that aspect, that might give me the next clue.  Unfortunately I have so many things on the go, that they all blur after a while, so if we have had previous interactions which might add knowledge into our puzzle, let me ask you to reference those as well.  I try and make each ticket self contained so that when new information arrives, I can re-read the one ticket and try and come back up to speed.

From the trace, the exception is occurring inside a BLE callback which runs in the BLE Stack.  The exception is being thrown in a call to release a "semaphore" which is a pretty innocuous thing to do.  Another test/clue we can perform is to switch on BLE trace using this recipe:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

(See section "Switching on debugging)

If we switch that on and re-rerun, we may also see more clues.

Akskaypohankar, 2017-09-19:
OHH god must be playing with me . . . . .
I did what you suggested in above comment (See section "Switching on debugging) and changed 
#define CONFIG_LOG_DEFAULT 1   to  5 
and restarted Arduino and reuploaded the same sketch (example /BLE_uart)
at first serial monitor was empty then after reset results was as fallows

e⸮⸮⸮⸮]	⸮⸮⸮⸮⸮ف⸮⸮⸮Iiqk⸮⸮⸮⸮⸮⸮.⸮⸮EAQ⸮⸮ׅ⸮⸮⸮%⸮)⸮%Q⸮⸮⸮t⸮⸮⸮⸮&⸮U⸮e⸮⸮⸮⸮⸮⸮⸮⸮ߢ⸮⸮⸮⸮#⸮崳⸮⸮]	⸮⸮⸮⸮⸮؁⸮⸮⸮IIyio
⸮⸮⸮⸮.⸮⸮⸮EaAQ⸮T⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮S,⸮⸮⸮⸮⸮⸮⸮⸮&$⸮V⸮e⸮⸮⸮21⸮⸮⸮⸮⸮O⸮⸮⸮⸮⸮⸮⸮K⸮s7⸮⸮iAaX⸮⸮⸮⸮⸮⸮⸮⸮⸮ٙ#⸮,⸮⸮E⸮⸮ya⸮`0⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮XVZE⸮⸮ya𰐜⸮⸮⸮E⸮⸮ya񰰜⸮⸮E⸮⸮ya𰐔⸮⸮E⸮⸮ya⸮0⸮⸮⸮⸮⸮⸮Y'T⸮⸮	5Ž⸮⸮A⸮鶚⸮⸮⸮⸮+K⸮⸮⸮M⸮⸮⸮a`⸮⸮⸮⸮⸮⸮⸮⸮C!⸮+K⸮⸮⸮M⸮⸮⸮a0⸮⸮⸮⸮Y˪⸮⸮6(⸮⸮⸮⸮qa⸮4⸮⸮⸮⸮⸮⸮⸮⸮⸮YK⸮*"⸮⸮⸮⸮u`⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮vYK⸮⸮⸮⸮4)⸮⸮⸮⸮⸮A0⸮⸮⸮⸮ʢa5⸮D (5991) BLEServer: >> registerApp - 0
V (5991) FreeRTOS: Semaphore taking: name: RegisterAppEvt (0x3ffd5230), owner: <N/A> for registerApp
V (5993) FreeRTOS: Semaphore taken:  name: RegisterAppEvt (0x3ffd5230), owner: registerApp
V (6000) FreeRTOS: Semaphore waiting: name: RegisterAppEvt (0x3ffd5230), owner: registerApp for registerApp
D (6001) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
D (6017) BLEUtils: GATT ServerEvent: ESP_GATTS_REG_EVT
D (6022) BLEUtils: [status: ESP_GATT_OK, app_id: 0]
D (6026) BLEServer: >> handleGATTServerEvent: ESP_GATTS_REG_EVT
V (6032) FreeRTOS: Semaphore released: name: RegisterAppEvt (0x3ffd5230), owner: registerApp
V (6032) FreeRTOS: Semaphore giving: name: RegisterAppEvt (0x3ffd5230), owner: registerApp
D (6040) BLEServer: << registerApp
D (6048) BLEServer: << handleGATTServerEvent
D (6052) BLEServer: >> createService - 6e400001-b5a3-f393-e0a9-e50e24dcca9e
V (6062) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffd4d58), owner: <N/A> for createService
V (6071) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffd4d58), owner: createService
D (6079) BLEService: >> executeCreate() - Creating service (esp_ble_gatts_create_service) service uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
V (6091) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffd4fc8), owner: <N/A> for executeCreate
V (6099) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffd4fc8), owner: executeCreate
V (6107) FreeRTOS: Semaphore waiting: name: CreateEvt (0x3ffd4fc8), owner: executeCreate for executeCreate
D (6108) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CREATE_EVT
D (6124) BLEUtils: GATT ServerEvent: ESP_GATTS_CREATE_EVT
D (6129) BLEUtils: [status: ESP_GATT_OK, service_handle: 0x28, service_id: [uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, inst_id: 0]]
D (6141) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
D (6147) BLEService: >> setHandle - Handle=0x28, service UUID=6e400001-b5a3-f393-e0a9-e50e24dcca9e)
D (6155) BLEService: << setHandle
V (6158) FreeRTOS: Semaphore released: name: CreateEvt (0x3ffd4fc8), owner: executeCreate
V (6158) FreeRTOS: Semaphore giving: name: CreateEvt (0x3ffd4fc8), owner: executeCreate
D (6166) BLEService: << executeCreate
V (6174) FreeRTOS: Semaphore giving: name: CreateEvt (0x3ffd4d58), owner: createService
V (6177) FreeRTOS: Semaphore waiting: name: CreateEvt (0x3ffd4d58), owner: createService for createService
D (6185) BLEServer: << handleGATTServerEvent
V (6194) FreeRTOS: Semaphore released: name: CreateEvt (0x3ffd4d58), owner: createService
D (6206) BLEServer: << createService
D (6210) BLEService: >> addCharacteristic()
D (6214) BLEService: Adding characteristic (esp_ble_gatts_add_char): uuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e to service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
D (6229) BLEService: << addCharacteristic()
D (6233) BLECharacteristic: >> addDescriptor(): Adding UUID: 00002902-0000-1000-8000-00805f9b34fb, handle: 0xffff to UUID: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, handle: 0xffff Notify 
D (6249) BLECharacteristic: << addDescriptor()
D (6254) BLEService: >> addCharacteristic()
D (6257) BLEService: Adding characteristic (esp_ble_gatts_add_char): uuid=6e400002-b5a3-f393-e0a9-e50e24dcca9e to service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
D (6273) BLEService: << addCharacteristic()
D (6277) BLEService: >> start(): Starting service (esp_ble_gatts_start_service): UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
D (6289) BLECharacteristic: >> executeCreate()
D (6293) BLECharacteristic: Registering characteristic (esp_ble_gatts_add_char): uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x28
V (6310) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffe2000), owner: <N/A> for executeCreate
V (6318) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffe2000), owner: executeCreate
V (6326) FreeRTOS: Semaphore waiting: name: CreateEvt (0x3ffe2000), owner: executeCreate for executeCreate
D (6327) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_ADD_CHAR_EVT
D (6343) BLEUtils: GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
D (6348) BLEUtils: [status: ESP_GATT_OK, attr_handle: 0x2a, service_handle: 0x28, char_uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e]
D (6360) BLEServer: >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
D (6366) BLECharacteristic: >> setHandle: handle=0x2a, characteristic uuid=6e400002-b5a3-f393-e0a9-e50e24dcca9e
D (6376) BLECharacteristic: << setHandle
V (6379) FreeRTOS: Semaphore giving: name: AddCharEvt (0x3ffd4f68), owner: <N/A>
V (6387) FreeRTOS: Semaphore giving: name: CreateEvt (0x3ffe2000), owner: executeCreate
V (6387) FreeRTOS: Semaphore released: name: CreateEvt (0x3ffe2000), owner: executeCreate
[... cut ...]
D (92617) BLECharacteristic: >> notify: length: 1
D (92621) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
D (92629) GeneralUtils: 0000 1e                                               .
D (92636) BLECharacteristic: << notifications disabled; ignoring

no apps just an example code from your library straightly uploaded to Sparkfun without any changes for Testing.
for our old communication, you can check your youtube video comment section but that's not important now that problem is solved

Akskaypohankar, 2017-09-19:
also, I used BLE uart terminal app from play store to see received data and send data. 
The data sent from this app was successfully received on Sparkfun 
*********
Received Value: hi
*********
D (940854) BLEServer: << handleGATTServerEvent
D (940859) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_RESPONSE_EVT
D (940866) BLEUtils: GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
D (940871) BLEUtils: [status: ESP_GATT_OK, handle: 0x2a]
D (940876) BLEServer: >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
D (940883) BLEServer: << handleGATTServerEvent
*** Sent Value: 121 ***
D (941249) BLECharacteristic: >> setValue: length=1, data=79, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e
D (941250) BLECharacteristic: << setValue
D (941254) BLECharacteristic: >> notify: length: 1
D (941258) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
D (941267) GeneralUtils: 0000 79                                               y
D (941274) BLECharacteristic: << notify
D (941275) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONF_EVT
D (941285) BLEUtils: GATT ServerEvent: ESP_GATTS_CONF_EVT
D (941290) BLEUtils: [status: ESP_GATT_OK, conn_id: 0x00]
D (941295) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
V (941301) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe24d4), owner: <N/A>
V (941308) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe1eb4), owner: <N/A>
D (941315) BLEServer: << handleGATTServerEvent
*** Sent Value: 122 ***
D (942278) BLECharacteristic: >> setValue: length=1, data=7a, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e
D (942279) BLECharacteristic: << setValue
D (942283) BLECharacteristic: >> notify: length: 1
D (942287) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
D (942296) GeneralUtils: 0000 7a                                               z
D (942303) BLECharacteristic: << notify
D (942304) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONF_EVT
D (942314) BLEUtils: GATT ServerEvent: ESP_GATTS_CONF_EVT
D (942319) BLEUtils: [status: ESP_GATT_OK, conn_id: 0x00]
D (942324) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
V (942330) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe24d4), owner: <N/A>
V (942337) FreeRTOS: Semaphore giving: name: ConfEvt (0x3ffe1eb4), owner: <N/A>
D (942344) BLEServer: << handleGATTServerEvent
D (942504) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_WRITE_EVT
D (942504) BLEUtils: GATT ServerEvent: ESP_GATTS_WRITE_EVT
D (942506) BLEUtils: [conn_id: 0, trans_id: 5, bda: 74:04:2b:2d:56:9c, handle: 0x2a, offset: 0, need_rsp: 1, is_prep: 0, len: 2]
D (942517) BLEUtils: [Data: 3130]
D (942520) BLEServer: >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
D (942526) BLECharacteristic: >> setValue: length=2, data=3130, characteristic UUID=6e400002-b5a3-f393-e0a9-e50e24dcca9e
D (942536) BLECharacteristic: << setValue
D (942540) BLECharacteristic:  - Response to write event: New value: handle: 2a, uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
D (942551) BLECharacteristic:  - Data: length: 2, data: 3130
*********
Received Value: 10
*********
D (942560) BLEServer: << handleGATTServerEvent
D (942565) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_RESPONSE_EVT
D (942572) BLEUtils: GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
D (942577) BLEUtils: [status: ESP_GATT_OK, handle: 0x2a]
D (942582) BLEServer: >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
D (942589) BLEServer: << handleGATTServerEvent
*** Sent Value: 123 ***

nkolban, 2017-09-19:
Howdy ... are we saying that the puzzle has resolved itself?

Akskaypohankar, 2017-09-19:
solved ! I don't think so
 1) after uploading it won't work directly you need to reset 
2) it gives a lot of other details in between actual data. We need to see 
is there any way to get rid of this data without compromising this result

nkolban, 2017-09-19:
Sorry ... I misunderstood ... I had assumed that the original problem was that it crashed ... and in the traces you sent, we weren't crashing any more and that made me think the issue might have been resolved.

In your Arduino IDE, what "board type" do you have selected from the options?

As for the current/new story ... I'm seeing indications that the BLE partner is trying to engage security and I haven't done any work on that yet.  Let me think on this a bit and see what it might take to add security exchanges.