nkolban/esp32-snippets #67: BLE_client sample crashes with own service UUID

pit001, 2017-09-19:
Hi, thank you for providing the BTLE C++ Api. To work with the nativ ESP-IDF it's like try to build a skyscraper but playing in the sand. 

I have checked your BLE_client sample in the Arduino IDE. It scans well. 
If I replace the service UUID with a own one (found before) it crashes at scan->stop. This happens in line
84: advertisedDevice.getScan()->stop();
and also after the 30 seconds scan timeout if I comment out the line 84. 

It doesn't find the UUID but crashes.

I have tried it with:
static BLEUUID serviceUUID("0000fe95-0000-1000-8000-00805f9b34fb"); //Flower care
static BLEUUID serviceUUID("0000aa80-0000-1000-8000-00805f9b34fb"); //Texas instruments CC2650 SensorTag

Peter

Starting Arduino BLE Client application...<\r><\n>
BLE Advertised Device found: Name: CC2650 SensorTag, Address: c4:be:84:71:39:84, manufacturer data: 0d00030000, serviceUUID: 0000aa80-0000-1000-8000-00805f9b34fb<\r><\n>
Found our device!  address: Forming a connection to c4:be:84:71:39:84<\r><\n>
 - Created client<\r><\n>
E (30669) BT: l2cble_start_conn_update, the last connection update command still pending.<\r><\n>
E (30670) BT: l2cble_start_conn_update, the last connection update command still pending.<\r><\n>
 - Connected to server<\r><\n>
E (31118) BT: l2cble_start_conn_update, staus = 1, line = 509<\r><\n>
E (31695) BT: bta_gattc_cmpl_cback unknown conn_id =  3, ignore data<\r><\n>
Failed to find our service UUID: 0000aa80-0000-1000-8000-00805f9b34fb<\r><\n>
Setting new characteristic value to "Time since boot: 4"<\r><\n>
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.<\r><\n>
Register dump:<\r><\n>
PC      : 0x400d622e  PS      : 0x00060130  A0      : 0x800d4913  A1      : 0x3ffdaff0  <\r><\n>
A2      : 0x00000088  A3      : 0x3ffdb020  A4      : 0x3ffe4506  A5      : 0x0000ff00  <\r><\n>
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d43b4  A9      : 0x3ffdafc0  <\r><\n>
A10     : 0x00000000  A11     : 0x3f4021aa  A12     : 0x3f4021b4  A13     : 0x00000000  <\r><\n>
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  <\r><\n>
EXCVADDR: 0x00000088  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  <\r><\n>
<\r><\n>
Backtrace: 0x400d622e:0x3ffdaff0 0x400d4910:0x3ffdb010 0x400d2dfc:0x3ffdb070 0x4018496e:0x3ffdb0c0<\r><\n>
<\r><\n>
Rebooting...<\r><\n>
ets Jun  8 2016 00:22:57<\r><\n>
<\r><\n>
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r><\n>
configsip: 0, SPIWP:0xee<\r><\n>
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r><\n>
mode:DIO, clock div:1<\r><\n>
load:0x3fff0010,len:4<\r><\n>
load:0x3fff0014,len:708<\r><\n>
load:0x40078000,len:0<\r><\n>
load:0x40078000,len:11460<\r><\n>
entry 0x400789f4<\r><\n>
Starting Arduino BLE Client application...

nkolban, 2017-09-19:
Howdy my friend,
Time for us to roll up our sleeves and find out what is going on.  The first thing ... in Arduino land, we have a utility that decodes the back trace to tell us the source code that threw the exception ... see:

https://github.com/me-no-dev/EspExceptionDecoder

Get that working and we'll find out where in the code the problem is occurring.

Next, we can switch on BLE C++ trace ... see:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

and the section called Switching on debugging.

chegewara, 2017-09-19:
It can be problem with this tag (just guessing)
https://github.com/sandeepmistry/node-sensortag/issues/45#issuecomment-118979119

nkolban, 2017-09-19:
Did you post to the wrong issue?

chegewara, 2017-09-19:
No, its same issue with same CC2650 SensorTag, just in different framework. I think it can be connected and related with tag not sending advertise data. As i understand they keep trying to scan with `SensorTag.SCAN_DUPLICATES = true;` or maybe im just a bit tired

pit001, 2017-09-19:
Ok, some more info. The CC2650 SensorTag has the newest FW, also it's only one of more test objects.
 I use the Arduino IDE 1.8.4 on Win7 64bit with ESP32_BLE.zip Sep 10, 2017.

In the #define sdkconfig.h there is no  CONFIG_LOG_DEFAULT, but a CONFIG_LOG_DEFAULT**_LEVEL** ... changed to 5.
 Good to know the the ExceptionDecoder also works with ESP32 ;)

Hope if helps !
Peter

// The remote service we wish to connect to.
//static BLEUUID serviceUUID("91bad492-b950-4226-aa2b-4ede9fa42f59"); //enabled in 1. run
static BLEUUID serviceUUID("0000aa80-0000-1000-8000-00805f9b34fb"); //enabled in 2. run

Starting Arduino BLE Client application...<\r><\n>
D (5037) BLEUtils: Received a GAP event: ESP_GAP_BLE_LOCAL_IR_EVT<\r><\n>
D (5084) BLEUtils: Received a GAP event: ESP_GAP_BLE_LOCAL_ER_EVT<\r><\n>
D (5230) BLEScan: >> start(30)<\r><\n>
V (5230) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffd61a0), owner: <N/A> for start<\r><\n>
V (5231) FreeRTOS: Semaphore taken:  name: ScanEnd (0x3ffd61a0), owner: start<\r><\n>
V (5236) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffd61a0), owner: start for start<\r><\n>
D (5239) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT<\r><\n>
D (5251) BLEUtils: [status: 0]<\r><\n>
D (5254) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT<\r><\n>
D (5261) BLEUtils: [status: 0]<\r><\n>
D (5550) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT<\r><\n>
D (5550) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: c4:7c:8d:62:e2:95, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -83, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 23, scan_rsp_len: 13<\r><\n>
D (5569) BLEAdvertisedDevice: - rssi: -83<\r><\n>
D (5573) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06<\r><\n>
D (5580) BLEAdvertisedDevice: Type: 0x02 (ESP_BLE_AD_TYPE_16SRV_PART), length: 2, data: 95fe<\r><\n>
D (5588) BLEAdvertisedDevice: - serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb<\r><\n>
D (5596) BLEAdvertisedDevice: Type: 0x16 (ESP_BLE_AD_TYPE_SERVICE_DATA), length: 14, data: 95fe310298000595e2628d7cc40d<\r><\n>
D (5606) BLEAdvertisedDevice: Unhandled type<\r><\n>
D (5610) BLEAdvertisedDevice: Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 11, data: 466c6f7765722063617265<\r><\n>
D (5620) BLEAdvertisedDevice: - name: Flower care<\r><\n>
BLE Advertised Device found: Name: Flower care, Address: c4:7c:8d:62:e2:95, serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb<\r><\n>
D (6451) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT<\r><\n>
D (6451) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: c4:7c:8d:62:e2:a4, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -81, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 23, scan_rsp_len: 13<\r><\n>
D (6470) BLEAdvertisedDevice: - rssi: -81<\r><\n>
D (6474) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06<\r><\n>
D (6481) BLEAdvertisedDevice: Type: 0x02 (ESP_BLE_AD_TYPE_16SRV_PART), length: 2, data: 95fe<\r><\n>
D (6489) BLEAdvertisedDevice: - serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb<\r><\n>
D (6497) BLEAdvertisedDevice: Type: 0x16 (ESP_BLE_AD_TYPE_SERVICE_DATA), length: 14, data: 95fe3102980004a4e2628d7cc40d<\r><\n>
D (6507) BLEAdvertisedDevice: Unhandled type<\r><\n>
D (6511) BLEAdvertisedDevice: Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 11, data: 466c6f7765722063617265<\r><\n>
D (6521) BLEAdvertisedDevice: - name: Flower care<\r><\n>
BLE Advertised Device found: Name: Flower care, Address: c4:7c:8d:62:e2:a4, serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb<\r><\n>
D (7655) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT<\r><\n>
D (7655) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: c4:7c:8d:61:b2:ff, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -58, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 27, scan_rsp_len: 13<\r><\n>
D (7674) BLEAdvertisedDevice: - rssi: -58<\r><\n>
D (7678) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06<\r><\n>
D (7685) BLEAdvertisedDevice: Type: 0x02 (ESP_BLE_AD_TYPE_16SRV_PART), length: 2, data: 95fe<\r><\n>
D (7693) BLEAdvertisedDevice: - serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb<\r><\n>
D (7701) BLEAdvertisedDevice: Type: 0x16 (ESP_BLE_AD_TYPE_SERVICE_DATA), length: 18, data: 95fe710298008bffb2618d7cc40d08100100<\r><\n>
D (7712) BLEAdvertisedDevice: Unhandled type<\r><\n>
D (7716) BLEAdvertisedDevice: Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 11, data: 466c6f7765722063617265<\r><\n>
D (7725) BLEAdvertisedDevice: - name: Flower care<\r><\n>
BLE Advertised Device found: Name: Flower care, Address: c4:7c:8d:61:b2:ff, serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb<\r><\n>
D (7977) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT<\r><\n>
D (7978) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: c4:be:84:71:39:84, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -54, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 14, scan_rsp_len: 27<\r><\n>
D (7996) BLEAdvertisedDevice: - rssi: -54<\r><\n>
D (8000) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06<\r><\n>
[... cut ...]
0x400d815d: FreeRTOS::Semaphore::take(std::__cxx11::basic_string  , std::allocator  >) at C:\...\portable\sketchbook\libraries\ESP32_BLE\src/FreeRTOS.cpp line 137
0x400d5214: BLERemoteCharacteristic::writeValue(std::__cxx11::basic_string  , std::allocator  >, bool) at C:\...\portable\sketchbook\libraries\ESP32_BLE\src/BLERemoteCharacteristic.cpp line 288
0x400d32a4: loop() at C:\SyncBase\Dropbox\Projekte\20170906_ESP32_Bluetooth\BLE_client/BLE_client.ino line 125
0x40186abe: loopTask(void*) at C:\...\hardware\espressif\esp32\cores\esp32/main.cpp line 18 (discriminator 1)

nkolban, 2017-09-20:
Working this now ...

First change made has been the addition of logging the "type" of undefined type that is being logged in the advertising responses as an "unhandled type".  Once that fix is pushed, we will then be able to at least learn more about the type that was unhandled beyond "one wasnt handled".

nkolban, 2017-09-20:
I had a look at the client source code and am puzzled as the path shown in the trace shouldn't be possible in the sample I provided.  Can you pastebin the exact source you are using that manifests this problem?

pit001, 2017-09-20:
It's exact the source you provided. For the test I get it new from examples. Only the line  with the UUID was changed... see below.

Tell me if I can further tests, Peter

// The remote service we wish to connect to.
//static BLEUUID serviceUUID("91bad492-b950-4226-aa2b-4ede9fa42f59"); //enabled in 1. run
static BLEUUID serviceUUID("0000aa80-0000-1000-8000-00805f9b34fb"); //enabled in 2. run

pit001, 2017-09-20:
what's surprising for me the hole time is, that we can get a service UUID before we have a connection  (line 77) !? Update: OK - there is one (GAP), but this one isn't available in connected mode (GATT). And we can not postulate that.

The normal case I know is to:  scan, find a match by name or mac, (stop scanning,)  connect, read GATT (services, characteristics, values, descriptions), (set notifications,) ...

OK, I have changed the match from serviceUUID to deviceUUID (MAC), now the device gets found. Also it reads the GATT service IDs (and the missed one is not listed). I uncomment the write to service (//pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());) and the device keeps living. 

Some finetuning... 

Now I can search for an GAPP service (i.e. f000aa00-0451-4000-b000-000000000000), followed by Characteristic and write the value... nice :-) 

Addons:
1)
I had to add a delay(); in connectToServer() otherwise I get scrambled text output. Something seems to be wrong.
delay(50); /*--D- -(-6-293--)- -B-LE-U--t-i-l-s : coRennceecivtTeodS ae rvGeAPr  -e-ve--n-t-: --E-S-P-_-GA--P-_-B-LE--_-S-C-A-N-_S--TO--P-_C-O--M-P-L-E-T-E-_E--VT-<\r>-<\n>
-D-- -(6--3-0-7<\r>)<\n>

2)
advertisedDevice.getApperance()); // Appearance ?

3)
E (11925) BT: bta_gattc_cmpl_cback unknown conn_id =  3, ignore data<\r> Is this OK?

4) in loop() if (connected) { ... we should only write to the char if we rearly rearly sure that the char (and service) was found/is existing (or the function handel this)!!

5) 
got this sometimes:
V (9522) FreeRTOS: Semaphore released: name: SearchCmplEvt (0x3ffd6504), owner: getServices<\r><\n>
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idD (9530) BLEClient: << getServices<\r><\n>
f-public/components/heap/./heap_caps.c", line 171, function: heap_caps_free<\r><\n>
abort() was called at PC 0x4015bedf on core 0<\r><\n>
<\r><\n>
Backtrace: 0x400887fc:0x3ffde060 0x400888fb:0x3ffde080 0x4015bedf:0x3ffde0a0 0x400839bb:0x3ffde0d0 0x40083e2d:0x3ffde0f0 0x4000bec7:0x3ffde110 0x400d9ac1:0x3ffde130 0x400d9b01:0x3ffde150 0x400d9c58:0x3ffde170 0x400d9cc4:0x3ffde190 0x400d9cd5:0x3ffde1b0 0x400dca3c:0x3ffde1d0 0x400d9ef4:0x3ffde1f0 0x400da661:0x3ffde210 0x400da679:0x3ffde230 0x400d823b:0x3ffde250 0x400d82d6:0x3ffde340 0x400d46a2:0x3ffde380 0x400d4a2d:0x3ffde410 0x400f4441:0x3ffde460 0x400f13b0:0x3ffde4e0<\r><\n>
<\r><\n>
Rebooting...<\r><\n>
6)
in line 88: Serial.print("Found our device!  address: ");  there is no following address outout ?

- Peter

LOG:
ets Jun  8 2016 00:22:57<\r><\n>
<\r><\n>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r><\n>
flash read err, 1000<\r><\n>
ets_main.c 371 <\r><\n>
ets Jun  8 2016 00:22:57<\r><\n>
<\r><\n>
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r><\n>
configsip: 0, SPIWP:0xee<\r><\n>
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r><\n>
mode:DIO, clock div:1<\r><\n>
load:0x3fff0010,len:4<\r><\n>
load:0x3fff0014,len:708<\r><\n>
load:0x40078000,len:0<\r><\n>
load:0x40078000,len:11460<\r><\n>
entry 0x400789f4<\r><\n>
Starting Arduino BLE Client application...<\r><\n>
D (5445) BLEUtils: Received a GAP event: ESP_GAP_BLE_LOCAL_IR_EVT<\r><\n>
D (5448) BLEUtils: Received a GAP event: ESP_GAP_BLE_LOCAL_ER_EVT<\r><\n>
D (5592) BLEScan: >> start(30)<\r><\n>
V (5592) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffd6200), owner: <N/A> for start<\r><\n>
V (5593) FreeRTOS: Semaphore taken:  name: ScanEnd (0x3ffd6200), owner: start<\r><\n>
V (5598) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffd6200), owner: start for start<\r><\n>
D (5601) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT<\r><\n>
D (5613) BLEUtils: [status: 0]<\r><\n>
D (5616) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT<\r><\n>
D (5623) BLEUtils: [status: 0]<\r><\n>
D (6211) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT<\r><\n>
D (6211) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: c4:be:84:71:39:84, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -50, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 14, scan_rsp_len: 27<\r><\n>
D (6230) BLEAdvertisedDevice: - rssi: -50<\r><\n>
[... 323 lines cut ...]
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    connectToServer(*pServerAddress); //this internal reads the GATT
    doConnect = false;
    connected = true;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected)
  {
    Serial.println("----------------- still connected ------------------------");
    String newValue = "Time since boot: " + String(millis() / 1000);

    if (characteristicFound == true)
    {
      Serial.println("Setting new characteristic value to \"" + newValue + "\"");
      // Set the characteristic's value to be the array of bytes that is actually a string.
      pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
    }
  }

  delay(1000); // Delay a second between loops.
} // End of loop

nkolban, 2017-09-20:
Howdy Peter,
Am at work just now but will study the report in detail this evening ... however the immediate thing I wanted to examine was the source of the "BLE_client.ino" Arduino source file.  The trace we were examining wasn't possible against the latest source.   For example, the trace showed that the function called "connectToServer" was detecting an error but yet the code following that was still running.  In the current source in Github, we check that connectToServer() succeeds before continuing.  Here is the latest source file in the Github repository:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino

Can I ask you to try this source file and make any changes you need/desire to this?  Create a new Arduino project rather than replace your own source ... that way you can refer to both as you work.  Look especially at the function called `connectToServer()`.   Notice that it returns a `bool` return code while the copy in your code does not.

Looking at the Git file history, the change was committed on 2017-09-07.

pit001, 2017-09-20:
Hi  - no freeze with this code. :-)
It also didn't find the service UUID, but this is well known - see the "Update:" in my last post.
And - my code was from the zip file (see my 2nt post) - I didn't aware the newer one, sorry.  
Peter

ets Jun  8 2016 00:22:57<\r><\n>
<\r><\n>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r><\n>
flash read err, 1000<\r><\n>
ets_main.c 371 <\r><\n>
ets Jun  8 2016 00:22:57<\r><\n>
<\r><\n>
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r><\n>
configsip: 0, SPIWP:0xee<\r><\n>
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r><\n>
mode:DIO, clock div:1<\r><\n>
load:0x3fff0010,len:4<\r><\n>
load:0x3fff0014,len:708<\r><\n>
load:0x40078000,len:0<\r><\n>
load:0x40078000,len:11460<\r><\n>
entry 0x400789f4<\r><\n>
Starting Arduino BLE Client application...<\r><\n>
D (4985) BLEUtils: Received a GAP event: ESP_GAP_BLE_LOCAL_IR_EVT<\r><\n>
D (4988) BLEUtils: Received a GAP event: ESP_GAP_BLE_LOCAL_ER_EVT<\r><\n>
D (5178) BLEScan: >> start(30)<\r><\n>
V (5178) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffd6200), owner: <N/A> for start<\r><\n>
V (5179) FreeRTOS: Semaphore taken:  name: ScanEnd (0x3ffd6200), owner: start<\r><\n>
V (5184) FreeRTOS: Semaphore taking: name: ScanEnd (0x3ffd6200), owner: start for start<\r><\n>
D (5187) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT<\r><\n>
D (5199) BLEUtils: [status: 0]<\r><\n>
D (5202) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT<\r><\n>
D (5208) BLEUtils: [status: 0]<\r><\n>
D (10656) BLEUtils: Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT<\r><\n>
D (10656) BLEUtils: search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: c4:be:84:71:39:84, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -50, ble_adv: ??, flag: 6, num_resps: 1, adv_data_len: 14, scan_rsp_len: 27<\r><\n>
D (10675) BLEAdvertisedDevice: - rssi: -50<\r><\n>
D (10679) BLEAdvertisedDevice: Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06<\r><\n>
D (10686) BLEAdvertisedDevice: Type: 0x02 (ESP_BLE_AD_TYPE_16SRV_PART), length: 2, data: 80aa<\r><\n>
D (10695) BLEAdvertisedDevice: - serviceUUID: 0000aa80-0000-1000-8000-00805f9b34fb<\r><\n>
D (10702) BLEAdvertisedDevice: Type: 0xff (ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE), length: 5, data: 0d00030000<\r><\n>
D (10712) BLEAdvertisedDevice: - manufacturer data: 0d00030000<\r><\n>
D (10717) BLEAdvertisedDevice: Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 16, data: 4343323635302053656e736f72546167<\r><\n>
D (10728) BLEAdvertisedDevice: - name: CC2650 SensorTag<\r><\n>
BLE Advertised Device found: Name: CC2650 SensorTag, Address: c4:be:84:71:39:84, manufacturer data: 0d00030000, serviceUUID: 0000aa80-0000-1000-8000-00805f9b34fb<\r><\n>
Found our device!  address: D (10749) BLEScan: >> stop()<\r><\n>
V (10752) FreeRTOS: Semaphore taken:  name: ScanEnd (0x3ffd6200), owner: start<\r><\n>
V (10752) FreeRTOS: Semaphore giving: name: ScanEnd (0x3ffd6200), owner: start<\r><\n>
V (10759) FreeRTOS: Semaphore giving: name: ScanEnd (0x3ffd6200), owner: start<\r><\n>
D (10766) BLEScan: << stop()<\r><\n>
D (10773) BLEScan: << start()<\r><\n>
FDor (mi1n0g7 7a8) c oBnLnEeUcttiliosn:  tRoe cce4i:vebed: 8a 4G:7A1P :3e9ve:8n4t:<\r><\n>
  E-S PC_rGeAPat_eBdL Ec_SlCiAenNt_<\r>S<\n>
TOP_COMPLETE_EVT<\r><\n>
D (10789) BLEClient: >> connect(c4:be:84:71:39:84)<\r><\n>
D (10790) BLEUtils: [status: 0]<\r><\n>
V (10795) FreeRTOS: Semaphore taking: name: RegEvt (0x3ffd63e0), owner: <N/A> for connect<\r><\n>
V (10806) FreeRTOS: Semaphore taken:  name: RegEvt (0x3ffd63e0), owner: connect<\r><\n>
V (10813) FreeRTOS: Semaphore waiting: name: RegEvt (0x3ffd63e0), owner: connect for connect<\r><\n>
D (10813) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT<\r><\n>
D (10828) BLEUtils: GATT Event: ESP_GATTC_REG_EVT<\r><\n>
D (10832) BLEUtils: [status: ESP_GATT_OK, app_id: 0x0]<\r><\n>
V (10837) FreeRTOS: Semaphore released: name: RegEvt (0x3ffd63e0), owner: connect<\r><\n>
V (10837) FreeRTOS: Semaphore giving: name: RegEvt (0x3ffd63e0), owner: connect<\r><\n>
V (10844) FreeRTOS: Semaphore taking: name: OpenEvt (0x3ffd62a4), owner: <N/A> for connect<\r><\n>
V (10859) FreeRTOS: Semaphore taken:  name: OpenEvt (0x3ffd62a4), owner: connect<\r><\n>
V (10867) FreeRTOS: Semaphore waiting: name: OpenEvt (0x3ffd62a4), owner: connect for connect<\r><\n>
E (14957) BT: l2cble_start_conn_update, the last connection update command still pending.<\r><\n>
E (14958) BT: l2cble_start_conn_update, the last connection update command still pending.<\r><\n>
D (14963) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT<\r><\n>
D (14970) BLEUtils: GATT Event: ESP_GATTC_CONNECT_EVT<\r><\n>
D (14975) BLEUtils: [staus: ESP_GATT_OK, conn_id: 0, remote_bda: c4:be:84:71:39:84]<\r><\n>
[... 35 lines cut ...]
D (16770) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT<\r><\n>
D (16775) BLEUtils: [srvc_id: uuid: f000aa70-0451-4000-b000-000000000000, inst_id: 0 [<Unknown Service>], instanceId: 0x00 conn_id: 0]<\r><\n>
D (16787) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_SEARCH_RES_EVT<\r><\n>
D (16794) BLEUtils: GATT Event: ESP_GATTC_SEARCH_RES_EVT<\r><\n>
D (16799) BLEUtils: [srvc_id: uuid: 0000ffe0-0000-1000-8000-00805f9b34fb, inst_id: 0 [Unknown], instanceId: 0x00 conn_id: 0]<\r><\n>
[... cut ...]
D (16917) BLEUtils: GATT Event: ESP_GATTC_SEARCH_CMPL_EVT<\r><\n>
D (16922) BLEUtils: [status: ESP_GATT_OK, conn_id: 0]<\r><\n>
V (16927) FreeRTOS: Semaphore giving: name: SearchCmplEvt (0x3ffd65b8), owner: getServices<\r><\n>
V (16927) FreeRTOS: Semaphore released: name: SearchCmplEvt (0x3ffd65b8), owner: getServices<\r><\n>
D (16943) BLEClient: << getServices<\r><\n>
Failed to find our service UUID: 0000aa80-0000-1000-8000-00805f9b34fb<\r><\n>
We have failed to connect to the server; there is nothin more we will do.<\r><\n>

pit001, 2017-09-20:
OT: I play the the API and loop through the services, characteristics an values. 
I miss the return of the m_characteristicMap reference in getCharacteristics() similar as in the getServices()

std::map<std::string, BLERemoteService*>* BLEClient::getServices()
void BLERemoteService::getCharacteristics()

I enhanced it to:
std::map<std::string, BLERemoteCharacteristic*>*  BLERemoteService::getCharacteristics()

Peter

nkolban, 2017-09-20:
Looking at your trace, I see that your BLE partner device returns the following list of services:

```
00001800-0000-1000-8000-00805f9b34fb
00001801-0000-1000-8000-00805f9b34fb
0000180a-0000-1000-8000-00805f9b34fb
f000aa00-0451-4000-b000-000000000000
f000aa20-0451-4000-b000-000000000000
f000aa40-0451-4000-b000-000000000000
f000aa80-0451-4000-b000-000000000000
f000aa70-0451-4000-b000-000000000000
0000ffe0-0000-1000-8000-00805f9b34fb
f000aa64-0451-4000-b000-000000000000
f000ac00-0451-4000-b000-000000000000
f000ccc0-0451-4000-b000-000000000000
f000ffc0-0451-4000-b000-000000000000
```

You are looking for a service called:

```
0000aa80-0000-1000-8000-00805f9b34fb
```

Can you post a link to the actual peer device you are using?  What makes you think that it exposes a service called `0000aa80-0000-1000-8000-00805f9b34fb`?  Do you have a data sheet or other documentation?

pit001, 2017-09-20:
Yes, that's the point - unconnected (GAP) it provides only one (and a unique) UUID - different from all athers in the GATT (connected). i din't know why there is in the GAP a service UUID at all. 

Iin the example I use the well known 'Texas Instruments Sensor TAG', but the same pappens with all atoer tested devices on my table (iTAG a kind of Keyfinder - more a an beacon, Flowercare, a humidity plant sensor). 

http://www.ti.com/tool/cc2650stk
www.dx.com/p/418518
https://www.open-homeautomation.com/de/2016/08/23/reverse-engineering-the-mi-plant-sensor/

Peter

nkolban, 2017-09-20:
I have just ordered a couple of CC2650STK devices.  Will await their arrival for testing.

pit001, 2017-09-20:
;) 

Keep in mind - the TI TAg is not in my focus, it's only a test object. 
As a hint - power the device externaly, the battery keeps only hours. There is a pinnconnector behind a plastic plate

pit001, 2017-09-20:
How can I access the remote descriptor and the handle of the remote characteristic ? Is this implemented yet?
Peter

nkolban, 2017-09-20:
Let us review the story.  We have an ESP32 as a BLE client and a "device" as the remote BLE Server.

A BLE device exposes one or more services.
A BLE service exposes one or more characteristics.
A BLE characteristic exposes 0 or more descriptors.

The current implementation of the BLE C++ library for ESP32 does not yet remote device BLE descriptor access ... i.e. When the ESP32 is being a client, we can't yet access descriptors of a characteristics of a service.  While that is not technically hard to do ... I have yet to find a device that actually exposes a descriptor of a characteristic ... and as such I am "demand driven" on what gets attention for implementation.  Do you know that you need to access a descriptor of a characteristics of a service of a remote BLE server device?  If so, can you help me understand what that device exposes in that fashion?

On the subject of handles,  Internally, BLE implementations use "handles" to refer to instances of characteristics and descriptors.  These then manifest themselves at API levels so that when you want to work with (for example), a characteristic instance, you need to know its handle.  The intent of the C++ wrapper library (as opposed to the underlying raw ESP32 BLE APIs) is to attempt to make things simpler.  To that end, concepts like "handles" are hidden from you.  Instead, you can get access to a class instance that "represents" the logical characteristic ... for example ... which is easier to work with:

An instance of a class of type BLERemoteCharacteristic with all its accessor methods or the integer value "0x2a" which might represent a handle to a remote characteristic where you feed that handle value into many other APIs in order to maintain state.

So in your last question ... there were two concepts ... remote BLE descriptors (not yet implemented but keen to see a need/use case) and low level handles to characteristics (modeled out of the C++ libraries).

pit001, 2017-09-20:
Right. One comment first. I'm not a expert ;) I work on this to lern how it work.

BTLE is Low Energy because one part is sleeping the most time, the other ony listening. 
The sleeping part ist often a sensor (Temperature, Hearttracker, ...), battery powered and called Peripheral, the listening part is the central or observer. (At this moment we don't have a connection and we don't speak about Server or Client). We use the GAP (here the A=Access). If the Central like to connect the Peripheral it has to wait (!) for a (weakup and) Broadcast (also Adverticement). Depending the Advertivemet type (more then 7 different), die Central is able to
- see (i.e. beacon),
- responce (i.e ACK), 
- responce and request (short exchange), 
...
- connect (long data exchance with GATT), ... the peripheral. (here the A=Attribute - rearly stupid)
...

Only in the GATT mode we can talk about Server (provide data) and Client (get data). 
Not every time the Peripheral is the Server. And a device can be both (Server and Client) at the same time in one connection. This maybe change the organisation of the API and also the naming of the Examples  of ESP-IDF.

So yes:
A BLE >Server< exposes one or more services.
A BLE service exposes one or more characteristics.
A BLE characteristic exposes 0 or more descriptors.
...
To find a device that exposes a descriptor - the TI Sensor TAG does this as an example. As a hint if the tool is not known - I check all the things with the Nordic Android App 'nRF Connect' (blue icon with a 2 pieces chain). 

Hmm, I can't post pictures here ?!

The descriptor describe a characteristic value with datatype (~unit) and string "Key press state".  

i.e. with string
[Service] Handle=0x005d Type=0xffe0
    [Characteristic] Handle=0x005e ValueHandle=0x005f Type=0xffe1 Properties=(Notify)
        [Descriptor]  Handle=0x0061 Type=0x2901(User Description)
            [Value] Key Press State
			
or with properties 'readable', could be important :)
[Service] Handle=0x0010 Type=0x180a(Device Information)
    [Characteristic] Handle=0x0011 ValueHandle=0x0012 Type=0x2a23(System ID) Properties=(Read)
        [Value] [FF306400004C99B4]

more: https://developer.microsoft.com/en-us/windows/iot/samples/blegattdump

For me it's interessting to scan all possible things, I try to unterstand BLE and this is the practical part. The standard documentation has 2400 pages ;) Maybe my first ESP32 sketch will be a BLE scanner. 

handles: They are interessting, because they are also used in the communication to reduce the need to transfere the compleate UUID. So, if you sniff a packet i.e with wireshark you only see the handle. Also often in the documentation or on webpages, in other tools and programms only the handles are printed.

hope it helps, its my view, no guaranty
Peter

nkolban, 2017-09-20:
Action: Wait for delivery of TI SensorTag device for testing same technology as issue raiser.

nkolban, 2017-09-23:
I have received a TI SensorTag.

nkolban, 2017-09-27:
I have the test and got working the TI SensorTag ... we we do a summary reset on this issue?  Since we both now have the SensorTag, lets see if we can't get to grips with a usage scenario we can both test.   Unfortunately Arduino-ESP32 with BLE C++ is currently broken while we wait on a fix for a toolchain issue ... but in the interim, if you still have a working Arduino environment because you haven't upgraded for a while ... lets pick just one of the puzzles and take it apart.  Let me know which one you would like to look at.

pit001, 2017-09-27:
Hello Neil. 
On my side I'm also working on this topic  - more in the use and my unterstanding... :-)
Yesterday I had a lot of problems and so I deceide to update all to the newst version (IDF and C++ API). 
I had do some mods to add my changes but it's compiling and some of my problems seems solved. 
I try yet to scan (discover) al from a peripheral (starting with the GATT of a server) and i get the following output (Humidity sensor):

[image]

I also want to add the values and descriptors. It would be great so get your support at this points, similar as at getAllServices I like to have the same methods for Characteristics (done) , Descriptors and Values. 
Also it would be nice to print the different UUID lenght types and the handles. 
If it is working well mybee you would like to add it to your examples.

Back to topic: I would say the crash is solved and it was not the main problem. More: " It's a good way to compare the UUID of a Service in GAP? Alternatives: Device UUID (MAC) or Name. It's IMO more commen, easy to understand and short to type in.

Peter

nkolban, 2017-09-27:
It sounds like the immediate puzzle is solved ... maybe we can close out this issue?  Remember, we can always create new ones or re-open an old one if needed.

The browser based Tree View is interesting me though.  There is a project called "ESP32_Explorer" (see https://github.com/nkolban/ESP32_Explorer) and it is being augmented to provide BLE visualization and editing.  Would that also be useful to your tests?

skyler14, 2017-09-28:
Hi Neil,

Is this related to the error I'm attaching below.  Tried to add BLE support
to another system about an hour ago and got this stacktrace:

C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:
In member function 'void
BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, esp_gatt_if_t,
esp_ble_gattc_cb_param_t*)':

C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:126:78:
error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no
member named 'srvc_id'

    BLERemoteService *pBLERemoteService =
getService(BLEUUID(evtParam->notify.srvc_id.id.uuid));

                                                             ^

In file included from
C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:128:104:
error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no
member named 'srvc_id'

     ESP_LOGE(LOG_TAG, "Could not find service with UUID %s for
notification",
BLEUUID(evtParam->notify.srvc_id.id.uuid).toString().c_str());

                           ^

C:\Program Files
(x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168:
note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >=
ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E,
format), esp_log_timestamp(), tag, ##__VA_ARGS__); }

         ^

C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:131:118:
error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no
member named 'char_id'

    BLERemoteCharacteristic* pBLERemoteCharacteristic =
pBLERemoteService->getCharacteristic(BLEUUID(evtParam->notify.char_id.uuid));

            ^

In file included from
C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:9:0:

C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:133:111:
error: 'struct esp_ble_gattc_cb_param_t::gattc_notify_evt_param' has no
member named 'char_id'

     ESP_LOGE(LOG_TAG, "Could not find characteristic with UUID %s for
notification", BLEUUID(evtParam->notify.char_id.uuid).toString().c_str());

^

C:\Program Files
(x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/log/esp_log.h:239:168:
note: in definition of macro 'ESP_LOGE'

 #define ESP_LOGE( tag, format, ... )  if (LOG_LOCAL_LEVEL >=
ESP_LOG_ERROR)   { esp_log_write(ESP_LOG_ERROR,   tag, LOG_FORMAT(E,
format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
[... 136 lines cut ...]
  ^

C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:29:2:
note:   no known conversion for argument 1 from 'esp_gatt_id_t' to
'esp_gatt_srvc_id_t'

C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7:
note: candidate: BLERemoteService::BLERemoteService(const BLERemoteService&)

 class BLERemoteService {

       ^

C:\Users\hansmalab\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLERemoteService.h:27:7:
note:   candidate expects 1 argument, 2 provided

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with

"Show verbose output during compilation"

option enabled in File -> Preferences.

chegewara, 2017-09-28:
Hi,
ive got the same errors. To be sure, is it on arduino with old toolchain and newest Kolban's cpp_utils?

nkolban, 2017-09-29:
Unfortunately this is related to #82 

We are awaiting toolchain fixes from Espressif which should be available in a couple of weeks.

pit001, 2017-09-29:
Hi Neil,
here a bug summery of this thread and more:

1) Typo: advertisedDevice.getApperance()); // Appearance ?
2) Change the Example to find a device by mac or name, not by serivce ID
3) Improve the writeValue() to self check if the Characteristic exists and if the device is connected.
4) BLEUtils.cpp place the large Strings (members,services) into the PROGMEN, move the Structs into seperate files.
5) Enhance the getCharacteristics() to also return a pointer to the m_characteristic map like getServices()
6) Add the getDescriptors() , ... finctionality
7) Add a gethandle function (no use in other calls needed, only to print out informal)
8) Order of #include <WiFi.h> and #include "BLEDevice.h" (working), (BLEDevice WiFi - not working - 'WiFiServer' does not name a type)

If you agree, I can create seperate issues for each ?
Peter

nkolban, 2017-09-29:
Howdy my friend ... I'm a big fan of separate trackable issues per task ... that allows us to not miss somethign and makes these more readable ... however ...  I do NOT want to put the burden of work on you (my customer/users).  As such, you decide.  My **preference** would be separate issues for each one ... 

Weekend is coming so many hours to work on issues.

pit001, 2017-09-29:
As you want ;)

Btw: the Editor looks fine ! Great to assess the ESP32 before using in own projects. 
Here is the state of the BLE Scanner:

[image]

nkolban, 2017-09-29:
The editor does look great.  Is this part of a project you will be releasing?

pit001, 2017-10-04:
If you like you can provide is a one of your examples or integrate it into your ESP32 Editor.
But in the moment, a lot of problems has to be solved :(
1) BLE things
2) BLE + Webserver work together
3) Upload data to Filesystem
4) ...
5) get enough coffee

chegewara, 2017-10-30:
@pit001 Any progress with this project?

pit001, 2017-11-06:
HI - I have learned that I have to investigate a lot of time (21 days) to realize BLE things with the yet version of API. To solve my actual functional need I switch to RPI with Bluez and Python and get it running in 2 days (and stud in vacation for 3 weeks ;) ) .  Here I will continue if Wi-Fi and BLE will clean work together in the Arduino enviroment.

chegewara, 2017-11-06:
I think it is possible with optimizations. I didnt check it with arduino, but ESP32Explorer is working on esp-idf. Since its pretty big app, and we made some optimizations to free ram which was hell issue, when you feel you are ready you can try.

Ive been asking because i like your services tree and been trying to do similar with jstree:
https://github.com/nkolban/ESP32_Explorer/issues/7#issuecomment-341195453

closed 2018-11-13 by chegewara