nkolban/esp32-snippets #69: Add support for BLE Security

nkolban, 2017-09-19:
Currently, the BLE libraries have no support for BLE security.  This hasn't been an issue but we are starting to see some users present traces where the BLE partner is attempting to get credentials.

sglvladi, 2017-09-26:
Hi, as I am new to BLE and bluetooth in general I was wondering if what you are referring herein is BLE Secure pairing?

Looking at the library and documentation, I cannot see any functions that deal with such a case. Any pointers to how this can be implemented?

nkolban, 2017-09-26:
Unfortunately, we have added **nothing** in the way of security handling to the library yet.  We are looking for user stories that involve security so that we can satisfy those.  In the meantime, we are reading about *theoretical* BLE security and will try and implement those use cases in the interim.

sglvladi, 2017-09-27:
Thanks for the swift response! If you are looking for use cases, I guess a simple starting point would be to add a functionality to pair two devices (i.e. a server and a client) given a pre-shared key. Not sure how easy or hard this would be,  but would provide an adequate initial security measure for most users. I for one would be very interested in one :)

ayavilevich, 2017-10-23:
Hi, I have noticed that ESP-IDF does mention security functions:
esp_ble_gap_set_security_param
esp_ble_gap_security_rsp
esp_ble_set_encryption
esp_ble_passkey_reply
esp_ble_confirm_reply

http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv230esp_ble_gap_set_security_param18esp_ble_sm_param_tPv7uint8_t

Is this available but just not implemented in the c++ library?

nkolban, 2017-10-23:
The BLE functions supplied in ESP-IDF do indeed contain a rich set of security capabilities.  These have not been mapped into the C++ library (yet).  Part of my puzzle is coming up with a test environment that would use security.  For example, if ESP32 were a BLE client, what would the security look like to a BLE server?  While it is pretty obvious that security in BLE land is important ... I haven't come across a test scenario which has said "This is what I want to achieve using the ESP32 BLE C++ classes and I can't".   If I have a use case that I can recreate the environment upon here, then I can start the task of building out the wrappers for the C++ classes.  My skills on BLE are not brilliant and I tend to learn best when I have a concrete puzzle to solve.  If we can work together to build out a use case that I can work on here, that would be ideal.

ayavilevich, 2017-10-24:
Hi Neil, happy to help. Regarding use cases, here are two with ESP32 being the BLE server:
1: IoT light switch with ESP32 as BLE server. You don't want anybody to connect to your switch and play with the lights. A good process would be to ask for a pin/pass on first attempt. If the client can provide the correct pin then establish a secure connection from now on between the two devices.
2: I have noticed in the past that some client phone devices insist on pin pairing on connecting. Happened to me with an HM-10 and a Samsung S4. If that doesn't work, they will disconnect. Perhaps it is an implementation issue. If the ESP32 can't confirm the pin or convince the client not to bother with security, then connection won't work, even though you might not be interested in security in the first place.

LMK if you have additional questions. At a later time I could look at supporting these issues. Currently dealing with more basic stuff. If you don't have the time or priority for this, we can leave it for later.

ayavilevich, 2017-10-24:
Correction, issue number 2, where pairing was required, happened to me on LG G5. It is a newer Android. Version 7 is I am not mistaken.

nkolban, 2017-12-12:
See also #260

nkolban, 2017-12-12:
Mr @chegewara has found a great article on BLE security ... logging here so we can study it:

https://eewiki.net/display/Wireless/A+Basic+Introduction+to+BLE+Security

chegewara, 2017-12-12:
From official bluetooth document about security we can learn that:

[image]
[image]

This mean that security can be implemented in esp32 only if esp32 have some sort of display and/or keyboard to display and/or input pin code to pair devices.

chegewara, 2017-12-12:
Its a lot to study:
https://blog.bluetooth.com/bluetooth-pairing-part-1-pairing-feature-exchange
https://blog.bluetooth.com/bluetooth-pairing-part-2-key-generation-methods
https://blog.bluetooth.com/bluetooth-pairing-passkey-entry?_ga=2.237580416.167769799.1512932724-1073796007.1512932724
https://blog.bluetooth.com/bluetooth-pairing-part-4
https://blog.bluetooth.com/bluetooth-pairing-part-5-legacy-pairing-out-of-band

chegewara, 2017-12-13:
I think this may be good case study example
https://github.com/kmackay/micro-ecc

ayavilevich, 2017-12-13:
Hi @chegewara, that example only shows pairing with no IO options. Which means that there is no authentication.
There are no explanations and no example of how to do something more secure with a passkey or buttons or a display. We really need more/better information from espressif.

chegewara, 2017-12-13:
I mean its good study case to me and mr @nkolban

chegewara, 2017-12-13:
And here is capability to use IO option
https://github.com/espressif/esp-idf/blob/bd6ee752fcf074f2fc681c001a329300c8a180d1/examples/bluetooth/gatt_security_client/main/example_ble_sec_gattc_demo.c#L368

chegewara, 2017-12-13:
@ayavilevich This is what i discovered as for now. If you will change this line https://github.com/espressif/esp-idf/blob/bd6ee752fcf074f2fc681c001a329300c8a180d1/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L521

with this code: `esp_ble_io_cap_t iocap = ESP_IO_CAP_OUT;` you will have security capability with pin code. Pin code will be generated and outputed to serial (just in this example), but you can easy display pin on any display connected to esp32.

```
I (0) cpu_start: App cpu up.
I (383) heap_init: Initializing. RAM available for dynamic allocation:
I (390) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (396) heap_init: At 3FFCAAB8 len 00015548 (85 KiB): DRAM
I (402) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (409) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (415) heap_init: At 400927E0 len 0000D820 (54 KiB): IRAM
I (421) cpu_start: Pro cpu start user code
I (104) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (146) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (406) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (406) SEC_GATTS_DEMO: app_main init bluetooth
I (496) SEC_GATTS_DEMO: The number handle = 8
I (506) SEC_GATTS_DEMO: advertising start success
I (40736) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
E (40896) SEC_GATTS_DEMO: The passkey Notify number:186182
I (56816) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC
I (56816) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID
I (56816) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LCSRK
I (56856) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
I (56886) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
I (56886) SEC_GATTS_DEMO: remote BD_ADDR: 781fdbc0c7f3
I (56886) SEC_GATTS_DEMO: address type = 0
I (56886) SEC_GATTS_DEMO: pair status = success
I (56896) SEC_GATTS_DEMO: Bonded devices number : 1

I (56906) SEC_GATTS_DEMO: Bonded devices list : 1

I (56906) SEC_GATTS_DEMO: 78 1f db c0 c7 f3 
I (74196) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
I (74206) SEC_GATTS_DEMO: advertising start success

```

If you need more assistance with example code just ask and i will try to explain all

ayavilevich, 2017-12-13:
@chegewara, nice. 
Can I set my own passkey per device and have the other side enter it? Seems like in the example the system decides what the passkey is for you.
Do you know what the difference is between modes "Display only" and "Display yes/no"?

chegewara, 2017-12-13:
This is how it works, but ive read somwhere, not related to esp-idf, that its possible to use static passkey, which is not as secure as random passkey of course. Maybe OOB will help with this, but requires more study.

Difference is simple:
- "Display only" - its only display and thare is no keyboard to input anything
- "Display yes/no" - its display and you have 2 keys to chose yes or no, or any other input option to chose and send yes or no

ayavilevich, 2017-12-13:
Seems to me a static key would be better than no authentication at all. Thanks for your research.

chegewara, 2017-12-13:
@ayavilevich It is possible to use "Display only" option. Display does not need to be classic display device. It just mean you have an option to visual presentation numbers, like with serial monitor or even with blinking LED

chegewara, 2018-02-15:
We have added security to library, maybe its time to close this issue?

closed 2018-11-13 by chegewara

guysmilez98, 2019-06-14:
@nkolban I have a very specific case to use a passkey with the esp32 where should I post this? Here?

xharkx, 2019-06-23:
@guysmilez98 that's what I want.
I try to create bluetooth controller for BMPCC4k, but I don't have knowledge to create.
So, I'm really happy, if you share you code, thank you.