nkolban/esp32-snippets #73: WebServer functions getMethod, getPath are kinda slow

Eloquence4, 2017-09-20:
The way they work right now is that they use the function mgStrToString to a string. Problem with that is, that it creates a bunch of unnecessary copying (the mongoose char* string to the std::string string) and it takes double the memory. My solution would be to just return another structure containing the length and char*, or split it to 2 functions getMethodStr and getMethodLen and the same for getPath.

There might be other similar functions. I'll be changing all of this for my code, i'm just wondering if you would like me to create a pull request for this, or if your argument would be that the getMethod and getPath strings are fairly small and will not take a lot of space or processing power anyway.

I'll make a pull request with the changes if you agree with me (asking because i have a loot of changes, and making pull requests for all of them will be a bit difficult as i'm not the most experienced git user)

nkolban, 2017-09-20:
Howdy my friend.

Memory utilization within the ESP32 ... especially for HTTP is indeed very important.  If you wish to perform surgery on the WebServer classes ... I have absolutely no issue with that and I'll keep my hands off completely.  If you are changing WebServer.h and WebServer.cpp I'll 100% leave those along until you tell me otherwise ... that way there won't be any form of a Git conflict.  If you make massive changes then before you perform the Pull Request ... I'll be delighted to work with you (if needed) to make sure that the Git merge is safe and clean.  Worst case, before you perform the Pull Request, take a complete backup of the source as it stands ... that way we won't lose anything and thus we can always correct.  The only thing I ask before any pull request is that the whole of cpp_utils compiles cleanly ... other than that, have at it.   You'll hopefully find me very friendly so if ever you want to discuss something, email, post here or ping on IRC.

My contract to you is simple ... if you issue a Pull Request then it will be merged ASAP.   If there are problems, I'll work with you immediately.  Too many open source projects make it hard for others to partner or collaborate ... I want to avoid that.

Neil

Eloquence4, 2017-09-20:
Alright, i'm fine with that.

And ya, i make sure everything compiles because i am testing the code i am writing in real time, thanks for the trust! And GIT automatically saves all of the previous revisions of the codes, you can always go back a few commits and revert the code, or just revert the commits

nkolban, 2017-09-20:
I forget to mention in my last post ... I'm tinkering with a whole new HttpServer story ... see the classes ... 

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/HttpServer.cpp

This is very much a work in progress.  But what it will be is *yet another* HTTP server.  This one though is written in (poor) C++ and tries to leverage the C++ methodologies.   For example, rather than provide memory buffers of the received HTTP content, it provides C++ std::stream handlers and leverages callbacks that inherit from abstract classes for indications.  I point it out just as an FYI.

Eloquence4, 2017-09-20:
I'll check it out, i'm not very familiar with the std::streams, i only know what the conceptually do. Thanks

nkolban, 2017-10-04:
Closing for just now.  Don't hesitate to re-open as needed.

closed 2017-10-04 by nkolban