nkolban/esp32-snippets #75: Latest zip file (and windows support)

skyler14, 2017-09-20:
Hi, just wanted to say I'm very interested in the work on your project. I couldn't help but notice there was a precompiled zip file, but I don't currently see the zip file anywhere in the newest commit. Since I'm running on a windows system I can't do the compilation atm. Can you provide the latest stable-ish release for the zip? If I'm just derping and there is a zip directory pointing me in that direction is also great. Thanks and good luck.

nkolban, 2017-09-20:
Howdy,
Is this for the Arduino BLE Library?  You didn't mention what you were referencing in the last post.

Assuming it is, the library can now be found directly in the Arduino IDE.  If you visit the Library manager and search for ESP32 BLE ... you will find that you can install the library from the Internet.

skyler14, 2017-09-20:
Thanks for the blazing fast response. This is for Arduino. Library accounted for, thanks!

nkolban, 2017-09-22:
When you are satisfied with the answer to the question, go ahead and close.  We can always re-open or create a new ticket as needed.

nkolban, 2017-10-06:
Closing for just now.  Don't hesitate to re-open as needed should the need arise.

closed 2017-10-06 by nkolban