nkolban/esp32-snippets #76: not workong on ios

ahachicha, 2017-09-21:
Hello Kolban,
it's not working on mac ios. i get this errors:

```
Arduino: 1.8.4 (Mac OS X), Board: "ESP32 Dev Module, 80MHz, 921600, None"

/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp: In static member function 'static std::__cxx11::string BLEUtils::gattClientEventTypeToString(esp_gattc_cb_event_t)':
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:344:8: error: 'ESP_GATTC_CONNECT_EVT' was not declared in this scope
   case ESP_GATTC_CONNECT_EVT:
        ^
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:346:8: error: 'ESP_GATTC_DISCONNECT_EVT' was not declared in this scope
   case ESP_GATTC_DISCONNECT_EVT:
        ^
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp: In static member function 'static void BLEUtils::dumpGattClientEvent(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:684:8: error: 'ESP_GATTC_CONNECT_EVT' was not declared in this scope
   case ESP_GATTC_CONNECT_EVT: {
        ^
In file included from /Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:22:0:
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:686:44: error: 'union esp_ble_gattc_cb_param_t' has no member named 'connect'
     BLEUtils::gattStatusToString(evtParam->connect.status).c_str(),
                                            ^
/Users/ahachicha/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:687:15: error: 'union esp_ble_gattc_cb_param_t' has no member named 'connect'
     evtParam->connect.conn_id,
               ^
/Users/ahachicha/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:688:26: error: 'union esp_ble_gattc_cb_param_t' has no member named 'connect'
     BLEAddress(evtParam->connect.remote_bda).toString().c_str()
                          ^
/Users/ahachicha/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:700:8: error: 'ESP_GATTC_DISCONNECT_EVT' was not declared in this scope
   case ESP_GATTC_DISCONNECT_EVT: {
        ^
In file included from /Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:22:0:
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:702:44: error: 'union esp_ble_gattc_cb_param_t' has no member named 'disconnect'
     BLEUtils::gattStatusToString(evtParam->disconnect.status).c_str(),
                                            ^
/Users/ahachicha/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:703:15: error: 'union esp_ble_gattc_cb_param_t' has no member named 'disconnect'
     evtParam->disconnect.conn_id,
               ^
/Users/ahachicha/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
/Users/ahachicha/Documents/Arduino/libraries/ESP32_BLE_Arduino/src/BLEUtils.cpp:704:26: error: 'union esp_ble_gattc_cb_param_t' has no member named 'disconnect'
     BLEAddress(evtParam->disconnect.remote_bda).toString().c_str()
                          ^
/Users/ahachicha/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log/esp_log.h:144:168: note: in definition of macro 'ESP_LOGD'
 #define ESP_LOGD( tag, format, ... )  if (LOG_LOCAL_LEVEL >= ESP_LOG_DEBUG)   { esp_log_write(ESP_LOG_DEBUG,   tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); }
                                                                                                                                                                        ^
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
 
```

I have the last version from Arduino and from the library

nkolban, 2017-09-21:
Howdy,
The good news is that we have seen this problem before ... can I ask you to look at this issue and resolution:

https://github.com/nkolban/esp32-snippets/issues/59

Can you upgrade your Arduino ESP32 base within your Arduino IDE and test again.  Hopefully that will resolve the problem ... but either way, hoping to hear back from you.

pdxalz, 2017-09-23:
I am also unable to get the BLE lib to work.  I've tried both Windows 10 and Ubuntu.  Everything was installed fresh today.  Arduino version 1.8.4.  I am able to run the wifi scan app just fine on the esp32.  
I get build errors when I build the BLE_Client app (same on windows and ubuntu): [https://pastebin.com/hQ7JDxXp](url)

On Windows I used the library manager and found version 0.4.2 (I tried all of them with the same result)
On Ubuntu I built the .zip file and used the install zip feature.

Alan

nkolban, 2017-09-23:
@pdxalz 
See issue #82 

The latest ESP-IDF has introduced a breaking change to the APIs used by the BLE C++ classes.  This is being addressed and tracked as #82.

nkolban, 2017-10-14:
The great news is that the Arduino ESP32 project has been updated.  Please re-build your base Arduino ESP32 environment.  The even better news is that the owner of that project has decided to include the BLE C++ classes as part of the distribution.  So it is now baked in.  Please re-test and let us now see where we stand.

nkolban, 2017-10-29:
Closing for just now.  Re-open as needed.

closed 2017-10-29 by nkolban