nkolban/esp32-snippets #77: Mac OS build issues

hosek, 2017-09-21:
Hi I had problem with building it on my Sierra 10.12.6
`cp --R tests/BLETests/Arduino Arduino/ESP32_BLE/examples
cp: illegal option -- -
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvX] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvX] source_file ... target_directory
make: *** [build_ble] Error 64
`
As error is self explanatory, edit 
Makefile.arduino:62 to
`	cp -R tests/BLETests/Arduino Arduino/ESP32_BLE/examples`

Helps, just mention it there for others if they hit same issue (I was not able to find this in issues, sorry if its duplicate)

nkolban, 2017-09-21:
Howdy,
MANY thanks for reporting this.  The original source line was:

```
cp --recursive tests/BLETests/Arduino Arduino/ESP32_BLE/examples
```
however it appears that the following:
```
cp -R tests/BLETests/Arduino Arduino/ESP32_BLE/examples
```
is also allowed on Linux/Unix ... am changing to that so that we will be consistent.

nkolban, 2017-09-21:
Change made.

closed 2017-09-21 by hosek