nkolban/esp32-snippets #78: Example code for running a specific service on the BLE server

JabberwockPL, 2017-09-21:
I am running the examples without problems, but I have a problem moving on. I would like ESP32 to run a specific BLE service (bike speed/cadence) and cannot even get the clients to recognize the service correctly... I am sending the service UUID as a char array with two values (18 and 16 for the UUID 0x1816) and it does not work. I am afraid that I will understand even less about sending specific values...

Is there an example with code that runs a specific GATT service as described in BT specifications?

nkolban, 2017-09-21:
Howdy my friend.  I'll try and assist as best I can.  First we must determine if you are running the ESP32 as a BLE Client or a BLE Server.  From your notes, I think a BLE Server.  This means that you are going to "expose" one or more services where each service has one or more "characteristics".  Both services and characteristics are specified using UUID (Universal Unique Identifiers).

Let us use this as an example ... https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleRead.cpp

Here we expose a service and a characteristics with their own UUIDs.  You should then be able to use the android nRF Connect app to connect to it and read the value of the characteristic.

So let us start slowly ... can you get this running?  Can you connect to it from nRF Connect?

Following that ... let us then think about what you want to achieve.  Spell out your goal and how you plan to go about it ... and then we can look and see what intermediate steps we can perform to make progress.

JabberwockPL, 2017-09-21:
Thank you for the prompt response! 

Yes, that works. The code I am using is the example code from the Arduino library, which I have tried to modify. However, it is somewhat different, it has:

```
#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
[..]
BLEService *pService = pServer->createService(SERVICE_UUID);
```
(there is no SERVICE_UUID_BIN part there). I have tried to replace it with:

```
char my_service[1];
[..]
my_service[0]=char(18);
my_service[1]=char(16);
[..]
BLEService *pService = pServer->createService(my_service);

```
to send a two-byte string with the service identificator (0x1816), but it is not recognized.

I admit I do not understand it very well, that is why I have asked for more examples to study...

nkolban, 2017-09-21:
A certain number of BLE services and characteristics are "prescribed".  What that means is that they are pre-speced out so that multiple vendors can implement the same type of device and have it accessed the same way.

A UUID is a unique identifier to a service or a characteristic.  A UUID is typically 128 bits of data ... but for some UUIDs, they need only be 16 bits of data.  These are the prescribed UUID.s  When we create a service, we must tell the service what UUID it is going to have.  The way we do that is to call the createService() method passing in a UUID.

There is an API function on BLEServer called `createService(BLEUUID uuid)`.

If we then look at the constructor for a BLEUUID we find that one of its constructors is:

BLEUUID(uint16_t uuid)

This will create a UUID from the 16 bit value.

Putting this together, your code would then become:

```
BLEService *pService = pServer->createService(BLEUUID((uint16_t)0x1816));
```

JabberwockPL, 2017-09-21:
Thank you very much for detailed explanation!

I see, I did not realize there can be many constructors for the same function... Now it makes sense and, more importantly, works - my phone does see the service as the bicycle sensor.

I will try to wrestle with the characteristics and the values on my own for a bit, I hope it will be a good learning opportunity. I might come back though :)

Thank you again for your assistance!

nkolban, 2017-09-21:
Come back anytime.  You might close out this issue if complete and create a new one so that we can work each puzzle as its own story.  I'm always delighted to spend time with users of code I have tinkered with.  Where possible, try and pass it on and help others where you can.

JabberwockPL, 2017-09-21:
Will do that, thanks!

closed 2017-09-21 by JabberwockPL