nkolban/esp32-snippets #79: setValue() argument as a string or a uint16

Tavo7995, 2017-09-21:
Hey so Ive been trying to use the ble uart example to send continuous sensor data to an android app, and I can send uint8 but I wanted to send uint16 so as to make decimal places and make the reading more accurate. I have tried setting the number as a string but it only accepts it if is within double quotations. 
so it will take pCharacteristic->setValue("sensorValue");
but not pCharacteristic->setValue(sensorValue); (sensorValue having been converted into a string earlier)
Is there something I am missing or some way around this? 
I am using the Arduino IDE.
Thanks!

nkolban, 2017-09-21:
The value for a characteristic is a "sequence of bytes".  These bytes (data) are then stored by the BLE Server and made available to a BLE client that requests them.  What the bytes represent is a private contract between the BLE Server and the BLE client.  The API exposed through these BLE classes allows you to set the value of a characteristic.  You can set the value to a sequence of bytes.

In C++, there is no "Buffer" data type for holding arbitrary sequences of bytes ... however, convention holds that we use the std::string data type.  This is the C++ standard data type that holds strings.  However, any sequence of bytes can be stored in a string.  When we create a std::string instance, we code the following:
```
std::string myString(pointerToData, lengthOfData);
```
To have a unit16_t stored  in the string we would thus code:
```
std::string myStringForUnit16((char*)&myUint16, 2);
```
we could then set the characteristic by pasing that string:
```
pCharacteristic->setValue(myStringForUnit16);
```

I'd suggest also grabbing a big cup of coffee and having a read of the std::string data type which you can find by googling.

nkolban, 2017-10-05:
Closing for just now.  Don't hesitate to re-open as needed.

closed 2017-10-05 by nkolban

Girish1389, 2018-04-14:
Hi this is my first post on BLE issue which i am facing.
Kindly help me out.
I am using the ble write example to read sensor data to an android app, whenever i am writing prefered string"sensor data". I am not able to read the sensor data, when i am calling the function return value in the code. kindly help me to solve the issue which i am facing.
Thanking you.
if i take pCharacteristic->setValue("sensorValue")
only able to read the sting not the sensor value in the app.

chegewara, 2018-04-14:
You dont have to set value as string:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.h#L76-L82
unless you are working with arduino-ide then it may be not added yet, then just copy paste BLECharacteristic.cpp and .h files into your library.

Girish1389, 2018-04-16:
Sir I am trying to read the sensor data, by calling the function return value but, I am not able to read the LDR sensor value into my android app. 
if (value=="LDR value")
        {
          pCharacteristic->setValue(LDR());
          LDR();
error: no matching function for call to 'BLECharacteristic::setValue(float)'
           pCharacteristic->setValue(LDR())
kindly suggest me.

Girish1389, 2018-04-17:
Sir I have added the above mentioned BLECharacteristic.h and BLECharacteristic.cpp into my source code folder.
I am attaching my code below.

#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <BLECharacteristic.h>
#define GAP_UUID  "6f3273a0-f831-4b0f-b609-2477208f8c2b"
#define GATT_UUID "0c401a07-3fef-4a77-9efc-36458a21c7ba"

#define SERVICE_UUID        GAP_UUID
#define CHARACTERISTIC_UUID GATT_UUID
int sensorPin = 34; // select the input pin for ldr
int sensorValue = 0;
float LDR(void)
{
float luminosity;
sensorValue = analogRead(sensorPin); // read the value from the sensor
luminosity = (3.3*sensorValue)/4095;
return (luminosity);  
}
class MyCallbacks: public BLECharacteristicCallbacks 
{
    void onWrite(BLECharacteristic *pCharacteristic) 
    {
      std::string value = pCharacteristic->getValue();
      if (value.length() > 0) 
      {
        //Serial.println("*********");
        //Serial.print("New value: ");
        for (int i = 0; i < value.length(); i++)
          Serial.print(value[i]);
        Serial.println();
        }
        if (value=="LDR value")
        {
          pCharacteristic->setValue("LDR()");
          LDR();
          Serial.println(LDR());
        }
      } 
    }       
};
void setup() {
  Serial.begin(115200);
  BLEDevice::init("test");
  BLEServer *pServer = BLEDevice::createServer();

  BLEService *pService = pServer->createService(SERVICE_UUID);

  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pCharacteristic->setCallbacks(new MyCallbacks());
    pService->start();

  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
  pinMode(led,OUTPUT);  pinMode(sensorPin,INPUT);
  LDR();
}
void loop() 
{
  // put your main code here, to run repeatedly:
  delay(2000);
}

I am only able to read in my serial monitor but not in my app.

dhavals1212, 2022-03-25:
> Sir I have added the above mentioned BLECharacteristic.h and BLECharacteristic.cpp into my source code folder. I am attaching my code below.
>
> #include <Arduino.h> #include <BLEDevice.h> #include [...]

Well, you have printed setValue("LDR()"); which I am going to allow you to guess what it means.