nkolban/esp32-snippets #80: Zeroes in the setvalue() string

JabberwockPL, 2017-09-22:
I am back, after all...

I have managed to get the bike sensor service working and it is recognized by various training apps on my phone, which is great. However, I have encountered an issue: everything works fine as long as there are no zeroes in the byte sequence. I suppose that it is because the string is treated in the Arduino code as null-terminated string... I suspect this might be rather a problem of my sketchy knowledge of C than an issue with this particular code, but I have no idea how to get around that. 

The relevant part of the code is:
```
char measure[12] = {0x03, 0x00, 0x00, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01, '\0'};
[..]
measurement->setValue(measure);
```

when I replace 0x00 with 0x01, for example, it works correctly (beside the fact that the distances get absurd values, which confuses the apps).

nkolban, 2017-09-22:
There was a similar post today ... which makes me think the documentation isn't good enough.  Have a read of the following:

https://github.com/nkolban/esp32-snippets/issues/79

A characteristics's value is an array of bytes and not a string 

Using your sample as an example, we would code:

```
char measure[12] = {0x03, 0x00, 0x00, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01, '\0'};
[..]
measurement->setValue(std::string(measure), sizeof(measure));
```
to pass an array of bytes.

JabberwockPL, 2017-09-22:
The first option throws an error at compilation:

C:\Users\mjarek\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLECharacteristic.h:76:7: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'BLECharacteristic::setValue(std::__cxx11::string, unsigned int)'

The second one compiles correctly, but the problem is the same: the sent bytes seem to be cut off at the first zero.

nkolban, 2017-09-22:
Sorry, my friend, I had a bracket in the wrong place ... I had coded:

```
measurement->setValue(std::string(measure), sizeof(measure));
```
as opposed to:

```
measurement->setValue(std::string(measure, sizeof(measure));
```

Can you clarify to me how we know the sent bytes are being cut off?  How are you determining that?  What if we switch on BLE C++ trace ... I'm thinking we should be able to see the size of the data actually sent.

JabberwockPL, 2017-09-22:
That new instruction compiled correctly (well, with one more bracket), but the issue remains.

I am using nRF Connect to test the emulated bike sensor. If I have specific non-zero values, the phone app sees it as correct bike data and shows e.g. the distance taken, wheel rotations, etc. (although the values are very large, as more significant bytes are non-zero).

When I put a zero on one position, e.g.:
`
static char measure[12] = {0x03, 0x01, 0x01, 0x00, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01};
`
the app shows:

Invalid data syntax: (0x) 03-01-01

For other cadence data apps it is the same - they are working (well, some of them, some are crashing due to unrealistic values) with non-zero values and not responding when there is a zero in the data.

nkolban, 2017-09-22:
There is **always** the opportunity that I have messed up implementation.  My saving grace is that I am not hesitant to acknowledge that possibility and will work hard to remedy it.  When a problem is reported, I think it through to see if I can sense the issue by looking at high level code and thinking it through.  When that doesn't work (for me) then I need help to see traces and attempt a recreate.  Do you know how to switch on BLE C++ trace?  From there we will be able to see the sequence of requests and responses send and received by the ESP32.  We might then be able to compare what we desire to be sent/received vs what is actually sent/received.  Obviously I don't want you to go to do un-necessary work or do more than you have to but by the same token, I need information against which to do my part.

Do you have a self contained Arduino script that illustrates the issue?  Ideally it should contain the minimum to recreate the story.  If we need, we can also do live discussions but they would have to be my evenings or weekends (US Central Time) as I work at a Job other times.

JabberwockPL, 2017-09-22:
Could it be that Arduino implementation is working somehow differently, in less binary-safe way?

I will look into BLE trace and provide you the relevant info. For now the code is this:

```
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

BLEDevice ble;
BLECharacteristic *measurement;
bool deviceConnected = false;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

static char feat[2] = {0x03, 0x00};
static char measure[12] = {0x03, 0x01, 0x01, 0x00, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01};
uint8_t pos = 6;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

      ble.init("MyESP32");
  BLEServer *pServer = new BLEServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEService *pService = pServer->createService(BLEUUID((uint16_t)0x1816));
  BLECharacteristic *feature = pService->createCharacteristic(
                                         BLEUUID((uint16_t)0x2A5C),
                                         BLECharacteristic::PROPERTY_READ
                                       );

  BLECharacteristic *location = pService->createCharacteristic(
                                         BLEUUID((uint16_t)0x2A5D),
                                         BLECharacteristic::PROPERTY_READ
                                       );

  BLECharacteristic *controlpoint = pService->createCharacteristic(
                                         BLEUUID((uint16_t) 0x2A55),
                                         BLECharacteristic::PROPERTY_WRITE |
                                         BLECharacteristic::PROPERTY_INDICATE
                                         );

  measurement = pService->createCharacteristic(
                                         BLEUUID((uint16_t) 0x2A5B),
                                         BLECharacteristic::PROPERTY_READ
                                       );

  feature->setValue((uint8_t*)feat, sizeof(feat));
  location->setValue(&pos, 1);
  measurement->setValue(std::string(measure, sizeof(measure)));
  
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  if (deviceConnected) {
    measure[10]= measure[10]+1;
    measure[8]= measure[8]+1;
    measure[4]= measure[4]+1;
    measure[6]= measure[6]+1;
    measurement->setValue(measure);
    measurement->notify();
  }
  delay(2000);
}
```

Basically, it is the examples of server and notify merged together with some addtions specific for the bike sensor.

JabberwockPL, 2017-09-22:
I have just realized that the above code makes little sense, as I have changed the measurement characteristic from NOTIFY to READ for debugging purposes (as nRF Connect does not seem to have a way to read notifications?). But the change does not affect the issue in question.

Here is the BLE trace when measurement is set to NOTIFY:

https://pastebin.com/MrX2kmsf

The setValue does seem truncated at three bytes...

chegewara, 2017-09-22:
Actually nRF has option to read notifications and it works really good. Its 3 arrows icon in characteristic, only you have to do is to add 0x2902 descriptor to characteristic which has notifications.

chegewara, 2017-09-22:
@JabberwockPL one suggestion if i can, 
`void loop() {
  // put your main code here, to run repeatedly:
  if (deviceConnected) {
    measure[10]= measure[10]+1;
    measure[8]= measure[8]+1;
    measure[4]= measure[4]+1;
    measure[6]= measure[6]+1;
    measurement->setValue(measure);
    measurement->notify();
  }
  delay(2000);
}`
try this to set value:
` measurement->setValue(std::string(measure, sizeof(measure)));`

From logs you show us it seems to work even with 0x00 value
`(6049) BLECharacteristic: >> setValue: length=12, data=030101000101010101010100, characteristic UUID=00002a5b-0000-1000-8000-00805f9b34fb`

JabberwockPL, 2017-09-22:
Yes, you are right, I feel like an idiot... I have changed the setvalue call in the setup according to Neil's instructions, but I forgot about the main loop one!

Now it works beatifully, I can have my bike trainer/race gamer report the distance as I wanted.

Thank you very much to both of you!

closed 2017-09-22 by JabberwockPL

nkolban, 2017-09-22:
Thank **you** Mr @JabberwockPL for getting back to us and letting us know that things are good.  Issues are cheap and easy, so feel free to raise as needed.

@chegewara Thank you sir for your assistance on the issue.  Awesome detective work!