nkolban/esp32-snippets #88: Setting the appearance characteristic of the device

JabberwockPL, 2017-09-26:
The code for the cadence/speed sensor is almost done, however, its application is hit-and-miss. One particular app that has worked before now refuses to do so, disconnecting with cryptic messages... Some apps do not recognize the sensor at all. On the other hand, some other apps do and they show the correct inputs. nRF Connect is always solid, showing exactly what I need (but the apps themselves are bad).

I have even downloaded the nRF SDK, as they have an example code for cadence/speed. The code does not look fundamentally different from mine (well, mostly yours :) ), but there is one notable difference: they set the appearance of the device to Speed and Cadence Sensor.

Naturally, I wanted to do the same, but it seems the characteristic is not exposed at all in the library - the device name is set with .init and that is about it. Is there any way I can set it?

nkolban, 2017-09-26:
Howdy,
The bad news is that the ESP-IDF master on Git changed significantly.  My guess is Espressif are getting closer to releasing their 3.0.  Two major areas changed.  First, the BLE API exposed by ESP-IDF changed in breaking ways which had to be resolved.  Following that, it was found that the toolchain (the C++ compiler) now failed to generate "good code" for execution.  See issue #82.

More than that, the Arduino-ESP32 library opts to build against the latest ESP-IDF and that means that the breakages in ESP-IDF have flowed through to breakages in Arduino-ESP32.

That said ... that isn't the issue you are discussing.

Help me to help you ... assume that I am juggling so many parallel stories that I can't keep context straight in my head.  Please try and make each issue as self contained as possible or, at a minimum, point to other issues that I should pre-read before reading the current one.

Can you clarify whether or not your ESP32 BLE app is a BLE Server or a BLE Client.   I am going to assume it is a BLE Server that advertises its presence.

By setting the appearance, can I assume that you mean exposing this characteristic?

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.appearance.xml

If yes, then my immediate guess is that you can create a BLE Server that contains a service that contains a characteristic with 16bit UUID 0x2A01 as documented and described above.  What you would do is create and expose a new characteristic.  Read the specification of that characteristic and see if it makes sense.

If after reading that and the BLE C++ documentation we make available it is still a puzzle ... then I *really* want to hear back.  If something is confusing, then I want to try and resolve that confusion in the documentation so that both you and future readers/users can understand.  If something isn't clear, then I consider that my problem to fix.

JabberwockPL, 2017-09-26:
I understand you have a lot to deal with, so I appreciate your help even more!

I have read the specs and considered creating the characteristic (yes, it is a BLE Server, sorry for not being more specific). I just thought that if I create it from pService, which in my case is cycling speed/cadence sensor, then it appears under that service and the Generic Access characterstic of Appearance would remain unaffected. I have just tested it and it is indeed the case - new Appearance shows up under the cycling service and the Generic Access Appearance remains Uknown.

And I do not know how I can create it from the Generic Access service (0x1800), as it does seem to be exposed in the code. If it is accessible, I do not know how to do that. In the nRF SDK the characteristic is set at the creation of the device, together with its name and connection parameters. In your BLEDevice.cpp it seems only the name can be assigned.

chegewara, 2017-09-26:
Can you paste link to nRF code you reference to?

JabberwockPL, 2017-09-26:
I have downloaded the nRF SDK, but the code is online here:
http://storage.tokor.org/pub/nrf51/doc/html/nrf6310_2s110_2ble__app__cscs_2main_8c_source.html

searching for 'appearance' should get you to the relevant part of the code.

nkolban, 2017-09-26:
@JabberwockPL Ohhh!!  I think I see what you are saying.   Let me play it back to you and see if this sound right ...

We have a service type with UUID 0x1800 called "Generic Access" .. see:

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.generic_access.xml

This service is a prescribed service and can have a variety of characteristic values including 0x2A01 (Appearance).   How do I use the BLE C++ libraries in order to expose characteristic 0x2A01 as belonging to service 0x1800?

----

Does that sound about right?  I don't have an immediate answer for that and will have to study to assist.

JabberwockPL, 2017-09-26:
Well, actually if I do not create the characteristic at all, nRF already shows that the Generic Access service has that characteristic, it only shows in the log  „'[0] Unknown' received” and the characterstic is visible under the service as '[0] Unknown'. But of course I would like to set to according to my needs.

chegewara, 2017-09-26:
Yes, but you can't add any characteristic to generic access 0x1800 and generic attributes 0x1801 service. You need to create service as you can see at 
> Step 4: Add our service
and i dont think you can add name to those services, but i will investigate it further.

JabberwockPL, 2017-09-26:
But  I do not want to add my own chatacteristic, it seems Appearance is created anyway at the device creation (as nRF Connect shows). I just want to set it, just like you allow setting the name of the device (characteristic 0x2A00) with BLEDevice init(). I cannot figure out from your code _how_ exactly you set the name (I.e. 0x2A00) at the device creation, but I guess the method (and time and place) should be the same for the appearance.

chegewara, 2017-09-26:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L191	
errRc = ::esp_ble_gap_set_device_name(deviceName.c_str());

Mail me and we can talk in Polish, imperiaonline4@gmail.com.

chegewara, 2017-09-26:
https://devzone.nordicsemi.com/question/52197/how-to-change-the-service-name/

nkolban, 2017-09-26:
Ooooh ... I had made a mental note to go research how "appearance" was going to be set system wide but it looks like you may have found the answer (you are AWESOME).  Are we saying that setting the device name when we create a BLEDevice() is what causes the value of 0x1800 -> 0x2a01 to be set?

nkolban, 2017-09-26:
Ohh ... wait again ... as soon as I re-read, it appears we have two characteristics ... 0x2a00 -> https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.device_name.xml and 0x2a01-> https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.appearance.xml which is appearance.

chegewara, 2017-09-26:
Yes, and both are mandatory, which means both are created somwhere in here: https://github.com/espressif/esp-idf/blob/a4fe12cb6d195a580a57690ceed45e1e06dc58e0/components/bt/bluedroid/stack/gap/gap_ble.c#L371

chegewara, 2017-09-26:
But i guess answer to the question is, you cant change appearance name in server. Its up to client to match service UUID and display desired name.

JabberwockPL, 2017-09-26:
Name... or an icon. So I think add icon is a good candidate, as it has two bytes, exactly as the apperance UUID.

JabberwockPL, 2017-09-26:
Or I could skip the guessing and check GATT_UUID_GAP_ICON:
https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/5738f83aeb59361a0a2eda2460113f6dc9194271/stack/include/gattdefs.h

... and it is a match: 0x2A01

chegewara, 2017-09-26:
Well, my guessing is not about how its handled, its about how good is this answer to you.

JabberwockPL, 2017-09-26:
Well, as I understand, the function is not exposed in the Arduino library, so for now I cannot do anything about it. However, I am not sure how useful it would be anyway - I mean, it would be extremely silly if the app developers have detected the sensors based on their _icons_ and not on the services they provided? So while I cannot exclude that, it might be that my emulated sensor differs in other ways from commercial sensors.

My next candidate is manufacturer's name, as most examples seem to have that set as well. That I can access creating the relevant service, so it should be no problem.

chegewara, 2017-09-26:
Almost found it, just need to investigate if it can be set in arduino.

chegewara, 2017-09-26:
OK, it is possible to do it with arduino ide. You need to those 2 includes:
`#include "gap_api.h"
#include "gattdefs.h"
`
this line 
`GAP_BleAttrDBUpdate(GATT_UUID_GAP_ICON, (tGAP_BLE_ATTR_VALUE ) *attr);`
and to figure out how to fill in attr. This is structure:
```
typedef union {
    tGAP_BLE_PREF_PARAM     conn_param;
    BD_ADDR                 reconn_bda;
    UINT16                  icon;
    UINT8                   *p_dev_name;
    UINT8                   addr_resolution;

} tGAP_BLE_ATTR_VALUE;
```

chegewara, 2017-09-26:
About manufacturer's name im not sure, but this is what i found. 
https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers

I found also https://github.com/nkolban/esp32-snippets/blob/92958bcb5193a902b6c8d44066658215de1ebc4b/cpp_utils/BLEAdvertising.cpp#L31. This is hard coded in Kolban's library, so its not very helpful.

JabberwockPL, 2017-09-26:
I have already set the manufacturer's name, it is a characteristic in a separate service (Device Info). I will try to use the code you have provided.

JabberwockPL, 2017-09-26:
I could not get your code to work, so I googled a bit and found this:
https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/5738f83aeb59361a0a2eda2460113f6dc9194271/stack/btm/btm_devctl.c#1086

I have copied the code verbatim and set p_name (just to test the name change), but it does not work, it throws 'unknown reference to' for the procedure.

I have dug some more and found that the advertising data also contains the appearance icon, which I thought might be even more import for detection. So between this:
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();

you just need to put this:
  pAdvertising->setAppearance(1157);

And it worked! nRF Connect even before the connection sees my device as a cycling sensor!

However, strangely enough, it _still_ asks for the characteristic 0x2A01 and it comes up with nothing. Oh, and the cycling apps still do not work... Back to investigations, I guess.

nkolban, 2017-09-26:
I have posed a question against the underlying ESP-IDF APIs ... see:

https://github.com/espressif/esp-idf/issues/1042

JabberwockPL, 2017-09-27:
Oh my, it seemed like a simple question at first! I hope for their answer...

chegewara, 2017-09-28:
I dont know where is the problem, but with this https://github.com/nkolban/esp32-snippets/issues/88#issuecomment-332282142 i am able to change advertising as you wish.

nkolban, 2017-09-28:
@chegewara My thinking goes as follows ...

The ESP-IDF exposes an API ... this is the API that Espressif wishes us to code against with the belief that the API exposed will be exposed the same on the ESP32 and potentially future Espressif products and stacks.  Today the ESP-IDF BLE happens to be implemented on top of a Bluetooth stack that I believe is called "Bluedroid" however there are other implementations/stacks in existence and the desire is to isolate/insulate ESP-IDF applications from a choice of an explicit stack.  The BLE C++ classes in this project code to the expose ESP-IDF and, so far, have not needed to drop lower to underlying implementation choices beneath the exposed ESP-IDF.

If there is a some merit in this thinking (and I could easily be mistaken or just plain wrong), then the ESP-IDF issue asks for an ESP-IDF mechanism to set the appearance/icon.  From what I understand, the referenced comment in the issue previously showed a technique where we could "drop down" through the ESP-IDF APIs and call the underlying BLE implementation code directly.  The request in the ESP-IDF Github repo is to expose an ESP-IDF API to achieve this task.  It would be extremely likely that the ESP-IDF implementation of that might be no more than a wrapped pass-through to the underlying implementation and may be exactly the API call that was previously referenced.  However, I'm holding out a desire for a *legal* ESP-IDF exposed API as opposed to coding through to an implementation.

chegewara, 2017-09-28:
No matter how it is exposed, this is how i see it. If you have field appearance in esp_ble_adv_data_t struct (as you can see its esp struct) and even if its set but isnt advertised or read from 0x1800 characteristic properly then something is setup wrong. Especially if this can be "fixed" by updating atrr db.

User should have no need to worry about appearance if during advertising initialization all is setup as he wish.

nkolban, 2017-09-28:
That's a good thought ... how separate should a "BLE Server" advertisement be from the actual properties that the advertised server truly exposes.  Should setting an attribute in the advertisement force the server to have the advertised property?  Should setting a property in the server force the advertisement to be consistent with the properties offered?   I can't say I disagree with either of those.

chegewara, 2017-09-28:
Its up to you. In my opinion it should be done in ble stack, but as for now you can add some code like:
`if(m_advData.appearanse != 0)
    GAP_BleAttrDBUpdate(GATT_UUID_GAP_ICON, &m_advData.appearance);
`
This can help users to set this properly and to avoid a lot of headache and/or deep digging.

JabberwockPL, 2017-09-28:
@chegewara 
I have tried to use your code and it did not work... Probably I have not initiated the data properly. Could you post the whole code?

chegewara, 2017-09-28:
@JabberwockPL  Like i said, it was tested with esp-idf, i have to prepare Arduino ide first but i have no time. But it should work ith arduino too
`#include "gap_api.h"`
`#include "gattdefs.h"
`
`tGAP_BLE_ATTR_VALUE attr;
`
`attr.icon = 0x485;
`
`GAP_BleAttrDBUpdate(GATT_UUID_GAP_ICON, &attr);
`

JabberwockPL, 2017-09-28:
This results in:

`undefined reference to `GAP_BleAttrDBUpdate(unsigned short, tGAP_BLE_ATTR_VALUE*)'
`

But no worries, it was most likely not the main source of the problem (but rather non-standard implementation of sensor handling by the apps, as I have written [here](https://github.com/nkolban/esp32-snippets/issues/89).

chegewara, 2017-09-28:
Try with this, should have working on arduino:

`extern "C"{
#include "gap_api.h"
#include "gattdefs.h"
}
`

JabberwockPL, 2017-09-28:
It compiles, but then goes into reboot loop:

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:732
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
Starting BLE work!
abort() was called at PC 0x400d975f on core 0

Backtrace: 0x40088330:0x3ffdb560 0x4008842f:0x3ffdb580 0x400d975f:0x3ffdb5a0 0x400dadce:0x3ffdb5c0 0x400daf5a:0x3ffdb5e0 0x400db2a5:0x3ffdb620 0x400db2e5:0x3ffdb650 0x400db2f5:0x3ffdb670 0x400d6a0a:0x3ffdb690 0x400d57dd:0x3ffdb6b0 0x400d461d:0x3ffdb6f0 0x4010e17d:0x3ffdb710 0x4010a03e:0x3ffdb750

```

Commenting out GAP_BleAttrDBUpdate(GATT_UUID_GAP_ICON, &attr); elimates the loop so it must be the cause.

JabberwockPL, 2017-10-08:
Update:
I have managed to set the appearance by using:

pAdvertising->setAppearance(1157);

However, after reading the specs again, I have realized that some apps might expect the service UUID to be exposed in advertising data themselves. For that purpose I have used:

pAdvertising->setServiceUUID(BLEUUID((uint16_t)0x1816));

but this returns the error:
E (5844) BLEAdvertising: << esp_ble_gap_config_adv_data: rc=258 Invalid argument

chegewara, 2017-10-09:
Good to hear.

@nkolban i have a question about BLEAdvertising class creator. There is 2 variables with exact same structure, m_advData and m_advDataScanResponse, and both are set as config adv data. What is purpose to have it both?

chegewara, 2017-10-09:
@JabberwockPL if you follow the routines then you can see that setServiceUUID() sets up m_advDataScanResponse.p_service_uuid, but this error is from this line: 
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp#L141

which is just after 
> errRc = ::esp_ble_gap_start_advertising(&m_advParams);

I dont know if somewhere else in your code m_advData.p_service_uuid is updated, but if not then from constructor it is nullptr. My advice is to try add this lines or change it with:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp#L100-L101

> m_advData.service_uuid_len = 2;
m_advData.p_service_uuid = reinterpret_cast<uint8_t*>(&espUUID->uuid.uuid16);

JabberwockPL, 2017-10-09:
No, that was not the problem. setServiceUUID is supposed to setup the value, so it does not need to be set up earlier. The actual problem is described here:

https://www.esp32.com/viewtopic.php?f=13&t=2253

All it took was to put the UUID in the long form:

pAdvertising->setServiceUUID("00001816-0000-1000-8000-00805f9b34fb");

now I can happily declare that almost all apps are detecting the sensor correctly!

Again, thank you very, very much, I could not have done it without the help...

nkolban, 2017-10-09:
@chegewara 
re: m_advData and m_advDataScanResponse

In BLE there are always *two* advertising packets.  One that is *always* broadcast and one that is returned to a client when *it* receives a broadcast advertisement and then says *I want the additional advertisement packet*.   I'm probably getting the technical words wrong here ... but I believe this is called a scan response.   At a high level ... imagine that the BLE Server is broadcasting adverts .... when a client receives an advert ... that may be all it needs ... however it is the clients *choice* to immediately turn around and ask the BLE Server *give me the scan response* advert.

In ESP-IDF, we define **two** adverts.  One which contains the always advertised packet ... and a second advert that is returned by the server *only* when the client asks for the scan response advert.

nkolban, 2017-10-09:
We may have a relationship between this issue and #115

chegewara, 2017-10-09:
Maybe this can explain something:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server_service_table/README.rst

Actually, there are two way to create server service and characteristics. One is use the esp_gatts_create_service or esp_ble_gatts_add_char and etc. The other way is use esp_ble_gatts_create_attr_tab. **The important things: the two ways cannot use in the same service, but can use in different service.**

chegewara, 2017-10-09:
https://github.com/nkolban/esp32-snippets/issues/88#issuecomment-335080299

@JabberwockPL ive found the problem why it is working this way
> pAdvertising->setServiceUUID("00001816-0000-1000-8000-00805f9b34fb");
but doesnt work that way
> pAdvertising->setServiceUUID(0x1816);
This is the clue:
> E (5844) BLEAdvertising: << esp_ble_gap_config_adv_data: rc=258 Invalid argument

@nkolban When we change this line it will works:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp#L100
m_advDataScanResponse.service_uuid_len = 16;

Why? I dont know. Just works.

nkolban, 2017-10-09:
The sad thing about this issue is that I created a post back in June warning others not to use a uuid_len of anything other than 16 ... and failed to heed my own advice:

https://www.esp32.com/viewtopic.php?f=13&t=2253

Taking an action to resolve properly.

nkolban, 2017-10-09:
Code changed ... commit pending.

nkolban, 2017-10-11:
Code changed.  Please let us know if we are good now.

JabberwockPL, 2017-10-11:
So I should now to try code it with:

pAdvertising->setServiceUUID(0x1816);

I wil do it, but it might take a bit, as it requires partial dismantling my bike installation (and I am having a blask using it :) ). But I do plan to have some other code changes, so I will get around to it.

nkolban, 2017-10-11:
The new API is:

```
pAdvertising->setServiceUUID(BLEUUID((uint16_t)0x1816);
```

JabberwockPL, 2017-10-11:
Ok, I will try that.

nkolban, 2017-10-29:
Closing for just now.  Re-open as needed.

closed 2017-10-29 by nkolban

muradjabir, 2024-03-13:
This can now be set using esp_ble_gap_config_local_icon() on both Arduino and the IDF.