nkolban/esp32-snippets #89: Issue with an application

JabberwockPL, 2017-09-27:
This is really low priority, so if the log does not tell you anything important right now, just ignore it.

Here is the log from communication with an app which was working fine just a few days ago. I do not think I have made any significant changes in the code, especially the parts related to BLE... My implementation is still BLEServer running Cadence/Speed sensor service.

```
D (221562) BLEServer: >> startAdvertising
D (221566) BLEAdvertising: >> start
D (221569) BLEAdvertising: << start
D (221572) BLEServer: << startAdvertising
D (221576) BLEServer: << handleGATTServerEvent
D (221582) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
D (221587) BLEUtils: [status: 0]
D (221590) BLEServer: BLEServer ... handling GAP event!
D (221595) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (221602) BLEUtils: [status: 0]
D (221605) BLEServer: BLEServer ... handling GAP event!
D (223462) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONNECT_EVT
D (223462) BLEUtils: GATT ServerEvent: ESP_GATTS_CONNECT_EVT
D (223464) BLEUtils: [conn_id: 0, remote_bda: 4a:de:a1:0e:d0:72, is_connected: 1]
D (223471) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
D (223477) BLEServer: << handleGATTServerEvent
D (223950) BLEUtils: gapEventToString: Unknown event type 0x14
D (223950) BLEUtils: Received a GAP event: Unknown event type
D (223950) BLEUtils: *** dumpGapEvent: Logger not coded ***
D (223955) BLEServer: BLEServer ... handling GAP event!
D (224269) BLEUtils: gapEventToString: Unknown event type 0x14
D (224269) BLEUtils: Received a GAP event: Unknown event type
D (224269) BLEUtils: *** dumpGapEvent: Logger not coded ***
D (224274) BLEServer: BLEServer ... handling GAP event!
D (224279) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_WRITE_EVT
D (224287) BLEUtils: GATT ServerEvent: ESP_GATTS_WRITE_EVT
D (224292) BLEUtils: [conn_id: 0, trans_id: 1, bda: 4a:de:a1:0e:d0:72, handle: 0x35, offset: 0, need_rsp: 1, is_prep: 0, len: 2]
D (224303) BLEUtils: [Data: 0100]
D (224306) BLEServer: >> handleGATTServerEvent: ESP_GATTS_WRITE_EVT
D (224312) BLEServer: << handleGATTServerEvent
D (224317) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_RESPONSE_EVT
D (224324) BLEUtils: GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
D (224330) BLEUtils: [status: ESP_GATT_OK, handle: 0x35]
D (224335) BLEServer: >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
D (224341) BLEServer: << handleGATTServerEvent
E (230557) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013
D (230558) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_DISCONNECT_EVT
D (230561) BLEUtils: GATT ServerEvent: ESP_GATTS_DISCONNECT_EVT
D (230567) BLEUtils: [conn_id: 0, remote_bda: 4a:de:a1:0e:d0:72, is_connected: 0]
D (230574) BLEServer: >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
D (230581) BLEServer: >> startAdvertising
D (230584) BLEAdvertising: >> start
D (230588) BLEAdvertising: << start
D (230591) BLEServer: << startAdvertising
D (230595) BLEServer: << handleGATTServerEvent
D (230601) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
D (230606) BLEUtils: [status: 0]
D (230609) BLEServer: BLEServer ... handling GAP event!
D (230614) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (230621) BLEUtils: [status: 0]
D (230624) BLEServer: BLEServer ... handling GAP event!
D (233499) BLEDevice: gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONNECT_EVT
D (233499) BLEUtils: GATT ServerEvent: ESP_GATTS_CONNECT_EVT
D (233501) BLEUtils: [conn_id: 0, remote_bda: 4a:de:a1:0e:d0:72, is_connected: 1]
D (233508) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
D (233515) BLEServer: << handleGATTServerEvent
D (233992) BLEUtils: gapEventToString: Unknown event type 0x14
D (233992) BLEUtils: Received a GAP event: Unknown event type
D (233992) BLEUtils: *** dumpGapEvent: Logger not coded ***
D (233997) BLEServer: BLEServer ... handling GAP event!

```

nkolban, 2017-09-27:
This is an awesome find ... there were a number of event types that we weren't decoding.  Even type 0x14 (20 decimal) was:

ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT

These have now been included in the debug log rather than saying "unknown type".  What remains to be done is to add then to the dumpGapEvent() logging.

nkolban, 2017-09-27:
We should be good now.

JabberwockPL, 2017-09-27:
Glad I could help! Do you think it is related to the reason why the app is not connecting?

nkolban, 2017-09-27:
Interestingly, I'm not seeing any actual *failures* in the trace ... just an inability to log the content of an incoming event that was previously unlogged.  It may be that event will hold some more clues.  If we can pin down the story some more .. the trace seems to show that the ESP32 is  being a BLE Server, is advertising, gets connected to by a BLE client ... which then writes two values and finally disconnects.  I'm not yet seeing a "problem".

JabberwockPL, 2017-09-27:
It is quite possible it is some bug or quirk in the app, as I wrote some other apps (but not all) work with the emulated sensor just fine. Only not those I would like to :)

I have noticed one more thing: I have turned off the BLE trace for a while to track other things on the serial and I have noticed the following sequence:
When trying to connect the faulty app I get:

E (287189) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013

(that one was in the trace)

When I try then nRF Connect (which still shows that all the data are correct), the log shows:

bta_gattc_process_indicate indication/notif for unknown device, ignore

which is a new one, and then again:

E (287189) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013

nkolban, 2017-09-27:
Those trace records are coming from the underlying ESP-IDF/BT/Bluedroid stack.   What I mean by that is you are programming to the BLE C++ APIs ... those are wrappers on the ESP-IDF API ... which are wrappers on the Bluedroid stack ... as I understand it.  My knowledge stops at the ESP-IDF API which is what the BLE C++ code is driving.  That said ... I'm VERY interested in things not working and well push hard to make them work.  The fact that we get "spurious" messages from lower stacks isn't that important (to me) until and unless they manifest in something that we expect to work not working.  Then they become clues and I'll hold my nose and jump into the swamp.

Am I correct in understanding that when you use nRF Connect as the partner to the ESP32, all the functions you expect to work ... actually do work?

JabberwockPL, 2017-09-27:
Yes, nRF Connect correctly displays all the data. There is also nRF Toolbox which is a more robust version that displays the cycling data in in correct context (i.e. converted to km/h etc.) and it also works perfectly. Curiously, simpler apps for reporting speed/cadence also work. The problem is with bigger sporting suits with logging abilities (which I would like to have) - they either do not detect the sensor at all (most of them) or detect it, but do not read the data.

I suspect that the simpler apps (and nRF) are built simply around the Bluetooth specs, just like your code and mine, so I am giving them exactly what they expect. On other hand, the sporting apps are developed for and tested with actual hardware commercial sensors, which are either doing something else or doing something not prescribed by the standard.

The only exception is that single app which both saw my sensor data AND allowed for online logging. But then it stopped working...

JabberwockPL, 2017-09-27:
Some new info... One difference between commercial sensors and my code seems to be that the sensors are sending the notifications at a steady rate, while my emulator sent notification only at the data change. This helped with detection in some applications. The rest of the apps might look for devices of specific vendors.

However, ESP32 does crash from time to time still. The log entries are now different:

```
E (75829) BT: Call back not found for application conn_id=3
E (145638) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013
E (148041) BT: lmp_version_below LMP version 6 < 8
E (148042) BT: l2cble_start_conn_update, the last connection update command still pending.
E (148176) BT: AES128_CMAC started, allocate buffer size = 80
E (148176) BT: flag = 0 round = 5
E (148195) BT: AES128_CMAC started, allocate buffer size = 32
E (148196) BT: flag = 1 round = 2
E (148196) BT: AES128_CMAC started, allocate buffer size = 64
E (148198) BT: flag = 0 round = 4
E (148202) BT: AES128_CMAC started, allocate buffer size = 64
E (148207) BT: flag = 0 round = 4
E (148210) BT: AES128_CMAC started, allocate buffer size = 80
E (148215) BT: flag = 0 round = 5
E (148219) BT: AES128_CMAC started, allocate buffer size = 80
E (148224) BT: flag = 0 round = 5
E (148246) BT: FOR LE SC LTK IS USED INSTEAD OF STK
E (154891) BT: Call back not found for application conn_id=3

```

I have also noticed that if the BLE trace is turned on, no notifications are sent (i.e. even nRF does not see them).

nkolban, 2017-09-27:
The phrase "AES" might be a clue for us.  I have seen that related to security phrases and there is currently nothing in the BLE C++ classes that is driving security.  Ive been waiting for a scenario where a user (maybe yourself) says "Hi ... I need security ... and here is the scenario and a test environment to get it working".

JabberwockPL, 2017-09-28:
I will try to help you, if I can... For now I have checked that reason 0x0013 is „connection terminate by peer user”. I have also found this:
https://github.com/espressif/esp-idf/issues/832

It also seems as if the other device disconnected due to reasons related to security (no dev CB).

chegewara, 2017-12-19:
@JabberwockPL Now we have introduced security in BLE stack, and from logs i can see you are trying to connect to device which requires secured connection, so we can try to make this happen.

JabberwockPL, 2017-12-19:
It seems the main problem was that the app did not advertise correctly its properties. Since I have fixed that, it has been running without a problem, although the sad truth is I do not use the bike that often (why make a new New Year's resolution list when you can recycle the old one...).

hansmbakker, 2018-12-04:
@JabberwockPL this issue is quite old. Is the issue still there? If not, please close the issue so that the issue list can be cleaned up a little :)

@chegewara

JabberwockPL, 2018-12-04:
Apologies, I was sure it was closed together with the reference!

closed 2018-12-04 by JabberwockPL