nkolban/esp32-snippets #92: BLE member List as String

pit001, 2017-09-28:
typedef struct {
  uint32_t assignedNumber;
  std::string name;
} member_t;

static const member_t members_ids[] = {
  {0xFE08, "Microsoft"},
  {0xFE09, "Pillsy, Inc."},
  {0xFE0A, "ruwido austria gmbh"},
  {0xFE0B, "ruwido austria gmbh"},
  {0xFE0C, "Procter & Gamble"},
  {0xFE0D, "Procter & Gamble"},
  {0xFE0E, "Setec Pty Ltd"},
  {0xFE0F, "Philips Lighting B.V."},
  {0xFE10, "Lapis Semiconductor Co., Ltd."},
  {0xFE11, "GMC-I Messtechnik GmbH"},
  {0xFE12, "M-Way Solutions GmbH"},
  {0xFE13, "Apple Inc."},
  {0xFE14, "Flextronics International USA Inc."},
  {0xFE15, "Amazon Fulfillment Services, Inc."},
  {0xFE16, "Footmarks, Inc."},
  {0xFE17, "Telit Wireless Solutions GmbH"},
  {0xFE18, "Runtime, Inc."},
  {0xFE19, "Google Inc."},
  {0xFE1A, "Tyto Life LLC"},
  {0xFE1B, "Tyto Life LLC"},
  {0xFE1C, "NetMedia, Inc."},
  {0xFE1D, "Illuminati Instrument Corporation"},
  {0xFE1E, "Smart Innovations Co., Ltd"},
  {0xFE1F, "Garmin International, Inc."},
  {0xFE20, "Emerson"},
  {0xFE21, "Bose Corporation"},
  {0xFE22, "Zoll Medical Corporation"},
  {0xFE23, "Zoll Medical Corporation"},
  {0xFE24, "August Home Inc"},
  {0xFE25, "Apple, Inc. "},
  {0xFE26, "Google Inc."},
  {0xFE27, "Google Inc."},
  {0xFE28, "Ayla Networks"},
  {0xFE29, "Gibson Innovations"},
  {0xFE2A, "DaisyWorks, Inc."},
  {0xFE2B, "ITT Industries"},
  {0xFE2C, "Google Inc."},
  {0xFE2D, "SMART INNOVATION Co.,Ltd"},
  {0xFE2E, "ERi,Inc."},
  {0xFE2F, "CRESCO Wireless, Inc"},
  {0xFE30, "Volkswagen AG"},
  {0xFE31, "Volkswagen AG"},
  {0xFE32, "Pro-Mark, Inc."},
  {0xFE33, "CHIPOLO d.o.o."},
  {0xFE34, "SmallLoop LLC"},
  {0xFE35, "HUAWEI Technologies Co., Ltd"},
  {0xFE36, "HUAWEI Technologies Co., Ltd"},
  {0xFE37, "Spaceek LTD"},
  {0xFE38, "Spaceek LTD"},
  {0xFE39, "TTS Tooltechnic Systems AG & Co. KG"},
  {0xFE3A, "TTS Tooltechnic Systems AG & Co. KG"},
  {0xFE3B, "Dolby Laboratories"},
  {0xFE3C, "Alibaba"},
  {0xFE3D, "BD Medical"},
  {0xFE3E, "BD Medical"},
  {0xFE3F, "Friday Labs Limited"},
  {0xFE40, "Inugo Systems Limited"},
  {0xFE41, "Inugo Systems Limited"},
  {0xFE42, "Nets A/S "},
  {0xFE43, "Andreas Stihl AG & Co. KG"},
  {0xFE44, "SK Telecom "},
  {0xFE45, "Snapchat Inc"},
  {0xFE46, "B&O Play A/S "},
  {0xFE47, "General Motors"},
  {0xFE48, "General Motors"},
  {0xFE49, "SenionLab AB"},
  {0xFE4A, "OMRON HEALTHCARE Co., Ltd."},
  {0xFE4B, "Philips Lighting B.V."},
  {0xFE4C, "Volkswagen AG"},
  {0xFE4D, "Casambi Technologies Oy"},
  {0xFE4E, "NTT docomo"},
  {0xFE4F, "Molekule, Inc."},
  {0xFE50, "Google Inc."},
  {0xFE51, "SRAM"},
  {0xFE52, "SetPoint Medical"},
  {0xFE53, "3M"},
  {0xFE54, "Motiv, Inc."},
  {0xFE55, "Google Inc."},
  {0xFE56, "Google Inc."},
  {0xFE57, "Dotted Labs"},
  {0xFE58, "Nordic Semiconductor ASA"},
  {0xFE59, "Nordic Semiconductor ASA"},
  {0xFE5A, "Chronologics Corporation"},
  {0xFE5B, "GT-tronics HK Ltd"},
  {0xFE5C, "million hunters GmbH"},
  {0xFE5D, "Grundfos A/S"},
  {0xFE5E, "Plastc Corporation"},
  {0xFE5F, "Eyefi, Inc."},
  {0xFE60, "Lierda Science & Technology Group Co., Ltd."},
  {0xFE61, "Logitech International SA"},
  {0xFE62, "Indagem Tech LLC"},
  {0xFE63, "Connected Yard, Inc."},
  {0xFE64, "Siemens AG"},
  {0xFE65, "CHIPOLO d.o.o."},
  {0xFE66, "Intel Corporation"},
  {0xFE67, "Lab Sensor Solutions"},
  {0xFE68, "Qualcomm Life Inc"},
  {0xFE69, "Qualcomm Life Inc"},
  {0xFE6A, "Kontakt Micro-Location Sp. z o.o."},
  {0xFE6B, "TASER International, Inc."},
  {0xFE6C, "TASER International, Inc."},
  {0xFE6D, "The University of Tokyo"},
  {0xFE6E, "The University of Tokyo"},
  {0xFE6F, "LINE Corporation"},
  {0xFE70, "Beijing Jingdong Century Trading Co., Ltd."},
  {0xFE71, "Plume Design Inc"},
  {0xFE72, "St. Jude Medical, Inc."},
  {0xFE73, "St. Jude Medical, Inc."},
  {0xFE74, "unwire"},
  {0xFE75, "TangoMe"},
  {0xFE76, "TangoMe"},
  {0xFE77, "Hewlett-Packard Company"},
  {0xFE78, "Hewlett-Packard Company"},
  {0xFE79, "Zebra Technologies"},
  {0xFE7A, "Bragi GmbH"},
  {0xFE7B, "Orion Labs, Inc."},
  {0xFE7C, "Telit Wireless Solutions (Formerly Stollmann E+V GmbH)"},
  {0xFE7D, "Aterica Health Inc."},
  {0xFE7E, "Awear Solutions Ltd"},
  {0xFE7F, "Doppler Lab"},
  {0xFE80, "Doppler Lab"},
  {0xFE81, "Medtronic Inc."},
  {0xFE82, "Medtronic Inc."},
  {0xFE83, "Blue Bite"},
  {0xFE84, "RF Digital Corp"},
  {0xFE85, "RF Digital Corp"},
  {0xFE86, "HUAWEI Technologies Co., Ltd. ( )"},
  {0xFE87, "Qingdao Yeelink Information Technology Co., Ltd. ( )"},
  {0xFE88, "SALTO SYSTEMS S.L."},
  {0xFE89, "B&O Play A/S"},
  {0xFE8A, "Apple, Inc."},
  {0xFE8B, "Apple, Inc."},
  {0xFE8C, "TRON Forum"},
  {0xFE8D, "Interaxon Inc."},
  {0xFE8E, "ARM Ltd"},
  {0xFE8F, "CSR"},
  {0xFE90, "JUMA"},
  {0xFE91, "Shanghai Imilab Technology Co.,Ltd"},
  {0xFE92, "Jarden Safety & Security"},
  {0xFE93, "OttoQ Inc."},
  {0xFE94, "OttoQ Inc."},
  {0xFE95, "Xiaomi Inc."},
  {0xFE96, "Tesla Motor Inc."},
  {0xFE97, "Tesla Motor Inc."},
  {0xFE98, "Currant, Inc."},
  {0xFE99, "Currant, Inc."},
  {0xFE9A, "Estimote"},
  {0xFE9B, "Samsara Networks, Inc"},
  {0xFE9C, "GSI Laboratories, Inc."},
  {0xFE9D, "Mobiquity Networks Inc"},
  {0xFE9E, "Dialog Semiconductor B.V."},
  {0xFE9F, "Google Inc."},
  {0xFEA0, "Google Inc."},
  {0xFEA1, "Intrepid Control Systems, Inc."},
  {0xFEA2, "Intrepid Control Systems, Inc."},
  {0xFEA3, "ITT Industries"},
  {0xFEA4, "Paxton Access Ltd"},
  {0xFEA5, "GoPro, Inc."},
  {0xFEA6, "GoPro, Inc."},
  {0xFEA7, "UTC Fire and Security"},
  {0xFEA8, "Savant Systems LLC"},
  {0xFEA9, "Savant Systems LLC"},
  {0xFEAA, "Google Inc."},
  {0xFEAB, "Nokia Corporation"},
  {0xFEAC, "Nokia Corporation"},
  {0xFEAD, "Nokia Corporation"},
  {0xFEAE, "Nokia Corporation"},
  {0xFEAF, "Nest Labs Inc."},
  {0xFEB0, "Nest Labs Inc."},
  {0xFEB1, "Electronics Tomorrow Limited"},
  {0xFEB2, "Microsoft Corporation"},
  {0xFEB3, "Taobao"},
  {0xFEB4, "WiSilica Inc."},
  {0xFEB5, "WiSilica Inc."},
  {0xFEB6, "Vencer Co, Ltd"},
  {0xFEB7, "Facebook, Inc."},
  {0xFEB8, "Facebook, Inc."},
  {0xFEB9, "LG Electronics"},
  {0xFEBA, "Tencent Holdings Limited"},
  {0xFEBB, "adafruit industries"},
  {0xFEBC, "Dexcom, Inc. "},
  {0xFEBD, "Clover Network, Inc."},
  {0xFEBE, "Bose Corporation"},
  {0xFEBF, "Nod, Inc."},
  {0xFEC0, "KDDI Corporation"},
  {0xFEC1, "KDDI Corporation"},
  {0xFEC2, "Blue Spark Technologies, Inc."},
  {0xFEC3, "360fly, Inc."},
  {0xFEC4, "PLUS Location Systems"},
  {0xFEC5, "Realtek Semiconductor Corp."},
  {0xFEC6, "Kocomojo, LLC"},
  {0xFEC7, "Apple, Inc."},
  {0xFEC8, "Apple, Inc."},
  {0xFEC9, "Apple, Inc."},
  {0xFECA, "Apple, Inc."},
  {0xFECB, "Apple, Inc."},
  {0xFECC, "Apple, Inc."},
  {0xFECD, "Apple, Inc."},
  {0xFECE, "Apple, Inc."},
  {0xFECF, "Apple, Inc."},
  {0xFED0, "Apple, Inc."},
  {0xFED1, "Apple, Inc."},
  {0xFED2, "Apple, Inc."},
  {0xFED3, "Apple, Inc."},
  {0xFED4, "Apple, Inc."},
  {0xFED5, "Plantronics Inc."},
  {0xFED6, "Broadcom Corporation"},
  {0xFED7, "Broadcom Corporation"},
  {0xFED8, "Google Inc."},
  {0xFED9, "Pebble Technology Corporation"},
  {0xFEDA, "ISSC Technologies Corporation"},
  {0xFEDB, "Perka, Inc."},
  {0xFEDC, "Jawbone"},
  {0xFEDD, "Jawbone"},
  {0xFEDE, "Coin, Inc."},
  {0xFEDF, "Design SHIFT"},
  {0xFEE0, "Anhui Huami Information Technology Co."},
  {0xFEE1, "Anhui Huami Information Technology Co."},
  {0xFEE2, "Anki, Inc."},
  {0xFEE3, "Anki, Inc."},
  {0xFEE4, "Nordic Semiconductor ASA"},
  {0xFEE5, "Nordic Semiconductor ASA"},
  {0xFEE6, "Silvair, Inc."},
  {0xFEE7, "Tencent Holdings Limited"},
  {0xFEE8, "Quintic Corp."},
  {0xFEE9, "Quintic Corp."},
  {0xFEEA, "Swirl Networks, Inc."},
  {0xFEEB, "Swirl Networks, Inc."},
  {0xFEEC, "Tile, Inc."},
  {0xFEED, "Tile, Inc."},
  {0xFEEE, "Polar Electro Oy"},
  {0xFEEF, "Polar Electro Oy"},
  {0xFEF0, "Intel"},
  {0xFEF1, "CSR"},
  {0xFEF2, "CSR"},
  {0xFEF3, "Google Inc."},
  {0xFEF4, "Google Inc."},
  {0xFEF5, "Dialog Semiconductor GmbH"},
  {0xFEF6, "Wicentric, Inc."},
  {0xFEF7, "Aplix Corporation"},
  {0xFEF8, "Aplix Corporation"},
  {0xFEF9, "PayPal, Inc."},
  {0xFEFA, "PayPal, Inc."},
  {0xFEFB, "Telit Wireless Solutions (Formerly Stollmann E+V GmbH)"},
  {0xFEFC, "Gimbal, Inc."},
  {0xFEFD, "Gimbal, Inc."},
  {0xFEFE, "GN ReSound A/S"},
  {0xFEFF, "GN Netcom"},
  {0, "" }
};

std::string getMember(uint32_t memberId)
{
  member_t* p = (member_t *)members_ids;
  
  while (p->name.length() > 0) {
    if (p->assignedNumber == memberId) {
      return p->name;
    }
    p++;
  }
  return "Unknown";
}

nkolban, 2017-09-28:
Thanks you again sir.  Another great set of changes.  These have now been made and committed into the code.  Thank you again.

pit001, 2017-10-09:
please add as n-1 line the 

`	{0xFFFF, "Reserved"},`

thx

nkolban, 2017-10-09:
Done and did ... make sure that you re-open issues when you need work done.  Posts to closed issues can get missed.

closed 2017-10-09 by nkolban