nkolban/esp32-snippets #100: Save memory/flash by optimize the Strings in BLEUtils.cpp

pit001, 2017-09-29:
In BLEUtils.cpp: Place the large Strings (members,services) into the PROGMEN (if this is existing with ESP32!?) and/or move the Structs into seperate files that the linker is able to only link the needed parts.

nkolban, 2017-09-29:
In Arduino, the notion of PROGMEM says that static data (strings, constants etc) should be stored in flash memory and referenced from there as opposed to being copied to RAM.  It is my understanding that memory on the ESP32 is managed differently and that data defined as "const" is by default kept in flash already as opposed to being copied to RAM.  Since we have 4 MBytes of flash to work with, I'm not necessarily concerned with running out of flash.  I'd like to suggest that we table this item until and unless we find applications that start to struggle with flash memory availability or until we find that what we though was going to be in flash has actually been moved to RAM.  My belief is that const data is in flash and not RAM.  The large set of strings are currently tagged as "static const" and hence should be in flash.

nkolban, 2017-10-14:
Closing for just now.  Don't hesitate to re-open as needed.

pit001, 2017-11-06:
Hi - I think you are with the PROGMEM at ESP32. However - we should move the Strings and the hcalling functions to a seperate .h file. If not used in the main programm the linker will not at it to the binary. 
 
Ram: in the Arduino enviroment we only have 1M Flash  - to compile my project i have to increase it to 2 (1,5) Mb.

chegewara, 2017-11-06:
Building app with this framework, or any framework, is like folding piece of paper. Every layer is increasing binary file. But there is difference between binary file size, which is consuming flash memory and ram/heap which is used to allocate memory for variables or stack.
This optimization was to save some ram/heap which is very quick consumed by every created task or initialized class. Also wifi and BT stacks allocate a lot ram/heap before our code even start running.

nkolban, 2017-11-06:
@pit001 See ... I think this is what was desired?

https://github.com/nkolban/esp32-snippets/issues/162

pit001, 2017-11-06:
no - it's different
with the use of only one fnc in the BLEhelp, but not the use of any String the complete c file (object file / binary) is linket into the flash. With different files the linker is able to only link the needed .o files. 

btw. good to see that you work at ram/flash optimizations :)

nkolban, 2017-11-06:
I think I see what you are saying.  If a source file contains two functions and we reference one of them in our code, both are linked with the final executable increasing its size.  The solution is to partition the source into multiple "translatable units" (source files) such that there are multiple object files created and then at link time, only those objects that need resolved get linked in.

Is that the notion?

Obviously, it is desirable not to waste resources however we have to balance that against the time it takes to refactor and re-work the code.  Like many open source projects, this one is built by hobbyists in evenings and weekends and as such there is limited time.  The movement of strings from RAM to flash was essential as projects were literally breaking because of lack of RAM.  With 4MBytes of flash, we haven't yet seen a project break because of running out of flash.

Do you have such a scenario where we are getting low on flash?   Next, we would have to segment and sort the methods and flash held data such that we could sensibly factor them into separate source files.  No use factoring code into a source file when the result would be that it is always needed and no use factoring out a source if it contributed only a few hundred bytes to the resulting compiled code.  I have been using "nm" and related tools to look for "data" (RAM) usage ... we would presumably use the same principle to look for flash usage.

Is this in line with your thinking?

pit001, 2017-11-06:
Yes, it is the notion. 
It's very siple to realize (only for the arrays of strings like ventor, Gatt, ...) and as I wrote - the Arduino IDE yet is configured for 1M flash (and this is not bad - we need space for OTA or FFS). In my Sniffer Example I have to increase it. Also in my understanding during program run parts of the program code is copied from flash to execution ram ...

nkolban, 2017-11-28:
Closing for just now.

closed 2017-11-28 by nkolban