nkolban/esp32-snippets #101: Feature: getHandle()

pit001, 2017-09-29:
For diagnostic a ~getHandle(uuid) function would be nice. 

It's not necessary that the API also is handle ready instead of UUID. 
Peter

nkolban, 2017-09-29:
Howdy Peter,
Can you expand with more commentary on this one.  In which class would you like a getHandle()?  Could the "toString()" methods of the existing classes be used for the diagnostics desired?

pit001, 2017-09-30:
Hi, I simply add 

uint16_t BLERemoteCharacteristic::getHandle() {
return getHandle();
}

in 

BLERemoteCharacteristic.cpp /.h
for Characteristic handles... (most interesting)

nkolban, 2017-09-30:
Part of my thinking ... and I could be very wrong here ... was trying to hide "handles" from the end user (the BLE programmer).  If one is writing a BLE app using these classes, I wanted one to think exclusively in terms of the class rather than the "handle".  The handle obviously exists ... it is part of the BLE world ... but I was trying to hide its existence from consumers.  Can you help me understand why we would want to expose a handle to a BLE C++ API programmer?

pit001, 2017-09-30:
As I allready try to show: It's informative only, but's interesting during development, because it's the only thing visable in the communication (i.e. with wireshark) and often also present in the web and in some documentation.

nkolban, 2017-09-30:
It looks like the toString() method of BLERemoteCharacteristic produces a string that could be logged for debugging which contains the handle, uuid and characteristic properties.  Would that not be sufficient for debugging purposes during development?

pit001, 2017-09-30:
both values are int16, as a String ist a bit bothersome to convert it to dec and hex.. but OK

nkolban, 2017-10-07:
You have happily convinced me that exposing the handles is *not a bad thing*.  As such BLERemoteService, BLERemoteCharacteristic and BLERemoteDescriptor now expose a method called "getHandle()".  Let me know if there are additional changes requested for this issue.

nkolban, 2017-10-23:
Assuming resolved.  Don't hesitate to re-open as needed.

closed 2017-10-23 by nkolban